openwrt/openwrt #13488: cannot build frp

ChaingTsung, 2023-09-19:
### Describe the bug

```log
Finding targets
go: downloading github.com/gabriel-vasile/mimetype v1.4.2
/home/yoki/op/dl/go-mod-cache/github.com/go-playground/validator/v10@v10.14.1/baked_in.go:25:2: github.com/gabriel-vasile/mimetype@v1.4.2: reading https://proxy.golang.org/github.com/gabriel-vasile/mimetype/@v/v1.4.2.zip: 403 Forbidden

Building target
go: downloading github.com/gabriel-vasile/mimetype v1.4.2
/home/yoki/op/dl/go-mod-cache/github.com/go-playground/validator/v10@v10.14.1/baked_in.go:25:2: github.com/gabriel-vasile/mimetype@v1.4.2: reading https://proxy.golang.org/github.com/gabriel-vasile/mimetype/@v/v1.4.2.zip: 403 Forbidden

make[3]: *** [Makefile:70: /home/yoki/op/build_dir/target-aarch64_cortex-a53_musl/frp-0.51.3/.built] Error 1
make[3]: Leaving directory '/home/yoki/op/feeds/packages/net/frp'
time: package/feeds/packages/frp/compile#0.27#0.18#0.78
    ERROR: package/feeds/packages/frp failed to build.

```

### OpenWrt version

`main` branch

### OpenWrt target/subtarget

banana pi r3

### Device

banana pi r3

### Image kind

Self-built image

### Steps to reproduce

- git pull source 
- update & install all feeds
- make -j1 V=s (with oldconfig)
- errors occurred

### Actual behaviour

```log
...
Finding targets
go: downloading github.com/gabriel-vasile/mimetype v1.4.2
/home/yoki/op/dl/go-mod-cache/github.com/go-playground/validator/v10@v10.14.1/baked_in.go:25:2: github.com/gabriel-vasile/mimetype@v1.4.2: reading https://proxy.golang.org/github.com/gabriel-vasile/mimetype/@v/v1.4.2.zip: 403 Forbidden

Building targets
go: downloading github.com/gabriel-vasile/mimetype v1.4.2
/home/yoki/op/dl/go-mod-cache/github.com/go-playground/validator/v10@v10.14.1/baked_in.go:25:2: github.com/gabriel-vasile/mimetype@v1.4.2: reading https://proxy.golang.org/github.com/gabriel-vasile/mimetype/@v/v1.4.2.zip: 403 Forbidden

make[3]: *** [Makefile:70: /home/yoki/op/build_dir/target-aarch64_cortex-a53_musl/frp-0.51.3/.built] Error 1
make[3]: Leaving directory '/home/yoki/op/feeds/packages/net/frp'
time: package/feeds/packages/frp/compile#0.27#0.18#0.78
    ERROR: package/feeds/packages/frp failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/frp/compile] Error 1
make[2]: Leaving directory '/home/yoki/op'
make[1]: *** [package/Makefile:114: /home/yoki/op/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/op'
make: *** [/home/yoki/op/include/toplevel.mk:232: world] Fehler 2
```

### Expected behaviour

built successfully

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ChaingTsung, 2023-09-19:
[seeds.config.txt](https://github.com/openwrt/openwrt/files/12661153/seeds.config.txt)

brada4, 2023-09-19:
Try setting `GOPROXY=direct`  and report to ../packages/ if there are any problems. with OpenWRT go package.