openwrt/openwrt #13489: [23.05.0-rc3] ASUS RT-AX53U | Status > Overview "InternalError  too much recursion"

SeimusS, 2023-09-19:
### Describe the bug

On the latest 23.05.0-rc3 the Status > Overview wont load it gives "InternalError  too much recursion". Tried as well 23.05.0-rc2 this issue was not seen on this release.

I checked with other user that is running same Router and same OS version/same snapshot and he doesnt have this issue. The main difference is that I am using VLANs and several VAPs

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Go to Status > Overview. Must have VLANs.

### Actual behaviour

Cant load Overview in luci

### Expected behaviour

Status > Overview should show general overview

### Additional info

Output from browser:

RPCError: RPC call to luci/getBuiltinEthernetPorts failed with error -32002: Access denied
  at handleCallReply (https://10.0.0.152/luci-static/resources/rpc.js?v=git-23.236.53405-fc638c8:11:3)
    raise https://10.0.0.152/luci-static/resources/luci.js?v=git-23.236.53405-fc638c8:155
    handleCallReply https://10.0.0.152/luci-static/resources/rpc.js?v=git-23.236.53405-fc638c8:11
[luci.js:161:9](https://10.0.0.152/luci-static/resources/luci.js?v=git-23.236.53405-fc638c8)
InternalError: too much recursion
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
[luci.js:161:9](https://10.0.0.152/luci-static/resources/luci.js?v=git-23.236.53405-fc638c8)
Uncaught (in promise) InternalError: too much recursion
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39
    resolveVLANPorts https://10.0.0.152/luci-static/resources/view/status/include/29_ports.js?v=git-23.236.53405-fc638c8:39

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-19:
clear web cache or try in private/incognito window.

brada4, 2023-09-19:
also try clearing cookies for the "site" and log in via deep links to pages:
/cgi-bin/luci/admin/network/network 

/cgi-bin/luci/admin/system/system

SeimusS, 2023-09-20:
Clearing cache and cookies doesnt help, I tried as well 3 different browsers still the same.

To be more precise on the issue, only the status/overview doesnt load "**/cgi-bin/luci/admin/status/overview**"

[image]

This references to _RPCError: RPC call to luci/getBuiltinEthernetPorts failed with error -32002: Access denied_

Everything else is working fine.

Regards,
S.

brada4, 2023-09-20:
Does 'logread' say anything bad from ubus or rpcd?

brada4, 2023-09-20:
It should go to ../luci as this seems to pertain js web app only.

SeimusS, 2023-09-20:
Thank you for the directions. I will close the ticket here and open on the luci subgit

Regards,
S.

closed 2023-09-20 by SeimusS