openwrt/openwrt #13492: No wifi on latest 23.05 Snapshot

mcaptur, 2023-09-20:
### Describe the bug

Wifi radio fails (on 2 x xiaomi mir3g routers) to load with following error in log

```
Wed Sep 20 03:31:33 2023 daemon.notice netifd: radio1 (2178): Command failed: ubus call hostapd config_set { "phy": "phy1", "config": "", "prev_config": "/var/run/hostapd-phy1.conf" } (Method not found)
Wed Sep 20 03:31:33 2023 daemon.notice netifd: radio1 (2178): Command failed: ubus call wpa_supplicant config_set { "phy": "phy1", "config": [] } (Method not found)
Wed Sep 20 03:31:33 2023 daemon.notice netifd: radio1 (2178): Failed to open "/usr/share/hostap/wdev.uc": No such file or directory
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Failed to open "/usr/share/hostap/wdev.uc": No such file or directory
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Command failed: ubus call wpa_supplicant config_set { "config": [ ], "phy": "phy1", "defer": true } (Method not found)
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Interface  setup failed: WPA_SUPPLICANT_FAILED
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Failed to parse json data: unexpected end of data
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Command failed: ubus call network.wireless notify { "command": 2, "device": "radio1", "data": { "pid": 0, "exe": "\/usr\/sbin\/wpad", "required": true, "keep": true } } (Invalid argument)
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Usage: ubus [<options>] <command> [arguments...]
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Options:
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  -s <socket>:		Set the unix domain socket to connect to
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  -S:			Use simplified output (for scripts)
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  -v:			More verbose output
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  -m <type>:		(for monitor): include a specific message type
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): 			(can be used more than once)
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Commands:
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  - list [<path>]			List objects
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  - call <path> <method> [<message>]	Call an object method
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  - listen [<path>...]			Listen for events
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  - send <type> [<message>]		Send an event
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):  - monitor				Monitor ubus traffic
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178):
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Command failed: ubus call hostapd config_set { "phy": "phy1", "config":"/var/run/hostapd-phy1.conf", "prev_config": "/var/run/hostapd-phy1.conf.prev"} (Method not found)
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Device setup failed: HOSTAPD_START_FAILED
Wed Sep 20 03:31:34 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
Wed Sep 20 03:31:34 2023 daemon.notice netifd: radio1 (2178): Failed to open "/usr/share/hostap/wdev.uc": No such file or directory
Wed Sep 20 03:31:34 2023 daemon.notice netifd: Wireless device 'radio1' is now up
```

### OpenWrt version

r23455-83bf45ea5c

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3G

### Image kind

Self-built image

### Steps to reproduce

Upgrade device..

Latest build i have which works is d536f398eb

### Actual behaviour

no wifi

### Expected behaviour

have wifi

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ne20002, 2023-09-20:
I have the same problem on BananaPi R3.

mcaptur, 2023-09-20:
> I have the same problem on mediatek/filogic BananaPi R3 after Attended Sysupgrade this morning
> 
> @nbd168 I believe this has something to do with this [commit](https://github.com/openwrt/openwrt/commit/09b9d732ec76a0cf0c80223f1c0b991139dd4e82)

no a build with this (commit 8b385a4)  fails also https://github.com/openwrt/openwrt/commit/8b385a45a65cb4721e6bb2fbb24ce080417e331f

so something btw commit 385a50f and commit 8b385a4 broke it

bfqrst, 2023-09-20:
So this landed a couple of hours ago: `https://github.com/openwrt/openwrt/commit/e9722aef9e84110331706f74f7de0942c8d657ed`. Build pipelines are still running...

marcust, 2023-09-20:
But this is on `master`, is that also affected? Because this bug is about 23.05 Snapshots, so I guess that would have to be backported to `23.05` then.

(I had the same issue on a WAX206)

bfqrst, 2023-09-20:
> But this is on `master`, is that also affected? Because this bug is about 23.05 Snapshots, so I guess that would have to be backported to `23.05` then.
> 
> (I had the same issue on a WAX206)

You're right, my guess would be that the `hostap` and `ucode` packages need to be bumped as well... At least in my case, after updating those few, wifi broke.

Code-TT, 2023-09-20:
Same issue on OpenWrt 23.05.0-rc3 build with firmware-selector.openwrt.org

mcaptur, 2023-09-21:
built commit eda1545 including
hostapd: add missing NULL pointer check in uc_hostapd_iface_stop
hostapd: fix a crash when disabling an interface during channel list …

still same error

mcaptur, 2023-09-21:
Built again now with image builder and it works fine.. so probably the packages needed to be rebuilt..

closed 2023-09-21 by mcaptur

kamil-michalak, 2023-09-22:
Still same issue on 23.05-rc3

ghoffman9, 2023-09-22:
on askey rac2v1k i use 23.05-snapshot with mbedtls (full).  wifi is not working since last good buid for me was r23429.  
if i use 23.05-rc3 and update via opkg update or auc, then wifi fails.
so something with wpad beyond mbedtls-basic?

Jonbar87, 2025-04-29:
I'm having the same problem trying to build an image for the GL-iNet GL-A1300 Slate Plus. I've tried manually creating phy0-sta0, creating the wifi networks manually, etc.

I'm using the following make command to build the image:
image_name=$(date +"%Y-%m-%d-%H%M")

make image EXTRA_IMAGE_NAME="$image_name" FILES="custom_files" PROFILE="glinet_gl-a1300" PACKAGES="openvpn-openssl coreutils-shuf kmod-ath10k-ct ath10k-firmware-qca4019-ct luci luci-app-opkg luci-app-firewall luci-app-commands openvpn-openssl luci-app-openvpn"

No matter what I try, I still get the following error fresh after building: 
logread | grep netifd
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'lan' is enabled
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'lan' is setting up now
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'lan' is now up
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'loopback' is enabled
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'loopback' is setting up now
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'loopback' is now up
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'wan' is enabled
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'wan6' is enabled
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Network device 'eth0' link is up
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Network device 'lo' link is up
Mon Oct  9 21:48:00 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon Oct  9 21:48:00 2023 daemon.notice netifd: radio0 (1588): WARNING: Variable 'data' does not exist or is not an array/object
Mon Oct  9 21:48:00 2023 daemon.notice netifd: radio0 (1588): Command failed: Not found
Mon Oct  9 21:48:00 2023 daemon.notice netifd: radio0 (1588): Command failed: Not found
Mon Oct  9 21:48:01 2023 daemon.notice netifd: radio0 (1588): sh: out of range
Mon Oct  9 21:48:03 2023 daemon.notice netifd: Network device 'lan1' link is up
Mon Oct  9 21:48:03 2023 daemon.notice netifd: bridge 'br-lan' link is up
Mon Oct  9 21:48:03 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Command failed: Request timed out
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Command failed: Not found
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Interface  setup failed: WPA_SUPPLICANT_FAILED
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Failed to parse json data: unexpected end of data
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Command failed: ubus call network.wireless notify { "command": 2, "device": "radio0", "data": { "pid": 0, "exe": "\/usr\/sbin\/wpad", "required": true, "keep": true } } (Invalid argument)
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Usage: ubus [<options>] <command> [arguments...]
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Options:
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  -s <socket>:             Set the unix domain socket to connect to
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  -t <timeout>:            Set the timeout (in seconds) for a command to complete
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  -S:                      Use simplified output (for scripts)
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  -v:                      More verbose output
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  -m <type>:               (for monitor): include a specific message type
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):                   (can be used more than once)
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  -M <r|t>         (for monitor): only capture received or transmitted traffic
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Commands:
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  - list [<path>]                  List objects
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  - call <path> <method> [<message>]       Call an object method
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  - subscribe <path> [<path>...]   Subscribe to object(s) notifications
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  - listen [<path>...]                     Listen for events
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  - send <type> [<message>]                Send an event
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  - wait_for <object> [<object>...]        Wait for multiple objects to appear on ubus
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):  - monitor                                Monitor ubus traffic
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588):
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Command failed: Not found
Mon Oct  9 21:48:31 2023 daemon.notice netifd: radio0 (1588): Command failed: Not found
Mon Oct  9 21:48:31 2023 daemon.notice netifd: Wireless device 'radio0' is now up