openwrt/openwrt #13493: The CPU has 8 core network cards and 8 queues. Each network card queue is bound to a CPU core. However, for some unknown reason, interruptions are concentrated in the eth3-fp-0 eth2-TxRx-6 queue bound to CPU core 6, resulting in cpu 6 core load is very high. eth3 is the wan port and eth2 is the lan port.

dongc123, 2023-09-20:
### Describe the bug

The CPU has 8 core network cards and 8 queues. Each network card queue is bound to a CPU core. However, for some unknown reason, interruptions are concentrated in the eth3-fp-0 eth2-TxRx-6 queue bound to CPU core 6, resulting in cpu 6 core load is very high. eth3 is the wan port and eth2 is the lan port.
`           CPU0       CPU1       CPU2       CPU3       CPU4       CPU5       CPU6       CPU7       
  0:          2          0          0          0          0          0          0          0   IO-APIC   2-edge      timer
  1:          9          0          0          0          0          0          0          0   IO-APIC   1-edge      i8042
  8:          0          1          0          0          0          0          0          0   IO-APIC   8-edge      rtc0
  9:          0          0          0          0          0          0          0          0   IO-APIC   9-fasteoi   acpi
 12:          0          0          0          0          0          0          0          5   IO-APIC  12-edge      i8042
 14:          0          0          0          0          0          0          0          0   IO-APIC  14-edge      ata_piix
 15:          0          0          0          0          0          0          0          0   IO-APIC  15-edge      ata_piix
 24:          0          0          0          0          0          0          0          0   PCI-MSI 129024-edge      vmw_vmci
 25:          0          0          0          0          0          0          0          0   PCI-MSI 129025-edge      vmw_vmci
 26:          0          0          0          0          0          0         21          0   PCI-MSI 1064960-edge      nvme0q0
 27:        195          0          0          0          0          0          0          0   PCI-MSI 1064961-edge      nvme0q1
 28:          0        154          0          0          0          0          0          0   PCI-MSI 1064962-edge      nvme0q2
 29:          0          0        152          0          0          0          0          0   PCI-MSI 1064963-edge      nvme0q3
 30:          0          0          0        167          0          0          0          0   PCI-MSI 1064964-edge      nvme0q4
 31:          0          0          0          0        168          0          0          0   PCI-MSI 1064965-edge      nvme0q5
 32:          0          0          0          0          0        269          0          0   PCI-MSI 1064966-edge      nvme0q6
 33:          0          0          0          0          0          0        199          0   PCI-MSI 1064967-edge      nvme0q7
 34:          0          0          0          0          0          0          0        182   PCI-MSI 1064968-edge      nvme0q8
 35:          0          0       2315          0          0          0          0          0   PCI-MSI 1048576-edge      eth0-rxtx-0
 36:          0          0          0       1457          0          0        112          0   PCI-MSI 1048577-edge      eth0-rxtx-1
 37:          0          0          0          0       1149          0          0         24   PCI-MSI 1048578-edge      eth0-rxtx-2
 38:         74          0          0          0          0        829        123          0   PCI-MSI 1048579-edge      eth0-rxtx-3
 39:          0        136          0          0          0          0       1143        362   PCI-MSI 1048580-edge      eth0-rxtx-4
 40:        376          0        146          0          0          0          0       2064   PCI-MSI 1048581-edge      eth0-rxtx-5
 41:       1137        124          0         58          0          0          0          0   PCI-MSI 1048582-edge      eth0-rxtx-6
 42:          0       1523        539          0        194          0          0          0   PCI-MSI 1048583-edge      eth0-rxtx-7
 43:          0          0          0          0          0          0          0          0   PCI-MSI 1048584-edge      eth0-event-8
 44:          0          0          0          3          1          0          0          0   PCI-MSI 1622016-edge      eth1
 45:          0          0          0          0     388900     200835          0          0   PCI-MSI 1622017-edge      eth1-TxRx-0
 46:          0          0          0          0          0      17027       3064          0   PCI-MSI 1622018-edge      eth1-TxRx-1
 47:          0          0          0          0          0          0      16807       3332   PCI-MSI 1622019-edge      eth1-TxRx-2
 48:       2956          0          0          0          0          0          0       8405   PCI-MSI 1622020-edge      eth1-TxRx-3
 49:       8944       3675          0          0          0          0          0          0   PCI-MSI 1622021-edge      eth1-TxRx-4
 50:          0     430073      24762          0          0          0          0          0   PCI-MSI 1622022-edge      eth1-TxRx-5
 51:          0          0       9413       3898          0          0          0          0   PCI-MSI 1622023-edge      eth1-TxRx-6
 52:          0          0          0       8136     197080          0          0          0   PCI-MSI 1622024-edge      eth1-TxRx-7
 53:    1566585          0          0          0          1          0          0          0   PCI-MSI 1605632-edge      eth2-TxRx-0
 54:          0    1710538          0          0          0          1          0          0   PCI-MSI 1605633-edge      eth2-TxRx-1
 55:          0          0    1548519          0          0          0          1          0   PCI-MSI 1605634-edge      eth2-TxRx-2
 56:          0          0          0    1494122          0          0          0          1   PCI-MSI 1605635-edge      eth2-TxRx-3
 57:          1          0          0          0    1866672          0          0          0   PCI-MSI 1605636-edge      eth2-TxRx-4
 58:          0          1          0          0          0    2194142          0          0   PCI-MSI 1605637-edge      eth2-TxRx-5
 59:          0          0          1          0          0          0   10098005          0   PCI-MSI 1605638-edge      eth2-TxRx-6
 60:          0          0          0          1          0          0          0    1592224   PCI-MSI 1605639-edge      eth2-TxRx-7
 61:          0          0          0          0          8          3          0          0   PCI-MSI 1605640-edge      eth2
 62:          0          0          0          0          0       6469          0          0   PCI-MSI 1572864-edge      eth3
 64:          0          0          0          0          0          0   10203047          0   PCI-MSI 1572866-edge      eth3-fp-0
 65:          0          0          0          0          0          0          0    2102315   PCI-MSI 1572867-edge      eth3-fp-1
 66:    2003234          0          0          0          0          0          0          0   PCI-MSI 1572868-edge      eth3-fp-2
 67:          0    2112603          0          0          0          0          0          0   PCI-MSI 1572869-edge      eth3-fp-3
 68:          0          0    2178438          0          0          0          0          0   PCI-MSI 1572870-edge      eth3-fp-4
 69:          0          0          0    1928720          0          0          0          0   PCI-MSI 1572871-edge      eth3-fp-5
 70:          0          0          0          0    2142720          0          0          0   PCI-MSI 1572872-edge      eth3-fp-6
 71:          0          0          0          0          0    2121919          0          0   PCI-MSI 1572873-edge      eth3-fp-7
NMI:          0          0          0          0          0          0          0          0   Non-maskable interrupts
LOC:     676554     684273     675933     674194     676485     673545     676728     677227   Local timer interrupts
SPU:          0          0          0          0          0          0          0          0   Spurious interrupts
PMI:          0          0          0          0          0          0          0          0   Performance monitoring interrupts
IWI:          0          0          0          0          0          0          0          0   IRQ work interrupts
RTR:          0          0          0          0          0          0          0          0   APIC ICR read retries
RES:       4686       5861       6429       5037       4255       5561       3965       3668   Rescheduling interrupts
CAL:      65682      82141      81583      90134      86307      81516      48431      74957   Function call interrupts
[... cut ...]
NPI:          0          0          0          0          0          0          0          0   Nested posted-interrupt event
PIW:          0          0          0          0          0          0          0          0   Posted-interrupt wakeup event
`
[image]

### OpenWrt version

OpenWrt 23.05.0-rc3 r23389-5deed175a5

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dongc123, 2023-09-20:
[image]

brada4, 2023-09-20:
Please attach output file
`irqbalance -o -d | tee /tmp/balance.txt`
Are all netcards same model? attach dmesg.
Also ethtool -g , -x , -c could help.

dongc123, 2023-09-20:
I have tried irqbalance dynamic scheduling interrupt and the situation is even worse. The system will generate a large number of sirqs. The wan port network card is Broadcom 57810 corresponding to eth3 and the lan port network card is Intel x540 corresponding to eth2.
Another server has a wan port intel i350 lan port intelx540, and the eight queue interrupts are balanced.
[image]

dongc123, 2023-09-20:
> Please attach output file `irqbalance -o -d | tee /tmp/balance.txt` Are all netcards same model? attach dmesg. Also ethtool -g , -x , -c could help.
I have tried irqbalance dynamic scheduling interrupt and the situation is even worse. The system will generate a large number of sirqs. The wan port network card is Broadcom 57810 corresponding to eth3 and the lan port network card is Intel x540 corresponding to eth2.Another server has a wan port intel i350 lan port intelx540, and the eight queue interrupts are balanced.
[image]

brada4, 2023-09-20:
You need to change rx hash to l4 or udp . The default does not work well if you have single gateway.

dongc123, 2023-09-20:
> 您需要将 rx hash 更改为 l4 或 udp 。如果您只有一个网关，则默认设置效果失败。

How should I modify it?

brada4, 2023-09-20:
Default is to spread load to irq-s based on MAC pair (which is fixed for all routed traffic)  or VLAN+IP, which will still remain fixed if you try benchmarks.
it is ethtool -n or -i
Also -g may be of help say clamp both cards to 4 queues so they leave space for rest.

dongc123, 2023-09-20:
> 默认情况下是根据 MA​​C 对（对于所有路由流量来说是固定的）或 VLAN+IP 将负载分散到 irq-s，如果您尝试基准测试，这仍然会保持固定。 它是 ethtool -n 或 -i 另外 -g 可能会有所帮助，例如将两张卡夹在 4 个队列中，以便它们留出休息空间。
eth3 is the wan port, eth3-fp-0 interrupt number 64 to eth3-fp-7 interrupt number 71, bind cpu 0 to cpu 7, it is found that a large number of interrupts are transferred to the queue corresponding to the cpu0 core. It seems that the system ranks first with eth3 A queue is mainly used to receive and send data, so the single core usage is high. However, my other server's lan port network card is also x540 and wan port i350. The queue interruption is very balanced. The system default configuration has not been modified
[image]

brada4, 2023-09-20:
It uses some algorithm to guess rx hash method, from human perspective unpredictable if it hits the spot.

LGA1150, 2023-09-21:
Enable RPS

LGA1150, 2023-09-21:
Do you use PPPoE by any chance?

brada4, 2023-09-21:
Irqbalance output absent?  Whats there to hide?
Minor cpu usage on core apparently forwarding traffic. No tellong its atom or epyc core... Thats default to re-send received frame on same CPU for both routing and nftables

dongc123, 2023-09-22:
> 启用RPS

网卡有rss队列功能不需要启用rps。

LGA1150, 2023-09-22:
> > 启用RPS
> > The network card has rss queue function and does not need to enable rps.

> Do you use PPPoE by any chance?