openwrt/openwrt #13503: No wifi on latest 23.05 Snapshot

ne20002, 2023-09-21:
### Describe the bug

Even though #13492 has been closed I still have this problem.

### OpenWrt version

23.05.0-rc3

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Steps to reproduce

Install image e.g. from image builder or with Attended Sysupgrade.

### Actual behaviour

Wifi is not working at all

### Expected behaviour

Wifi should work

### Additional info

I've set up BananaPi with frech install of latest firmware 23.05.0-rc3: Wifi worked. 
I then added a few additional packages (unbound, wireguard, ..) but I did not update packages list as upgradable: Wifi worked
I than used Attended SysUpgrade to build an image to flash: Wifi stopped working.

This is from syslog:
```
Thu Sep 21 20:35:16 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
Thu Sep 21 20:35:16 2023 daemon.err hostapd: Line 57: unknown configuration item 'radio_config_id'
Thu Sep 21 20:35:16 2023 daemon.err hostapd: 1 errors found in configuration file '/var/run/hostapd-phy1.conf'
Thu Sep 21 20:35:16 2023 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy1.conf
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739): Command failed: ubus call hostapd config_add {"iface":"phy1-ap0", "config":"/var/run/hostapd-phy1.conf"} (Invalid argument)
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739): Usage: ubus [<options>] <command> [arguments...]
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739): Options:
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  -s <socket>:		Set the unix domain socket to connect to
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  -S:			Use simplified output (for scripts)
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  -v:			More verbose output
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  -m <type>:		(for monitor): include a specific message type
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739): 			(can be used more than once)
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739): Commands:
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  - list [<path>]			List objects
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  - call <path> <method> [<message>]	Call an object method
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  - listen [<path>...]			Listen for events
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  - send <type> [<message>]		Send an event
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):  - monitor				Monitor ubus traffic
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739):
Thu Sep 21 20:35:16 2023 daemon.notice netifd: radio1 (7739): Device setup failed: HOSTAPD_START_FAILED
Thu Sep 21 20:35:16 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
Thu Sep 21 20:35:16 2023 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Thu Sep 21 20:35:16 2023 daemon.notice netifd: Wireless device 'radio1' is now down

```

Result: I don't get a version with wifi working from SysUpgrade.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lu-zero, 2023-09-21:
The problem exists also for others https://forum.openwrt.org/t/ax3600-hostapd-unknown-configuration-item-radio-config-id-snapshot-r23375-cdfcac6e24/169368/10

Comment the line 
```
echo "radio_config_id=${radio_md5sum}" >> $hostapd_conf_file
```

in `/lib/netifd/wireless/mac80211.sh` and you can get back the system working until you updated `kmod-cfg80211` again.

Code-TT, 2023-09-21:
Yes, but there will be still mesh not working.

pezz, 2023-09-21:
Thanks @lu-zero - I had this issue on my WRT32X as well after package updates and commenting this line has fixed it.

cyniu88, 2023-09-22:
In my case (wrt1900acs) commenting this line has NOT fixed it.

lu-zero, 2023-09-22:
```
kmod-cfg80211 - 5.15.131+6.5-1
```

does not seem to contain those lines anymore and

```
wpad-mbedtls - 2023-09-08-e5ccbfc6-3
```

Seems working fine.

ne20002, 2023-09-22:
@lu-zero 
Thank you very much, commenting the line did the trick. I'm back on rc3. :)
I believe I can close the issue as the developers are aware of the problem.

lu-zero, 2023-09-22:
I'd keep it open until a 23.05 is out w/out the problem.

solsticedhiver, 2023-09-23:
I was on 23.05-rc3 and run **auc** to get latest update so ended up on latest snapshot? and was hit by this bug.
I commented the line in `/lib/netifd/wireless/mac80211.sh` and **it fixed it**

on dlink, dap-2695-a1, ath79/generic, mips_24kc 23.05.0-rc3 r23389-5deed175a5 and this is the update that broke my config
```
Requesting package lists...
 ubus: 2022-06-15-9913aa61-1 -> 2023-06-05-f787c97b-1
 libubus-lua: 2022-06-15-9913aa61-1 -> 2023-06-05-f787c97b-1
 ucode-mod-ubus: 2023-04-03-51638672-1 -> 2023-06-06-c7d84aae-1
 ucode-mod-uci: 2023-04-03-51638672-1 -> 2023-06-06-c7d84aae-1
 netifd: 2023-06-04-ec9dba72-1 -> 2023-09-19-7a58b995-1
 ubusd: 2022-06-15-9913aa61-1 -> 2023-06-05-f787c97b-1
 libubus: 2022-06-15-9913aa61-1 -> 2023-06-05-f787c97b-1
 hostapd-utils: 2023-06-22-599d00be-1.2 -> 2023-09-08-e5ccbfc6-3
 ucode-mod-math: 2023-04-03-51638672-1 -> 2023-06-06-c7d84aae-1
 ucode-mod-fs: 2023-04-03-51638672-1 -> 2023-06-06-c7d84aae-1
 ucode: 2023-04-03-51638672-1 -> 2023-06-06-c7d84aae-1
 wpad-mbedtls: 2023-06-22-599d00be-1.2 -> 2023-09-08-e5ccbfc6-3
 libucode: 2023-04-03-51638672-1 -> 2023-06-06-c7d84aae-1
 hostapd-common: 2023-06-22-599d00be-1.2 -> 2023-09-08-e5ccbfc6-3
```

predators46, 2023-09-25:
Has the package been updated to openwrt 23.05.0-rc3 for this problem?

timur-davletshin, 2023-09-25:
OpenWrt 23.05-SNAPSHOT, r23467-9349302bca - just tested, works without that one line removal patch.

oopen, 2023-09-25:
> Has the package been updated to openwrt 23.05.0-rc3 for this problem?
> 
> Looks like the package has been updated

no the package is not updated on 23.05.0-rc3 
upgrade cause wifi lost
```
root@OpenWrt:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="23.05.0-rc3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.0-rc3"
VERSION_ID="23.05.0-rc3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r23389-5deed175a5"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3-d16"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.0-rc3 r23389-5deed175a5"

root@OpenWrt:~# opkg list-upgradable
netifd - 2023-06-04-ec9dba72-1 - 2023-09-19-7a58b995-1

```

predators46, 2023-09-25:
@nbd168 
please provide update wifi stuck on openwrt 23.05.0-rc3

erayrafet, 2023-09-26:
I also have no wifi working with a fresh build of RC3 from the Firmware Selector on my Archer C50 V4.

`Fri Sep 22 07:41:58 2023 daemon.notice netifd: Interface 'wan' is now up
Fri Sep 22 07:42:02 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wifi4) --> new PHY
Fri Sep 22 07:42:02 2023 daemon.err hostapd: Line 20: unknown configuration item 'radio_config_id'
Fri Sep 22 07:42:02 2023 daemon.err hostapd: FT (IEEE 802.11r) requires nas_identifier to be configured as a 1..48 octet string
Fri Sep 22 07:42:02 2023 daemon.err hostapd: 2 errors found in configuration file '/var/run/hostapd-phy0.conf'
Fri Sep 22 07:42:02 2023 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025): Command failed: ubus call hostapd config_add {"iface":"wifi4", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025): Usage: ubus [<options>] <command> [arguments...]
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025): Options:
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  -s <socket>:		Set the unix domain socket to connect to
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  -S:			Use simplified output (for scripts)
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  -v:			More verbose output
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  -m <type>:		(for monitor): include a specific message type
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025): 			(can be used more than once)
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025): Commands:
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  - list [<path>]			List objects
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  - call <path> <method> [<message>]	Call an object method
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  - listen [<path>...]			Listen for events
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  - send <type> [<message>]		Send an event
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):  - monitor				Monitor ubus traffic
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025):
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio0 (2025): Device setup failed: HOSTAPD_START_FAILED
Fri Sep 22 07:42:02 2023 daemon.notice netifd: Wireless device 'radio0' set retry=0
Fri Sep 22 07:42:02 2023 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Fri Sep 22 07:42:02 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wifi5) --> new PHY
Fri Sep 22 07:42:02 2023 daemon.err hostapd: Line 23: unknown configuration item 'radio_config_id'
Fri Sep 22 07:42:02 2023 daemon.err hostapd: FT (IEEE 802.11r) requires nas_identifier to be configured as a 1..48 octet string
Fri Sep 22 07:42:02 2023 daemon.err hostapd: 2 errors found in configuration file '/var/run/hostapd-phy1.conf'
Fri Sep 22 07:42:02 2023 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy1.conf
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026): Command failed: ubus call hostapd config_add {"iface":"wifi5", "config":"/var/run/hostapd-phy1.conf"} (Invalid argument)
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026): Usage: ubus [<options>] <command> [arguments...]
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026): Options:
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  -s <socket>:		Set the unix domain socket to connect to
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  -S:			Use simplified output (for scripts)
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  -v:			More verbose output
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  -m <type>:		(for monitor): include a specific message type
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026): 			(can be used more than once)
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026): Commands:
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  - list [<path>]			List objects
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  - call <path> <method> [<message>]	Call an object method
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):  - listen [<path>...]			Listen for events
[... cut ...]
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026):
Fri Sep 22 07:42:02 2023 daemon.notice netifd: radio1 (2026): Device setup failed: HOSTAPD_START_FAILED
Fri Sep 22 07:42:02 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
Fri Sep 22 07:42:02 2023 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Fri Sep 22 07:42:03 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Fri Sep 22 07:42:03 2023 daemon.notice netifd: Wireless device 'radio1' is now`

Commenting the line above didn't make a change.

Loafter, 2023-09-26:
Tue Sep 26 11:19:17 2023 daemon.err hostapd: Line 6: Invalid country_code '00'
Looks like you must set country code also, in my case it resolve issues

szero, 2023-09-26:
@erayrafet Looks like you are using Fast Transition, please check my separate bug report and check if the fix works for you. https://github.com/openwrt/openwrt/issues/13546

erayrafet, 2023-09-26:
> @erayrafet Looks like you are using Fast Transition, please check my separate bug report and check if the fix works for you. #13546

Sorry. I don't know how to change the NAS id.

erayrafet, 2023-09-26:
I actually went ahead and restored the working RC3 build I had from last week.

erayrafet, 2023-09-26:
But I might give this a try if it's important. Please, do let me know.

szero, 2023-09-26:
> > @erayrafet Looks like you are using Fast Transition, please check my separate bug report and check if the fix works for you. #13546
> 
> Sorry. I don't know how to change the NAS id.

Go to the "WLAN Roaming" tab of all your wireless networks and there should be NAS ID option there, set it to something like `112233445566` (set it to something different for each network) and it should work after hitting Save and Apply

erayrafet, 2023-09-27:
> Go to the "WLAN Roaming" tab of all your wireless networks and there should be NAS ID option there, set it to something like `112233445566` (set it to something different for each network) and it [...]

That error exists only with the buggy RC3 build.

predators46, 2023-09-29:
https://downloads.openwrt.org/releases/23.05.0-rc4/

oopen, 2023-09-30:
🆗 Issue fixed with [23.05.0-rc4](https://downloads.openwrt.org/releases/23.05.0-rc4/) 

(on my WRT3200ACM)

erayrafet, 2023-09-30:
I can confirm.

closed 2023-09-30 by nbd168

ne20002, 2023-09-30:
Yes, it's fixed with rc4