openwrt/openwrt #13505: mpc85xx: Watchguard T10 commit description bricks device and support for T10-W

Shine-, 2023-09-22:
### Describe the bug

I already commented this [in the commit itself](https://github.com/openwrt/openwrt/commit/35f6d795134e9b089c4e763a7f58cba7d4e15e42#commitcomment-120607714), but this may have gone unnoticed. So I'll also open an issue here, to prevent others from bricking their unit, who perhaps haven't kept a NOR dump for recovery.

First, I tried to follow @blocktrron's commit description to install the T10 image on a T10-W (which is nothing but a T10 with the Mini-PCIe slot populated by an AR9582 card).

At step 7 (increase loadable kernel size) in the installation instructions, `fw_setenv` threw a checksum error and cleared the whole Uboot environment, rendering the device unbootable. I had to recover it using the NOR flash dump.

I suggest using `setenv` / `saveenv` within Uboot, instead of `fw_setenv` within OpenWrt.

Second, as mentioned above, the T10-W is nothing but a T10 with an ath9 based card in its PCIe slot (`pci1: pcie@ffe0a000`). Since the T10 image has all required packages and drivers already present, it's sufficient to enable this PCIe slot in the DTS for the T10 image, to fully support the T10-W as well. No harm done to the T10 by enabling this PCIe port.
```
--- a/target/linux/mpc85xx/files/arch/powerpc/boot/dts/firebox-t10.dts
+++ b/target/linux/mpc85xx/files/arch/powerpc/boot/dts/firebox-t10.dts
@@ -183,7 +183,18 @@
 	};
 
 	pci1: pcie@ffe0a000 {
-		status = "disabled";
+		reg = <0 0xffe0a000 0 0x1000>;
+		ranges = <0x2000000 0x0 0x80000000 0 0x80000000 0x0 0x20000000
+			  0x1000000 0x0 0x00000000 0 0xffc00000 0x0 0x10000>;
+		pcie@0 {
+			ranges = <0x2000000 0x0 0x80000000
+				  0x2000000 0x0 0x80000000
+				  0x0 0x20000000
+
+				  0x1000000 0x0 0x0
+				  0x1000000 0x0 0x0
+				  0x0 0x100000>;
+		};
 	};
 
 	ifc: ifc@ffe1e000 {
-- 
```
Note: This is copied from similar devices' DTS, such as the TL-WDR4900 or the Sophos Red, not extracted from the vendor DTS.

I suggest enabling this PCIe port in the T10 DTS and renaming the device to something like "T10 / T10-W" or "T10(-W)".

### OpenWrt version

22.03 or later

### OpenWrt target/subtarget

mpc85xx/p1010

### Device

Watchguard Firebox T10-W

### Image kind

Official downloaded image

### Steps to reproduce

1. Follow commit 35f6d795134e9b089c4e763a7f58cba7d4e15e42 description to install OpenWrt on a Watchguard T10 or T10-W. When trying to use `fw_setenv` to increase kernel size within OpwnWrt initramfs image, it will throw a checksum error and clear the complete Uboot env. At least this is what happened to my (previously untampered) T10-W.
2. Recover and install the T10 image on a T10-W.

### Actual behaviour

1. After `fw_setenv` errors out and clears the Uboot env, the device is unbootable and has to be recovered, e.g. by using a SPI programmer to flash a NOR dump that contains the proper Uboot env.
2. WiFi is not detected and not working on the T10-W

### Expected behaviour

1. Installation instructions for OpenWrt should be updated to not brick the unit.
2. The T10 image should also fully work with T10-W, since the hardware is identical and all required WiFi packages and drivers are already present anyway.

### Additional info

n/a

### Diffconfig

```text
n/a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Hurricos, 2023-09-25:
Seems more like a bug in fw_setenv or our u-boot config for this board. I'm fairly sure the u-boot-env package knows how to manage this checksum.

>  it's sufficient to enable this PCIe slot in the T10 DTS, to fully support the T10-W as well. No harm done to the non-WiFi-equipped T10 version by enabling this PCIe port.

ACK. This is how the MX60 vs MX60W is handled.

Shine-, 2023-10-07:
> Seems more like a bug in fw_setenv or our u-boot config for this board. I'm fairly sure the u-boot-env package knows how to manage this checksum.

More or lesss... Looks like it's a mixture of both.

DTS describes uboot offset `0x90000` to `0x9ffff` as uboot-env partition. On a stock, untampered uboot, however, this space is **empty** (filled with `0xff`). A copy of the actual factory default uboot env is (on my device) stored at uboot offset `0x62fd0` to `0x639ff`. Once you use `setenv` / `saveenv` in uboot, the factory default env will be copied to offset `0x90000` and a proper checksum calculated. Only from then, it's usable from OpenWrt with `fw_setenv`.

So it looks like @blocktrron did play around with the env within uboot, ie. he didn't work off a factory default uboot dump. Therefore, the partition at offset `0x90000` was already filled with the proper uboot env.

**Conclusion:** OpenWrt's DTS config for the NOR layout is in fact correct. But **only after** you used `saveenv` at least once in uboot. So you **have to** change the SysAKernSize variable **from within uboot**, not after booting OpenWrt initramfs.

Installation instructions should definitely be corrected/updated to reflect that information.

Side note: After performing a vendor firmware recovery, any changed uboot env is also cleared (only env, not the uboot password). So if you ever went back to stock firmware by the vendor's method, and want to install OpenWrt again, you will need to set SysAKernSize again **from within uboot** to get the uboot-env partition populated properly in the NOR. Otherwise, any use of `fw_setenv` within OpenWrt will soft-brick your T10 again (vendor firmware recovery or NOR reflash required).