openwrt/openwrt #13506: MT7688 (MediaTek LinkIt Smart 7688) watchdog does trigger the reset but the board crashs/does not reboot (OpenWRT 22.03.5 and Openwrt-23.05)

biboc, 2023-09-22:
### Describe the bug

Hi,
I'm on OpenWrt 22.03.5, MT7688 (MediaTek LinkIt Smart 7688) **with 32MB flash** (instead of 16MB) so I added broken-flash-reset; in flash@0 in .dts

I want to control manually the watchdog, I do
```
ubus call system watchdog '{"magicclose": true}'
ubus call system watchdog '{"stop": true}'
ubus call system watchdog
{
        "status": "offline",
        "timeout": 30,
        "frequency": 5,
        "magicclose": true
}
echo 1 > /dev/watchdog

```
I wait until WDT reset the board, after 30 sec I lose the connection/serial and nothing happens, board gets stuck 

By the way, it works on OpenWRT 19 and it does **not** work on OpenWrt 23 either

root@OpenWrt:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.5 r20134-5f15225c1e'
DISTRIB_TAINTS='no-all'
```

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

MediaTek LinkIt Smart 7688

### Image kind

Self-built image

### Steps to reproduce

```
git clone https://github.com/openwrt/openwrt && cd openwrt
git checkout tags/v22.03.5 -b v22.03.5
make menuconfig
# Select target (MediaTek LinkIt Smart 7688)
# Change password
# Edit .dts with broken-flash-reset; in flash@0

ssh [BOARD]
ubus call system watchdog '{"magicclose": true}'
ubus call system watchdog '{"stop": true}'
echo 1 > /dev/watchdog
# Wait reboot
```

### Actual behaviour

The wdt seems to trigger a reset because I lose connection/serial but the board does not reboot

### Expected behaviour

The board should reboot

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt76x8=y
CONFIG_TARGET_ramips_mt76x8_DEVICE_mediatek_linkit-smart-7688=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

biboc, 2023-09-22:
@DragonBluep You solved https://github.com/openwrt/openwrt/issues/13437 and it seems related
Any ideas?

DragonBluep, 2023-09-22:
> @DragonBluep You solved #13437 and it seems related Any ideas?

You need to add a hack to reset spi flash on watchdog driver.

biboc, 2023-09-22:
But it is a hardware watchdog so it does the reset by itself and I don't see an early warning irq available on this whatchdog
How would you do it?

Isn't it possible to come back to how spi flash was managed before? Before the needed hack (broken-flash-reset;) ?

DragonBluep, 2023-09-22:
> Isn't it possible to come back to how spi flash was managed before? Before the needed hack (broken-flash-reset;) ?

May you can have a look on these patches.
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ramips/patches-4.14/0053-mtd-spi-nor-add-w25q256-3b-mode-switch.patch;h=0da49b0ff20d9a65c25937c25e1f1429c466b8e8;hb=refs/heads/openwrt-19.07

https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ramips/patches-4.14/0054-mtd-spi-nor-w25q256-respect-default-mode.patch;h=419a9719b5b49a6f161cbaee0a22a3282f5a66e0;hb=refs/heads/openwrt-19.07

biboc, 2023-09-22:
I see, it is what I need I think
Because my flash needs 3B read mode to correctly boot, I need it to always be in this mode except if it wants to read above 16MB in flash 
The patches you mentioned (this one: https://github.com/openwrt/openwrt/blob/openwrt-19.07/target/linux/ramips/patches-4.14/0053-mtd-spi-nor-add-w25q256-3b-mode-switch.patch) is not anymore in OpenWrt new release? I don't find it and I don't find the applied patch in linux

Moreover, if I apply this patch, I should be able to remove broken-flash-reset;

Did this patch abandoned? If it exists, should I set something in .dts to take it into account?

Thanks

biboc, 2023-09-22:
Someone added a patch in kernel 5.4 but it has also been removed, I don't understand why
https://patchwork.ozlabs.org/project/openwrt/patch/20200406114428.1189632-1-gch981213@gmail.com/#2403617

https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg61908.html

biboc, 2023-09-22:
In fact, this is not a problem of OpenWrt but in linux spi driver for winbond w25q256 flash
Explained here: https://patchwork.ozlabs.org/project/linux-mtd/patch/874lkmw54j.fsf@notabene.neil.brown.name/

> i.e. the problem isn't 4-byte mode exactly.  The problem is the Extended
> Address Register being set implicitly, and not being zero at reboot.
> It looks like we need to clear the extended address register before
> reboot, either by:
> - software-reset the flash at shutdown
> - write '0' in the shutdown handler
> - write '0' after every transfer (or every transfer beyond 16M).

Because I don't use 16MB to 32MB of my flash, I can work with 3Bytes address
For those who wants to use until 32MB, IMO, there is no solution 100% reboot safe. Even if we set extended address register after each transfer, there is always some ms where reboot is not safe

I close the issue

Thanks @DragonBluep for your help

closed 2023-09-22 by biboc