openwrt/openwrt #13516: Dynamic IPv6 prefix, old prefix not deprecated (by Apple clients, MacOS & iOS)

ahuse, 2023-09-23:
### Describe the bug

Hi,
like described in [this thread](https://forum.openwrt.org/t/dynamic-ipv6-prefix-old-prefix-not-deprecated-by-some-clients/163081) in the forums my ISP forces me to reconnect the wan interface every 24 hours and with the reconnect i get not only a new IPv4 address, but also a new IPv6 prefix. Odhcp announces the change of prefix correctly and if the apple devices are used (not asleep) during prefix change, every thing works as expected.

But if the devices are asleep (during scheduled reconnect at night) the iOS and MacOS devices get confused and accumulate new IPv6 addresses via SLAAC (not deleting old ones), but IPv6 connectivity is often lost.

I tried to optimise nearly every different setting in odhcp, but had no luck so far. Can someone help me debug this from the openwrt side or should i file a bug report with apple?

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

iOS or MacOS device in standby mode (for a longer time). Change IPv6 prefix and wake device after a few hours.

### Actual behaviour

Devices are accumulating IPv6 addresses via SLAAC and not getting new addresses via DHCPv6. IPv6 connectivity is often lost.

### Expected behaviour

Devices should get new IPv6 addresses via SLAAC and DHCPv6 at latest a few minutes after wake from sleep and deprecate old addresses after a few hours.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-23:
Can you try filtering ipv6 acquisition into pcap:
`tcpdump -p -i br-lan -w /tmp/sleepy.pcap -c 1000 'ether mac 10:20:30:40:50'`
Dont post in whole, just try to highlight the differences between slumber and awake modes first yourself and save out just limited stream if there is any difference.
Obviously if there is no difference on the wire/wifi openwrt is of no help and problem is with client device treating same event differently.
(1000 packets is less than 2 megs, even weakest router can hold that till morning)

ahuse, 2023-09-23:
Can you clarifiy? Sorry, i am no expert in tcpdumb
Should i put in the mac address of my device? host, src or dst?

If i use `tcpdump -p -i br-lan -w /tmp/sleepy.pcap -c 1000 'ether host 10:20:30:40:50'` with MacBook mac address it captures 1000 packets in a few seconds.

brada4, 2023-09-23:
Of the other device, the iphone etc.

dave14305, 2023-09-23:
You can restrict to RAs and DHCPv6 with:
```
tcpdump -n -v '(icmp6 and ip6[40] == 134) or (udp and port 547)' and 'ether host 10:20:30:40:50' -i br-lan 
```

brada4, 2023-09-24:
It was late then, write the files and then look at them with wireshark side by side. Using dave's expression for more exact filter.

ahuse, 2023-10-09:
Thanks for your help and sorry for the late feedback. I did some further testing and found a configuration that seems like it is working for me. Both MacOS and iOS just keep the old IPv6 SLAAC addresses longer than they should. But with ìfconfig en0` i was able to check that at least MacOS deprecates the old IPv6 addresses immediately. One thing i noticed is that i have to restart my MacBook (or at least turn off and on WiFi) after a router restart to get it working properly. 

Excerpt of /etc/config/dhcp
```
config dhcp 'lan'
	option interface 'lan'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	option ra_mininterval '30'
	option ra_maxinterval '100'
	option ra_useleasetime '1'
```

closed 2023-10-09 by ahuse