openwrt/openwrt #13526: ASUS RT-AC88U 802.11r Fast Transition breaks down WiFi Connection, with WPA-PSK2 and/or WPA3 SAE

theyo-tester, 2023-09-24:
### Describe the bug

Wifi connection fails as soon as 802.11r Fast Transition is enabled and WPA-PSK2 is used.  This happens with both drivers Generic MAC80211 802.11b/g/n & Generic MAC80211 802.11ac/n

Later I discovered that with WPA PSK the connection with mobile devices gets established, but I did not do further checks, because  WPA PSK  is insecure anyway, so its not a viable option.

Tried without success:
- specified NAS ID manually
- specified Mobility Domain manually
- deactivated WMM Mode
- changed Reassociation Deadline
- deactivated Disassociate On Low Acknowledgement
- changed DTIM Interval
- disabled Short Preamble
- enabled Coverage cell density 
- enabled/disabled Distance Optimization
- changed Fragmentation Threshold to something customized
- changed Beacon Interval to something customized
- rebooting device after enabled 802.11r
- 802.11w Management Frame Protection enabled & disabled, **BTW: This option enabled (required or optional)  breaks wifi connectivity even when 802.11r is disabled** Maybe another issue should be opened for this
- With and without WPA3 SAE, **BTW: This option  breaks wifi connectivity even when 802.11r is disabled**  Maybe another issue should be opened for this
- with manually selected wifi channel. BTW: With this snapshot release (r23995-ce7209bd21) wifi only works when the channel is manually selected, Auto option breaks the wifi connectivity alltogether, the SSID doesn't get broadcasted anymore and the transmit power drops to 0dbm. With Previous stable release (21, as with 22 there is a boot loop) this was working.   Maybe another issue should be opened for this.

### OpenWrt version

r23995-ce7209bd21 or any other functionial version for this device

### OpenWrt target/subtarget

bcm53xx/generic

### Device

ASUS RT-AC88U

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-24:
Permission denied looks like ujail?

theyo-tester, 2023-09-25:
I don't know what ujail does in this regard, but I'm not sure if that's the root cause.

brada4, 2023-10-02:
Disable ujail in init script....

theyo-tester, 2023-10-02:
Thanks for the answer. 
You mean that I change the wpad init script in init.d so that it doesn't use ujail anymore?
Would it also disable ujail if I unsinstall the package procd-ujail ?

brada4, 2023-10-02:
I think its just commenting a line in wpad script.

theyo-tester, 2023-10-09:
I have tested also with removing procd-ujail. The permission denied error went away. BUT the wifi with enabled fast roaming is still not usable. Like before: SSID gets broadcasted but no client can connect to it

brada4, 2023-10-09:
Is it bridged to br-lan? I.e dhcp and internet

raenye, 2023-10-11:
Did 802.11r ever work for you? `brcmfmac` is quite picky.
See here: https://forum.openwrt.org/t/netgear-r8000-support-for-802-11r/95675