openwrt/openwrt #13528: Unable to install build dependency distutils

huaracheguarache, 2023-09-24:
### Describe the bug

I'm currently running the Fedora 39 Silverblue beta, and when I try to use the build system I'm unable to satisfy the Python3 distutils dependency. When I run make menuconfig I get the following error:

```
Build dependency: Please install the Python3 distutils module

Prerequisite check failed. Use FORCE=1 to override.
make: *** [/var/home/michael/src/router_stuff/openwrt/include/toplevel.mk:182: /var/home/michael/src/router_stuff/openwrt/staging_dir/host/.prereq-build] Error 1
```

I'm unable to install distutils because it has been deprecated, and in Python 3.12 it has been removed (which Fedora 39 uses): https://docs.python.org/3/whatsnew/3.10.html#distutils-deprecated

### OpenWrt version

75ee5546e9b7cfa5bbfd6f844ab8c5fffd5bb594

### OpenWrt target/subtarget

NA

### Device

NA

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-24:
Try installing setuptools dnf/yum package and then manually try to `import distutils` .

huaracheguarache, 2023-09-24:
Thanks, that worked! The build system passed the dependency checks and I'm currently trying to make a new build.

I don't know what the right way to go about this issue is. Should the text about which module to install possibly be changed to include setuptools?

brada4, 2023-09-24:
It is in wiki here, install setuptools and buildutils both, missing packages dont make error on install.
https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem#centosfedora

brada4, 2023-09-24:
(it actually does not mention builtin distutils which are now in external setuptools)