openwrt/openwrt #13535: Build openwrt for Xiaomi Mini MT7620 error

InBomBi, 2023-09-25:
### Describe the bug

When i build openwrt for Xiaomi Mini, MT7620, i have got this erro.
How do i know exactly what error is?
Thanks

`mt7620/linux-5.15.132/include/ -undef -D__DTS__  -o /home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_planex_cs-qr10.dtb.tmp ../dts/mt7620a_planex_cs-qr10.dts
rm -f /home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_phicomm_psg1208.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.132/arch/mips/boot/dts -I/home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.132/arch/mips/boot/dts/include -I/home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.132/include/ -undef -D__DTS__  -o /home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_planex_db-wrt01.dtb.tmp ../dts/mt7620a_planex_db-wrt01.dts
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.132/arch/mips/boot/dts -I/home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.132/arch/mips/boot/dts/include -I/home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.132/include/ -undef -D__DTS__  -o /home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_planex_mzk-750dhp.dtb.tmp ../dts/mt7620a_planex_mzk-750dhp.dts
../dts/mt7620a_phicomm_psg1218a.dts:1:10: fatal error: mt7620a_phicomm_psg1218.dtsi: No such file or directory
    1 | #include "mt7620a_phicomm_psg1218.dtsi"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [Makefile:234: /home/vovin/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_phicomm_psg1218a.dtb] Error 1
make[5]: *** Waiting for unfinished jobs....`

### OpenWrt version

oepnwrt 23.05

### OpenWrt target/subtarget

MT7620

### Device

Xiaomi Miwifi Mini 

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7620=y
CONFIG_TARGET_ramips_mt7620_DEVICE_xiaomi_miwifi-mini=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
CONFIG_PACKAGE_batctl-full=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_kmod-batman-adv=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-batman-adv=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_luci-theme-openwrt-2020=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_uboot-envtools=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wpad=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_WPA_MBO_SUPPORT=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

LGA1150, 2023-09-25:
Is your repo clean? There is no `mt7620a_phicomm_psg1218a.dts` in the 23.05 branch.

InBomBi, 2023-09-25:
Yes, i knew. I will clean and remake again to check.

LGA1150, 2023-09-26:
Please check your `git status`.

closed 2023-10-18 by InBomBi