openwrt/openwrt #13536: TP-Link Archer C6 v3: hostapd: DFS start_dfs_cac() failed, -1 // 5G Wifi not working on DFS Channels

micw, 2023-09-25:
### Describe the bug

On TP-Link Archer C6 v3 Wifi seems not to work on any 5G DFS channel. I have 4 SSIDs configured. On start it fails with DFS setup. All other SSIDs are not configured.

### OpenWrt version

OpenWrt 22.03.5, r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6 v3

### Image kind

Official downloaded image

### Steps to reproduce

Setup wifi to a DFS channel (e.g. Country Code DE Channel 100)
Start Wifi 5G (`wifi down radio1; sleep 10; wifi up radio1`)

### Actual behaviour

Wifi is started but fails to init DFS. Only the first SSID is confdigured.

Log:

Mon Sep 25 09:30:32 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wifi-lan-5g) --> new PHY
Mon Sep 25 09:30:32 2023 kern.info kernel: [316584.485700] br-lan: port 10(wifi-lan-5g) entered blocking state
Mon Sep 25 09:30:32 2023 kern.info kernel: [316584.491767] br-lan: port 10(wifi-lan-5g) entered disabled state
Mon Sep 25 09:30:32 2023 kern.info kernel: [316584.498624] device wifi-lan-5g entered promiscuous mode
Mon Sep 25 09:30:32 2023 daemon.notice hostapd: wifi-lan-5g: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Sep 25 09:30:32 2023 daemon.notice hostapd: wifi-lan-5g: interface state COUNTRY_UPDATE->HT_SCAN
Mon Sep 25 09:30:33 2023 daemon.notice hostapd: wifi-lan-5g: interface state HT_SCAN->DFS
Mon Sep 25 09:30:33 2023 daemon.notice hostapd: wifi-lan-5g: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
Mon Sep 25 09:30:33 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Mon Sep 25 09:30:33 2023 daemon.err hostapd: Interface initialization failed
Mon Sep 25 09:30:33 2023 daemon.notice hostapd: wifi-lan-5g: interface state DFS->DISABLED
Mon Sep 25 09:30:33 2023 daemon.notice hostapd: wifi-lan-5g: AP-DISABLED
Mon Sep 25 09:30:34 2023 daemon.notice netifd: Wireless device 'radio1' is now up

### Expected behaviour

1. DFS is set up properly
2. All SSIDs are configured

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-25:
You need to provide fallback channels like
```
option chanmel '100'
option chamnels '36 100 149'
```
The 'issue' you complain about is called legislation. What you get is actually 5GHz working correctly.

micw, 2023-09-25:
I don't think so. Expected behaviour with a single channel would be to pause the channel for a defined time, then try again. Also it should bring up the other SSIDs.

I did another test with option channels '100 124 136'. It behaves exactly the same.

micw, 2023-09-25:
> The message is that radar beeped while initial channel clearance was done

Where exactly do you see that message?

I tried with VHT40 and `option channels '100 108 116 124 132 140'` - same result. I also tried every single channel, it's always exact the same behaviour.

brada4, 2023-09-25:
Does it still happen if you set channel to auto? (With channels limited to DFS only)

brada4, 2023-09-25:
try
channel 100
 channels 36-173
does it ever land on a DFS channel or it always escapes to 36?

Shine-, 2023-09-25:
@micw The mt76 driver doesn't currently support DFS with MT7613/MT7663 radios (your device has MT7613BEN for 5GHz). Also see https://openwrt.org/unsupported/dfs (that page is lacking a comprehensive list of affected devices, though).

brada4, 2023-09-25:
You essentially confirm that device is locked to 36-40-44-48 while in the EU ..... (to exclude superweak 149+)

micw, 2023-09-25:
That matches my observations, right. Too bad, it's oherwise an excelent, cheap device.

@Shine- I wonder why the information "not implemented" is not exposed by the driver to it's clients so that a meaningful error message can be shown.

micw, 2023-09-25:
@Shine- How can I add more models to https://openwrt.org/unsupported/dfs ? When I edit the page, the datatable has no content where I could add someting.

Shine-, 2023-09-25:
Don't ask me, I'm only a user, just like you! :-)

There's a link at the bottom of the Wiki pages that directs you to a forum topic that explains how to apply for a Wiki account. Though imo, that page shouldn't have a separate, manually maintained table, but should instead link to a ToH view that filters for MT7613 and MT7663.

micw, 2023-09-25:
Yeah, figured it out. Just edit the data table of the device and add "DSF" to unsupported field

brada4, 2023-09-25:
Should be likely caveat heading like 4MB flash space.

micw, 2023-09-25:
How can I check the radio? My "Xiaomi Mi Router 4A Gigabit Edition" is very similar in Hardware but has working DFS.

micw, 2023-09-25:
Since that's not a bug but a missing feature, I close this issue here.

closed 2023-09-25 by micw