openwrt/openwrt #13548: bcm53xx: Linksys EA9200 ethernet is broken

raenye, 2023-09-26:
### Describe the bug

Linksys EA9200 cannot communicate on ethernet ports (lan1-4 and wan).

### OpenWrt version

a4792d79e8

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Linksys EA9200

### Image kind

Official downloaded image

### Steps to reproduce

This worked fine on 23.05-rc2 and did not work on 23.05-rc3 (and also snapshot).
I performed `git bisect` to pinpoint commit a4792d79e8 as the one introducing the problem.
Moreover, this commit includes several kernel patches; I traced the problem and verified it is because of
```
target/linux/bcm53xx/patches-5.15/036-v6.5-0012-ARM-dts-BCM5301X-Describe-switch-ports-in-the-main-D.patch
```
which moved BCM5301X register definitions from individual DTS files to `arch/arm/boot/dts/bcm-ns.dtsi`.
Temporarily deleting this file fixes the problem.

### Actual behaviour

Pings cannot be sent from/to the device (answers to DHCP requests do not arrive, so no auto-assigned IP, but I used static assigned IP to test).
`tcpdump` running on a 2nd device connected via cable shows nothing.
Two devices connected to lan1 and lan2 ports **can** communicate, so the switch is active and operating, so it seems the cpu port is somehow not connected.

### Expected behaviour

Ethernet works (DHCP, ping, etc.)

### Additional info

Since `target/linux/bcm53xx/patches-5.15/304-ARM-dts-BCM5301X-Specify-switch-ports-for-remaining-.patch` adds port definitions with registers to `arch/arm/boot/dts/bcm4709-linksys-ea9200.dts`, I tried to fix the problem, along the lines of `036-v6.5-0012-ARM-dts-BCM5301X-Describe-switch-ports-in-the-main-D.patch`, using the following patch, but it did not work.

```
diff --git a/target/linux/bcm53xx/patches-5.15/304-ARM-dts-BCM5301X-Specify-switch-ports-for-remaining-.patch b/target/linux/bcm53xx/patches-5.15/304-ARM-dts-BCM5301X-Specify-switch-ports-for-remaining-.patch
index d15506f6d7..67ce6fe113 100644
--- a/target/linux/bcm53xx/patches-5.15/304-ARM-dts-BCM5301X-Specify-switch-ports-for-remaining-.patch
+++ b/target/linux/bcm53xx/patches-5.15/304-ARM-dts-BCM5301X-Specify-switch-ports-for-remaining-.patch
@@ -395,7 +395,7 @@ Signed-off-by: Rafał Miłecki <rafal@milecki.pl>
 +};
 --- a/arch/arm/boot/dts/bcm4709-linksys-ea9200.dts
 +++ b/arch/arm/boot/dts/bcm4709-linksys-ea9200.dts
-@@ -47,3 +47,45 @@
+@@ -47,3 +47,33 @@
  &usb3_phy {
        status = "okay";
  };
@@ -405,39 +405,27 @@ Signed-off-by: Rafał Miłecki <rafal@milecki.pl>
 +
 +      ports {
 +              port@0 {
-+                      reg = <0>;
 +                      label = "lan1";
 +              };
 +
 +              port@1 {
-+                      reg = <1>;
 +                      label = "lan2";
 +              };
 +
 +              port@2 {
-+                      reg = <2>;
 +                      label = "lan3";
 +              };
 +
 +              port@3 {
-+                      reg = <3>;
 +                      label = "lan4";
 +              };
 +
 +              port@4 {
-+                      reg = <4>;
 +                      label = "wan";
 +              };
 +
 +              port@8 {
-+                      reg = <8>;
 +                      label = "cpu";
-+                      ethernet = <&gmac2>;
-+
-+                      fixed-link {
-+                              speed = <1000>;
-+                              full-duplex;
-+                      };
 +              };
 +      };
 +};
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-09-26:
@rmilecki Please advise; thank you in advance.

rmilecki, 2023-09-26:
I decompiled `bcm4709-linksys-ea9200.dtb` from the latest `openwrt-23.05` and after removing `036-v6.5-0012-ARM-dts-BCM5301X-Describe-switch-ports-in-the-main-D.patch`. The only difference are two extra switch ports that **should not** matter as they don't have any label. I believe only switch port `cpu` and its Ethernet controller should be used. In your case that's `port@8`.

Just to be 100% sure can you:
1. Try the latest `openwrt-23.05` branch and attach boot log from it
2. Try my https://git.openwrt.org/?p=openwrt/staging/rmilecki.git;a=shortlog;h=refs/heads/linksys-ea9200-23.05 and attach boot log from it

raenye, 2023-09-26:
I had saved two bootlogs (via UART) during the bisect process (neither latest openwrt-23.05 nor your tree, but I think it serves to make the point).

[Here is a diff](https://www.diffchecker.com/qeqexchR/) between them showing that you said (left is good, right is bad).
Ignoring differences related to having a USB drive connected only on the left bootlog, the main issue is whether CPU uses `eth0` (port 5) or `eth2` (port 8).

## Highlights
1. One vs. three CPU ports;
1. `eth2` vs. `eth0`.
### Good
```
[    3.745440] b53-srab-switch 18007000.ethernet-switch: found switch: BCM53012, rev 5	
[    3.868153] b53-srab-switch 18007000.ethernet-switch: Using legacy PHYLIB callbacks. Please migrate to PHYLINK!	
[    3.879353] b53-srab-switch 18007000.ethernet-switch lan1 (uninitialized): PHY [dsa-0.0:00] driver [Generic PHY] (irq=POLL)	
[    3.891883] b53-srab-switch 18007000.ethernet-switch lan2 (uninitialized): PHY [dsa-0.0:01] driver [Generic PHY] (irq=POLL)	
[    3.904424] b53-srab-switch 18007000.ethernet-switch lan3 (uninitialized): PHY [dsa-0.0:02] driver [Generic PHY] (irq=POLL)	
[    3.917019] b53-srab-switch 18007000.ethernet-switch lan4 (uninitialized): PHY [dsa-0.0:03] driver [Generic PHY] (irq=POLL)	
[    3.929576] b53-srab-switch 18007000.ethernet-switch wan (uninitialized): PHY [dsa-0.0:04] driver [Generic PHY] (irq=POLL)
...
[    7.223991] b53-srab-switch 18007000.ethernet-switch lan1: configuring for phy/gmii link mode	
[    7.232650] bgmac_bcma bcma0:5 eth2: Link is Up - 1Gbps/Full - flow control off	
[    7.240172] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
...
[   19.762632] bgmac_bcma bcma0:5 eth2: Link is Up - 1Gbps/Full - flow control off	
[   19.773660] b53-srab-switch 18007000.ethernet-switch lan1: configuring for phy/gmii link mode	
[   19.785177] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready	
[   19.793615] br-lan: port 1(lan1) entered blocking state	
[   19.798970] br-lan: port 1(lan1) entered disabled state	
[   19.804815] device lan1 entered promiscuous mode	
[   19.809522] device eth2 entered promiscuous mode
```
### Bad
```
[    3.735710] b53-srab-switch 18007000.ethernet-switch: found switch: BCM53012, rev 5
[    3.858457] b53-srab-switch 18007000.ethernet-switch: Using legacy PHYLIB callbacks. Please migrate to PHYLINK!
[    3.868660] b53-srab-switch 18007000.ethernet-switch: Using legacy PHYLIB callbacks. Please migrate to PHYLINK!
[    3.878803] b53-srab-switch 18007000.ethernet-switch: Using legacy PHYLIB callbacks. Please migrate to PHYLINK!
[    3.889985] b53-srab-switch 18007000.ethernet-switch lan1 (uninitialized): PHY [dsa-0.0:00] driver [Generic PHY] (irq=POLL)
[    3.902547] b53-srab-switch 18007000.ethernet-switch lan2 (uninitialized): PHY [dsa-0.0:01] driver [Generic PHY] (irq=POLL)
[    3.915150] b53-srab-switch 18007000.ethernet-switch lan3 (uninitialized): PHY [dsa-0.0:02] driver [Generic PHY] (irq=POLL)
[    3.927727] b53-srab-switch 18007000.ethernet-switch lan4 (uninitialized): PHY [dsa-0.0:03] driver [Generic PHY] (irq=POLL)
[    3.940278] b53-srab-switch 18007000.ethernet-switch wan (uninitialized): PHY [dsa-0.0:04] driver [Generic PHY] (irq=POLL)
...
[    7.235353] b53-srab-switch 18007000.ethernet-switch lan1: configuring for phy/gmii link mode
[    7.243972] bgmac_bcma bcma0:3 eth0: Link is Up - 1Gbps/Full - flow control off
[    7.251956] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
...
	[   19.687488] bgmac_bcma bcma0:3 eth0: Link is Up - 1Gbps/Full - flow control off
[   19.694872] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   19.703438] b53-srab-switch 18007000.ethernet-switch lan1: configuring for phy/gmii link mode
[   19.718548] br-lan: port 1(lan1) entered blocking state
[   19.723830] br-lan: port 1(lan1) entered disabled state
[   19.729855] device lan1 entered promiscuous mode
[   19.734519] device eth0 entered promiscuous mode
```

raenye, 2023-09-26:
I was not able to compile snapshot (probably broken now; complains about missing `staging_dir/target-arm_cortex-a9_musl_eabi/image/dlink_dir-890l-u-boot.bin`).
Nevertheless I can confirm that patched snapshot-23.05 (1fd72867163) works fine.

Should I open a PR? (hope the commit message is fine)
If so, should it be for snapshot-23.05, snapshot, or both?

Thanks for your help @rmilecki :)

pauliuspetk, 2023-10-12:
Netgear R8000 is also affected, no Ethernet traffic between CPU and switch from 23.05.0-rc3 to today's 23.05.0 stable release (inclusive).
23.05.0-rc2 works fine.

raenye, 2023-10-12:
Do you want me to try to adapt my EA9200 patch for R8000 too?

pauliuspetk, 2023-10-12:
Yes, please. Could it be that more bcm53xx devices are affected as well?

raenye, 2023-10-12:
> Yes, please.

Here you go. Try to copy [this file](https://github.com/openwrt/openwrt/files/12887515/930-ARM-dts-BCM5301X-Delete-Netgear-R8000-unused-switch.patch) to `target/linux/bcm53xx/patches-6.1` or to `target/linux/bcm53xx/patches-5.15` and to compile.
Be sure to enable kernel support for USB storage and for VFAT filesystem. While bisecting, I found the easiest way to reflash after a bad version is to have a good version on USB.

I assume you have UART access?

> Could it be that more bcm53xx devices are affected as well?

Quite likely, I'm not sure why they let 23.05 out without the EA9200 fix (and without testing `bcm53xx`).

rmilecki, 2023-10-13:
I sent patch for upstream that I believe to be correct fix:
[ARM: dts: BCM5301X: Explicitly disable unused switch CPU ports](https://patchwork.kernel.org/project/linux-arm-kernel/patch/20231013103314.10306-1-zajec5@gmail.com/)

You can test my branch that should fix both: Netgear R8000 and Linksys EA9200:
https://git.openwrt.org/?p=openwrt/staging/rmilecki.git;a=shortlog;h=refs/heads/bcm53xx-ports

raenye, 2023-10-15:
@rmilecki I don't have an AC87U, but this seems like a typo (`port@7` disabled instead of `port@8`?):
```
 --- a/arch/arm/boot/dts/bcm4709-asus-rt-ac87u.dts
 +++ b/arch/arm/boot/dts/bcm4709-asus-rt-ac87u.dts
-@@ -66,6 +66,32 @@
+@@ -66,6 +66,40 @@
        status = "okay";
  };

@@ -286,6 +342,14 @@ Signed-off-by: Rafał Miłecki <rafal@milecki.pl>
 +                      label = "lan3";
 +              };
 +
++              port@5 {
++                      status = "disabled";
++              };
++
++              port@7 {
++                      status = "disabled";
++              };
++
 +              port@7 {
 +                      label = "cpu";
 +              };
```

closed 2023-10-24 by openwrt-bot