openwrt/openwrt #13552: TP-Link TL-WDR4900 v1 WLAN switch button behavior is wrong

cewbdex, 2023-09-27:
### Describe the bug

This router has a WLAN switch, however, it acts like a button, not a switch (have to flip twice back and forth to change the state). I tried to look how to fix it: [here is the DTS line I think is relevant](https://github.com/openwrt/openwrt/blob/ff95f859ebf710d2914472a3feeeb0d187d14459/target/linux/mpc85xx/files/arch/powerpc/boot/dts/tl-wdr4900-v1.dts#L304), I don't know how to change it, so it acts like a switch.

### OpenWrt version

22.3.05

### OpenWrt target/subtarget

mpc85xx/p1010

### Device

TP-Link TL-WDR4900 v1

### Image kind

Official downloaded image

### Steps to reproduce

Use WLAN switch to turn off and on WLAN interfaces.

### Actual behaviour

Acts like a button -  I have to move the switch to the other side and back so it has toggles.

### Expected behaviour

I only switch the switch to on and off state and the WLAN will be on and off.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Shine-, 2023-09-27:
Probably missing a line `linux,input-type = <EV_SW>;` there, maybe also an additional `debounce-interval = <60>;` to work properly (judging from the DTS of similar TP-Link devices that have a WiFi switch instead of a button).

cewbdex, 2023-09-27:
Can I change the DTS without reflashing the firmware? The kmod would be enough to install via opkg I guess? Would like to avoid flashing (and then restoring), as this router's service is important for my network, but if no other option, I will flash it.