openwrt/openwrt #13554: Compile the firmware of the TP-Link XDR6088 using the firmware-selector. WIFI will not work

ysqspdk, 2023-09-27:
### Describe the bug

This is the IPK I set

`
base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kmod-crypto-hw-safexcel kmod-gpio-button-hotplug kmod-leds-gpio kmod-mt7915e kmod-mt7986-firmware kmod-nft-offload kmod-usb3 libc libgcc libustream-mbedtls logd mt7986-wo-firmware mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-mbedtls luci luci-compat luci-lib-base luci-lib-ipkg luci-i18n-base-zh-cn luci-i18n-firewall-zh-cn luci-i18n-opkg-zh-cn
`

After burning the firmware, WIFI did not work, but I compiled it once in 2023.9.15 with the same IPK package, and the compiled firmware WIFI worked at that time, which was very strange

[image]

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

mediatek/filogic

### Device

TP-Link TL-XDR6088

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-27:
Just use default package selection? Add luci-mod-admin-full for common web interface?

DragonBluep, 2023-09-28:
The default WiFi status is **OFF**. You need to manually **Enable** it. The image above shows it's still off.