openwrt/openwrt #13561: ipq-wifi makefile instructions are not working

protectivedad, 2023-09-28:
### Describe the bug

The makefile in `package/firmware/ipq-wifi` says:
```
# Add board name to ALLWIFIBOARDS
# Place files in this directory as board-<devicename>.<qca4019|qca9888|qca9889|qca9984|qca99x0|ipq8074>
# Add $(eval $(call generate-ipq-wifi-package,<devicename>,<display name>))
```

But the `board-<devicename>.<qca4019|qca9888|qca9889|qca9984|qca99x0|ipq8074>` filess are not being copied to the build directory. Instead they are being downloaded from `PKG_SOURCE_URL=$(PROJECT_GIT)/project/firmware/qca-wireless.git`.

### OpenWrt version

openwrt-23.05

### OpenWrt target/subtarget

ipq40xx/generic

### Device

New device whw03

### Image kind

Self-built image

### Steps to reproduce

Add device to ipq-wifi package with board-2.bin files. Compile image.

### Actual behaviour

Files are never copied into build directory. A package file is created but it has no files in it.

### Expected behaviour

The board files are copied over and added to the package.

### Diffconfig

```text
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_linksys_whw03=y
CONFIG_DEVEL=y
CONFIG_ATH10K_LEDS=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
CONFIG_DOWNLOAD_FOLDER="/home/anthony/Downloads"
CONFIG_INCLUDE_CONFIG=y
CONFIG_KERNEL_BUILD_DOMAIN="sepanet"
CONFIG_KERNEL_BUILD_USER="protectivedad"
CONFIG_PACKAGE_ath10k-firmware-qca4019=y
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct is not set
CONFIG_PACKAGE_ath10k-firmware-qca9888=y
# CONFIG_PACKAGE_ath10k-firmware-qca9888-ct is not set
CONFIG_PACKAGE_batctl-default=y
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_iperf3=y
# CONFIG_PACKAGE_jansson is not set
CONFIG_PACKAGE_kmod-ath10k=y
# CONFIG_PACKAGE_kmod-ath10k-ct is not set
CONFIG_PACKAGE_kmod-batman-adv=y
CONFIG_PACKAGE_kmod-bluetooth=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-ecdh=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-hid=y
# CONFIG_PACKAGE_kmod-hwmon-core is not set
CONFIG_PACKAGE_kmod-input-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
# CONFIG_PACKAGE_kmod-nf-conntrack is not set
# CONFIG_PACKAGE_kmod-nf-conntrack6 is not set
# CONFIG_PACKAGE_kmod-nf-flow is not set
# CONFIG_PACKAGE_kmod-nf-log is not set
# CONFIG_PACKAGE_kmod-nf-log6 is not set
# CONFIG_PACKAGE_kmod-nf-nat is not set
# CONFIG_PACKAGE_kmod-nf-reject is not set
# CONFIG_PACKAGE_kmod-nf-reject6 is not set
# CONFIG_PACKAGE_kmod-nfnetlink is not set
# CONFIG_PACKAGE_kmod-nft-core is not set
# CONFIG_PACKAGE_kmod-nft-fib is not set
# CONFIG_PACKAGE_kmod-nft-nat is not set
# CONFIG_PACKAGE_kmod-nft-offload is not set
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-regmap-core=y
CONFIG_PACKAGE_kmod-spi-dev=y
CONFIG_PACKAGE_libiperf3=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libmbedtls is not set
# CONFIG_PACKAGE_libmnl is not set
CONFIG_PACKAGE_libncurses=y
# CONFIG_PACKAGE_libnftnl is not set
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-wolfssl=y
CONFIG_PACKAGE_libwolfssl=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-advanced-reboot=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-app-watchcat=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-batman-adv=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_mtd is not set
CONFIG_PACKAGE_nano-full=y
# CONFIG_PACKAGE_nftables-json is not set
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_watchcat=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-wolfssl=y
CONFIG_WOLFSSL_HAS_NO_HW=y
# CONFIG_WOLFSSL_HAS_TLSV10 is not set
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_WPA_WOLFSSL=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

protectivedad, 2023-09-28:
The `package/firmware/ipq-wifi` was changed to pull files from a git repository. It no longer looks at the files in the makefile directory. This was done in commit f576762. What are the new instructions needed when adding a new device? I also noticed that files that firmware files exist in the repo that are no longer referenced in the makefile. After adding a new device and the board.bin files are added to the official board-2.bin file how are the temp board files cleaned up?

Ansuel, 2023-10-02:
@protectivedad thanks for pointing this out! I will take care of updating the comments...

Uploading the bin is now a 2 step process.
First make a pr in the ipq-wifi repository
Second (ideally in the device pr) the ipq-wifi package is bumped to the new version and reference to the bin is added.