openwrt/openwrt #13562: Increased memory usage potentially due to hostapd updates in Snapshot

edrikk, 2023-09-28:
### Describe the bug

Hi,

As posted in the [openwrt forum,](https://forum.openwrt.org/t/potential-memory-leak-introduced-in-snapshot-august-sept-2023/172291/18), I have found that builds after 7f54d9ba1a cause my IPQ807 device (Xiaomi AX3600) to OOM.

I have performed a number of tests, and have found that:
* Packages (e.g. openvpn, dawn, pbr, sqm) being enabled or disabled do not impact the results in any of the builds I've tried.
* The issue started after build 7f54d9ba1a 
* Under ideal conditions, it is known that IPQ807 devices test to have a higher memory usage than e.g. MediaTek.  The hostapd updates after the above commit seem to have increased the memory useage under load.

Some tests that I performed:

**DOWNGRADE to build 7f54d9ba1a**
DISABLED:  Qualcomm Atheros QCA9887 802.11ac/b/g/n   
ENABLED:   Qualcomm Atheros IPQ8074 802.11ac/ax/n
ENABLED:  Qualcomm Atheros IPQ8074 802.11ax/b/g/n

Reboot Router.  After ~5 minutes:

	Total Available:   ~174 MiB / 406.23 MiB
	Used:   ~260 MiB / 406.23 MiB
	Cached	~65 MiB / 406.23 MiB

Transfer ~5GB at roughly 22MiB/Sec WAN->Wired LAN download and simultaneously start wifi data stream (e.g. Youtube) - this case the device was attached to 5GHz Radio:
	Total Available:   ~138 MiB / 406.23 MiB
	Used:   ~295 MiB / 406.23 MiB
	Cached	~65 MiB / 406.23 MiB

**UPGRADE to build ff95f859eb**

DISABLED:  Qualcomm Atheros QCA9887 802.11ac/b/g/n   
ENABLED:   Qualcomm Atheros IPQ8074 802.11ac/ax/n
DISABLED:  Qualcomm Atheros IPQ8074 802.11ax/b/g/n

Reboot Router.  After ~5 minutes:

	Total Available:   ~182 MiB / 406.23 MiB
	Used:   ~245 MiB / 406.23 MiB
	Cached	~58 MiB / 406.23 MiB

Transfer ~5GB at roughly 22MiB/Sec WAN->Wired LAN download and simultaneously start wifi data stream (e.g. Youtube) - this case the device was attached to 5GHz Radio:
	Total Available:   ~75 MiB / 406.23 MiB
	Used:   ~350 MiB / 406.23 MiB
	Cached	~58 MiB / 406.23 MiB

DISABLED:  Qualcomm Atheros QCA9887 802.11ac/b/g/n   
ENABLED:   Qualcomm Atheros IPQ8074 802.11ac/ax/n
ENABLED:  Qualcomm Atheros IPQ8074 802.11ax/b/g/n

Reboot Router.  After ~5 minutes:

	Total Available:   ~143 MiB / 406.23 MiB
	Used:   ~284 MiB / 406.23 MiB
	Cached	~59 MiB / 406.23 MiB

Transfer ~5GB at roughly 22MiB/Sec WAN->Wired LAN download and simultaneously start wifi data stream (e.g. Youtube) - this case the device was attached to 5GHz Radio:
	**Total Available:   ~33 MiB / 406.23 MiB
	Used:   ~368 MiB / 406.23 MiB
	Cached	~32 MiB / 406.23 MiB**

Although in the above tests the router did not OOM, it was because there was a single Wifi stream going while the large WAN->Wired LAN transfer was occurring.  Under normal situations where family has multiple wifi streams going, the wifi will OOM.

As an additional note, this behavior was observed with all two builds that I have access to between the above two ( a33f1d3515 ,  e3559fb445  ) .  Both these builds follow large changes in hostapd.

### OpenWrt version

ff95f859eb

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Start a large e.g. 5GB+ download from WAN -> Wired LAN with higher speeds.  e.g. 150Megabit (22MB) / second.
While WAN->LAN transfer is occurring, start one or more wireless streams such as a youtube video.

### Actual behaviour

Depending on number of wireless streams, the device will consume large amounts of memory to the point of OOM.
In version up to and including 7f54d9ba1a (prior to hostapd updates in snapshot) the device memory usage was much more stable.

### Expected behaviour

Memory consumption should not increase to the point of OOM.  i.e. similar to 7f54d9ba1a behavior.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

edrikk, 2023-10-07:
Have updated to [a181b9f0f9](https://github.com/openwrt/openwrt/commit/a181b9f0f9269525319024f53d83c7efe9da544b)  and the issue still remains.
After rebooting the router, if I simpliy do a speed test at fast.com the free memory drops from 140MB to 106MB
So with no wireless activity, starting a fast.com speed test WAN -> Wired LAN consumes 34MB of RAM!

I disabled SQM and rebooted before this test.

Rolling back resolves the issue.

reyhard, 2023-10-08:
I also have similar issue on Xiaomi AX3600  after updating to latest snapshot. I use rather humble set of additional packages (adblock, upnp, asu, wireguard) and previously I had rather stable memory usage - basically never dipping below 90MB of free RAM

julk77, 2023-10-09:
I noticed such behaviour too, since I installed build r24043-ff95f859eb on a Xiaomi AX3600.
I was using wpad-openssl initially.
Moving to wpad-basic-mbedtls instead seems to have solved this issue.
Not at home currently, hard to make more tests.

samed, 2023-10-16:
> I noticed such behaviour too, since I installed build r24043-ff95f859eb on a Xiaomi AX3600. I was using wpad-openssl initially. Moving to wpad-basic-mbedtls instead seems to have solved this issue. Not at home currently, hard to make more tests.

I also had the same issue and I was using wpad-wolfssl. After switching to wpad-mbedtls, memory usage went back to a reasonable level.

edrikk, 2023-11-18:
Using various resources and leveraging analysis by bitthief, dchard, McGiverGim, and Spacebar, I put together the below generic script (which I also posted in the [forum thread](https://forum.openwrt.org/t/potential-memory-leak-introduced-in-snapshot-august-sept-2023/172291/45)).  The results look very good as I have not had a single incidence since, and others on the thread experiencing the same issue also shared that they have stopped getting OOM.

The issue could very well have propped up because moving from Kernel 5.15 to 6.1 the IRQs changed, and most people (including myself) had hardcoded IRQ values or none at all.  This script will lookup the IRQ values dynamically and as such is not impacted by IRQ number changes.

@robimarko I know you're considering committing a version of the script.  Would you prefer/like if I close this ticket or just keep it open until later?

Thanks,

```
##########
# IPQ807 does not properly define interrupts, so irqbalance does not work.  Set static values.
# https://github.com/Irqbalance/irqbalance/issues/258
cat > /tmp/set-ipq807-affinity.sh << 'EOF'

#!/bin/sh

set_affinity() {
    irq=$(awk "/$1/{ print substr(\$1, 1, length(\$1)-1); exit }" /proc/interrupts)
    [ -n "$irq" ] && echo $2 > /proc/irq/$irq/smp_affinity
    logger -t /tmp/set-ipq807-affinity.sh "Setting Affinity: $1 ($irq) to $2"
}

# assign 4 rx interrupts to each core
set_affinity 'reo2host-destination-ring1' 1
set_affinity 'reo2host-destination-ring2' 2
set_affinity 'reo2host-destination-ring3' 4
set_affinity 'reo2host-destination-ring4' 8

# assign 3 tcl completions to last 3 cores
set_affinity 'wbm2host-tx-completions-ring1' 2
set_affinity 'wbm2host-tx-completions-ring2' 4
set_affinity 'wbm2host-tx-completions-ring3' 8

# assign 3 ppdu mac interrupts to last 3 cores
set_affinity 'ppdu-end-interrupts-mac1' 2
set_affinity 'ppdu-end-interrupts-mac2' 4
set_affinity 'ppdu-end-interrupts-mac3' 8

# assign lan/wan to core 4
set_affinity 'edma_txcmpl' 8
set_affinity 'edma_rxfill' 8
set_affinity 'edma_rxdesc' 8
set_affinity 'edma_misc' 8

exit 0
EOF
chmod 777 /tmp/set-ipq807-affinity.sh
/tmp/set-ipq807-affinity.sh
##########
```

sppmasterspp, 2023-12-29:
@edrikk 
Currently I test your script on my QNAP QHora-301W with NSS wifi offload build compiled 3 days ago.
For now I have 100-150 MB more available memory compared to only 20-30 MB previously sometimes (260-300 MB free most of the time) and seems that memory recovers faster than before. Let's see how it goes with it.

edrikk, 2024-01-22:
Closing this issue as https://github.com/openwrt/openwrt/commit/ad25cfb6b8a282e47344fba59f42af61e115a8df has been merged into snapshot.

closed 2024-01-22 by edrikk