openwrt/openwrt #13565: Mount Points completely broken on x86/64 Machines

Loafter, 2023-09-29:
### Describe the bug

Mount Points completely broken on x86/64 Machines
I have 3 routers with x86 motherboard and after update to OpenWrt 23.05.0-rc3 system can't auto mount any partitions
at mount points menu it shown error:
[image]

`The block mount command failed with code 127`
i check execution this command at terminal 
and result is:
`root@ProletarskiyOpenWrt:~# block mount
Error loading shared library libubus.so.20220615: No such file or directory (needed by /sbin/block)
Error relocating /sbin/block: ubus_free: symbol not found
Error relocating /sbin/block: ubus_lookup_id: symbol not found
Error relocating /sbin/block: ubus_connect: symbol not found
Error relocating /sbin/block: ubus_invoke_fd: symbol not found
`
Look like we have library linking issues.

### OpenWrt version

r23389-5deed175a5

### OpenWrt target/subtarget

x86/64

### Device

LENOVO 32642U2 and LENOVO 10GB000URU and Acer Veriton N2510G

### Image kind

Self-built image

### Steps to reproduce

install mount options for luci
and call in terminal  
`block mount`

### Actual behaviour

It's shown error

### Expected behaviour

Normal command execution

### Diffconfig

```text
./scripts/diffconfig.sh: No such file or directory
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-29:
You need to do a clean install, seems you run a mix of v22 and v23 userland.

jow-, 2023-10-19:
Indeed, your rootfs is unclean. Using a v22 overlay with a new v23 system maybe? Nothing we can fix on our end.

closed 2023-10-19 by jow-