openwrt/openwrt #13570: opkg: upgrade should only upgrade, not install

brianjmurrell, 2023-09-29:
### Describe the bug

Here is the output of doing an `opkg upgrade libpcap1` on two different devices:

```
Upgrading libpcap1 on root from 1.10.1-3 to 1.10.1-4...
Downloading http://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/base/libpcap1_1.10.1-4_mips_24kc.ipk
Configuring libpcap1.
```

```
Installing libpcap1 (1.10.1-4) to root...
Downloading http://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/base/libpcap1_1.10.1-4_mips_24kc.ipk
Configuring libpcap1.
```

The difference in output is leading me to believe that on the second case, lipcap1 was not actually installed and `opkg upgrade` resulted in installing it.

`opkg upgrade` should not install anything new.  It should be a NOOP in cases where the requested upgraded package is not actually installed.

### OpenWrt version

22.03.05

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-09-30:
Pre-upgrade check does not check package presence
https://github.com/openwrt/opkg-lede/blob/d038e5b6d155784575f62a66a8bb7e874173e92e/libopkg/opkg_upgrade.c#L32

KanjiMonster, 2023-09-30:
> Pre-upgrade check does not check package presence https://github.com/openwrt/opkg-lede/blob/d038e5b6d155784575f62a66a8bb7e874173e92e/libopkg/opkg_upgrade.c#L32

The caller of this function https://github.com/openwrt/opkg-lede/blob/d038e5b6d155784575f62a66a8bb7e874173e92e/libopkg/opkg.c#L404 does.

Though it might also need to check `pkg_t::state_status` as `opkg_remove_package()` https://github.com/openwrt/opkg-lede/blob/d038e5b6d155784575f62a66a8bb7e874173e92e/libopkg/opkg.c#L355 does.

brianjmurrell, 2025-04-16:
So can this be fixed then, or are we just going to disregard it in light of the switch to `apk`?  Is that going to happen any time soon?