openwrt/openwrt #13573: Broken MAC addresses on rtl838x since 4ebba8a05d

Borromini, 2023-09-30:
### Describe the bug

Commit 4ebba8a05d09cc3cdefce585ed51ccec3c6b1206 seems to break MAC addresses on RTL838x devices, resetting them to 00:00:00:00:00, with only the switch interface retaining a valid MAC. Reverting the commit is a crude fix, but so far noone has been able to pinpoint the actual issue.

See report [here](https://forum.openwrt.org/t/openwrt-23-05-0-rc3-third-release-candidate/169419/15) in the 23.05 RC3 thread, as well as further attempts to pinpoint the issue [here in the realtek development thread](https://forum.openwrt.org/t/support-for-rtl838x-based-managed-switches/57875/2569).

Tagging @CHKDSK88 as the commit author, @hailfinger has reported it as well on the forums.

### OpenWrt version

Main / 23.03 branch

### OpenWrt target/subtarget

realtek/rtl838x

### Device

ZyXEL GS1900-8HP v1 / Netgear GS108T v3

### Image kind

Self-built image

### Steps to reproduce

Compile main or 23.05 HEAD image, flash onto device, reboot, check MAC addresses.

### Actual behaviour

All MAC addresses are reset to zeroes and only the switch interface retains the actual MAC address.

### Expected behaviour

All interfaces obtaining their MAC address as prior to 4ebba8a05d09cc3cdefce585ed51ccec3c6b1206.

### Diffconfig

```text
N/A, this happens with both self built and buildbot images.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Borromini, 2023-09-30:
@hauke If noone is able to fix this, I suggest we do revert the commit before releasing 23.05, since you do not want a whole target without MAC addresses.

brada4, 2023-09-30:
Same thread says no functional impairment? Switch ports do not need mac addresses.

dave14305, 2023-09-30:
https://github.com/openwrt/openwrt/blob/5d111b67140d26dee4c8c4f2a6baf4cf077840a4/package/base-files/files/lib/functions/uci-defaults.sh#L110

Shouldn’t e the hyphen be an underscore? Noticed by janh in the forum…

dave14305, 2023-10-01:
Issue was identified in commit comments a year ago. A name with hyphen will get converted to underscore when loaded by jshn.

https://github.com/openwrt/openwrt/commit/9290539ca9c7b296891b1b386052c0fe308e9a62#r80107017

Borromini, 2023-10-01:
Thanks @dave14305. So renaming to `network_device` will fix the symptom here, but I suppose `libubox/jshn.sh` needs to be modified to stop this from happening?

dave14305, 2023-10-01:
The character swap actually happens in the jshn.c in libubox, as I posted later in the forum thread. netifd would also need to be updated if the rename happens in uci-defaults.sh.

Ansuel, 2023-10-02:
@nbd168 wonder if you can help here?

CHKDSK88, 2023-10-03:
I copied `03_gpio_switches` from different target.  Why it's is problematic only in realtek?

dave14305, 2023-10-03:
> I copied `03_gpio_switches` from different target. Why it's is problematic only in realtek?

Only a couple targets call `ucidef_set_network_device_mac` in 02_network.

howels, 2023-10-17:
Looks like this should be linked to the PR https://github.com/openwrt/netifd/pull/13

Borromini, 2023-10-20:
@Ansuel This can be closed now, thanks to everyone for the sleuthing and the fixes (@MadnessASAP).

Ansuel, 2023-10-20:
Fixed by https://github.com/openwrt/openwrt/pull/13622

closed 2023-10-20 by Ansuel