openwrt/openwrt #13576: ksmbd: request one commit backport

odkrys, 2023-09-30:
### Describe the bug

https://github.com/cifsd-team/ksmbd/issues/599

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=39e0c8afdce04157ce229c3d4aa21f80356e88c1

This issue has been resolved upstream. However, I don't know when it will be backported to 5.15.x. I hope that commit will be backported to openwrt before 23.05 is released.

### OpenWrt version

23.05

### OpenWrt target/subtarget

mvebu

### Device

fortigate 50e

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2023-10-02:
@odkrys is it critical? If the patch has been pushed correctly, in 1-2 kernel release we should have the fix backported automatically.

odkrys, 2023-10-03:
It's hard to say it's critical.  It's just a kernel warning.  Just I don't know if this commit will be backported upstream to the 5.15.x stable kernel.

Neustradamus, 2023-10-22:
@neheb: What do you think?

neheb, 2023-10-24:
better to push an update.

closed 2023-12-23 by odkrys

Neustradamus, 2023-12-23:
@neheb: Can you request the ksmbd for Kernel 6.1.x?
The repository has been removed.

Maybe the time to switch to 6.6.x, the new LTS.

There is only 5.15.x: https://github.com/namjaejeon/stable-linux-5.15-ksmbd

Thanks in advance.

Neustradamus, 2023-12-23:
@neheb: Can you request the ksmbd for Kernel 6.1.x?
The repository has been removed:
- https://github.com/namjaejeon/stable-linux-6.1-ksmbd

Maybe the time to switch to 6.6.x, the new LTS.

There is only 5.15.x: https://github.com/namjaejeon/stable-linux-5.15-ksmbd

Thanks in advance.

Source:
- https://github.com/cifsd-team/ksmbd/issues/594#issuecomment-1867335157
- https://github.com/cifsd-team/ksmbd/issues/599#issuecomment-1774022209

Neustradamus, 2024-01-20:
Linked to:
- https://github.com/openwrt/openwrt/issues/13576
- https://github.com/openwrt/openwrt/pull/14406
- https://github.com/openwrt/openwrt/pull/14408
- https://github.com/openwrt/openwrt/pull/14409 (merged)

Merged commit:
- https://github.com/openwrt/openwrt/commit/038b0cb4f1ac08803d601cea695c659c10629196

cc: @odkrys, @neheb, @pesa1234, @ptpt52.