openwrt/openwrt #13582: Can't start xray service

snowwolf007cn, 2023-10-01:
### Describe the bug

installed xray-core 1.8.3 using opkg
the service script was installed to /etc/init.d/xray, but the service wasn't autostart.
try to service start xray, it still didn't work.

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/nand

### Device

Netgear WNDR4300

### Image kind

Official downloaded image

### Steps to reproduce

installed xray-core 1.8.3 using opkg

### Actual behaviour

the service script was installed to /etc/init.d/xray, but the service wasn't autostart.
try to service start xray, it still didn't work.

### Expected behaviour

1. xray service started automatically after installation
2. service can be started, stopped, reload by 'service' command

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-01:
Should be reported in ../packages/
What do you see after boot with:
`logread -e xray` 
in the meantime
Also please confirm version over there:
`opkg list xray-core`

snowwolf007cn, 2023-11-21:
@brada4 I give the info you need below:
```shell
root@OpenWrt:~# opkg list xray-core
xray-core - 1.8.3-1
```
```shell
logread -e xray
<blank>
```
I think the service is not started, so there won't be any log.

brada4, 2023-11-21:
Service is disabled by default, check /etc/config/xray

CravateRouge, 2024-01-03:
Indeed, I had the same issue and you need to switch `option enabled '0'` to '1' in /etc/config/xray before doing `service xray start`

mohsentaleb, 2024-03-08:
Make sure there's only one `config.json` file in `/etc/xray/` otherwise service script can't decide which one to pick and fails starting.