openwrt/openwrt #13586: [MT7628] Xiaomi Router 4A kernel panic when using aireplay-ng

ShredRum, 2023-10-02:
### Describe the bug

Hello. I put the router into monitor mode, then try to deauthenticate via mac, after which the router immediately reboots

I'm not sure if this is a kernel panic, but I found a similar report here
https://github.com/openwrt/openwrt/issues/5102
https://github.com/openwrt/mt76/issues/504
https://github.com/openwrt/mt76/issues/504

Also, I don’t know where it’s better to publish the issue, in mt76 or here, so forgive me if I’m confused.
I'm ready to check this in more detail if you provide me with information on how to record a log/dump. Thank you.

### OpenWrt version

r24054-fe10f97439

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Xiaomi Mi Router 4A (100M International Edition)

### Image kind

Official downloaded image

### Steps to reproduce

1. Install aircrack-ng and airmon-ng
2. Switch the access point to monitor mode (airmon-ng start)
3. Run aireplay-ng -0 100 -a DC:2C:6E:44:D5:83 wlan0mon --ignore-negative-one

The device will freeze and reboot; only the line will appear in the terminal

Waiting for beacon frame (BSSID: DC:2C:6E:44:D5:83) on channel -1

### Actual behaviour

The device will freeze and reboot; only the line will appear in the terminal

Waiting for beacon frame (BSSID: DC:2C:6E:44:D5:83) on channel -1

### Expected behaviour

Sending deauthentication packets to a given MAC without failure

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-02:
Try `logread -f` while crashing the router.
Victim model likely absolutely required to continue.

ShredRum, 2023-10-02:
> Try `logread -f` while crashing the router. Victim model likely absolutely required to continue.

After the crash, nothing new appears. But after switching the adapter to monitor mode, I see repeated spam in the terminal with the lines

Mon Oct 2 22:44:27 2023 daemon.err hostapd: Failed to set beacon parameters

ShredRum, 2023-10-02:
Also the problem does NOT occur with the 5GHz network

ShredRum, 2023-10-05:
Since this problem most likely relates to a specific driver, I decided to close this issue and move it to the appropriate discussion.
https://github.com/openwrt/mt76/issues/825

closed 2023-10-05 by ShredRum