openwrt/openwrt #13597: Bug in MT-7615e

Oozil, 2023-10-03:
### Describe the bug

There is a problem searching for networks in the country code of 00, which shows only channel 36 and channel 40 

How do I make the default driver mode choose another country instead of 00 to any other country on the defualt driver mode?

### OpenWrt version

master

### OpenWrt target/subtarget

mt7621at

### Device

MSG1500

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-03:
Typically one installs a supported release 22.03.5 on your device and change country code via web.

`iw reg set 'DE'` on command line
or 
`option country 'DE'` in /etc/config/wireless

Otherwise - your network card driver does not support passive scan or your AP-s do not beacon often enough to guarantee been seen. Same applies for channels 12/13.
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n12

Oozil, 2023-10-03:
> Otherwise - your network card driver does not support passive scan or your AP-s do not beacon often enough to guarantee been seen. Same applies for channels 12/13

I want to make the automatic country selection select a country other than "00"

brada4, 2023-10-03:
Install the release version, country code will be in the web gui, upper section for an AP for each wireless radio device.
(by virtue of having this question at all I assume its your start moments at openwrt, so go with the "default" good experience, it will be kind of annoyance learning VI editor when you can have it working today)

Namely sysupgrade file erasing all past config.
https://firmware-selector.openwrt.org/?version=22.03.5&target=ramips%2Fmt7621&id=raisecom_msg1500-x-00

The steps:
1)
Download the file to memory disk on the device:
`wget -O /tmp/firmware.bin https://downloads.openwrt.org/releases/22.03.5/targets/ramips/mt7621/openwrt-22.03.5-ramips-mt7621-raisecom_msg1500-x-00-squashfs-sysupgrade.bin`
(i copied the link fro mthe web page.

2)
Check the checksum of file vs the web one:
`sha256sum /tmp/firmware.bin`
Now search displayed firmware checksum in the web page.

3) 
Install firmware wiping configuration 
`sysupgrade -v -n /tmp/firmware.bin`

4)
(webpage on openwrt.lan / 192.168.1.1)