openwrt/openwrt #13600: procd returning error when starting ubusd

oskarirauta, 2023-10-03:
### Describe the bug

[image]

Screenshot, because error does not seem to occur in syslog; on physical screen only (if emulated screen such as qemu counts).

disclaimer:
I readed procd sources partially, all the information about procd code might not be accurate.

Failed to connect to ubus. By using more verbose boot logging, it is revealed that procd is attempting to execute `/etc/init.d/ubus running` when this error occurs. This file does not exist, because ubusd does not need initscript, procd starts it. I created a quite empty init script for testing purposes, which doesn't _start_ anything. But has _running_ function, yet error existed, as after that _succeeding_ it procd attempts to set details of started service of ubus, **to** ubus. Problem is that ubus daemon is not yet ready/does not accept this connection, hence the error.

Yet, when I use `ubus call service list` - proper details about ubus service do exist, I didn't search details where these are set, as clearly when described error occurs, these won't be set - maybe ubus sets it's own state independently, or details are updated later? Here's output of ubus service details:
```
	"ubus": {
		"instances": {
			"instance1": {
				"running": true,
				"pid": 803,
				"command": [
					"/sbin/ubusd"
				],
				"term_timeout": 5,
				"respawn": {
					"threshold": 3600,
					"timeout": 1,
					"retry": 0
				},
				"user": "ubus",
				"group": "ubus"
			}
		}
	},
```

Details seem to be correct.

I assume ubus startup is made in state.c of procd near line 155 - and fix to this would be to alter ubus startup to ignore updating ubus details as a service on the "first run".. Only when updated etc..

I know, it's merely a cosmetic bug. But yet, it's a bug...

### OpenWrt version

r24069-954142f477

### OpenWrt target/subtarget

x86/64

### Device

Red Hat KVM

### Image kind

Self-built image

### Steps to reproduce

boot without log suppression to quiet level

### Actual behaviour

unnecessary error

### Expected behaviour

success without error and not to attempt update details of ubus, when ubus is not ready/available yet

### Additional info

tagging here few people who've been lately involved with project, as original maintainer doesn't seem to be available at GitHub for the moment.

@nbd168 @dangowrt @hauke

### Diffconfig

```text
unrelevant
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

oskarirauta, 2023-10-03:
@nbd168 @dangowrt @hauke

brada4, 2023-10-03:
Picture shows no sign of error?

oskarirauta, 2023-10-03:
@brada4 what you mean? It says it right there;

`Failed to connect to ubus` at the 4th line from bottom.

Its-Just-Nans, 2025-08-25:
Note for future reader

On my side, the issue was that ubusd wasn't working because `/var/run` wasn't existing

So I added

```sh
mkdir -p /var/run
```

Just before the ubusd start

Maybe related to #9511