openwrt/openwrt #13611: odhcp6c: Failed to send SOLICIT message to ff02::1:2 (Permission denied)

septem9er, 2023-10-05:
### Describe the bug

Hey,
I do get the following error messages from odhcp:
```
odhcp6c[16857]: Failed to send RS (Permission denied)
odhcp6c[16857]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
```
I am using a turris omnia, my netconfig looks as follows:
```
config interface 'loopback'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'
        option device 'lo'

config globals 'globals'
        option ula_prefix 'fd42:44d4:f241::/48'

config interface 'lan'
        option proto 'static'
        option ip6assign '64'
        option device 'br-lan'
        option _turris_mode 'managed'
        list ipaddr '192.168.178.1/24'

config interface 'wan'
        option ipaddr '192.168.8.10'
        option netmask '255.255.255.0'
        option gateway '192.168.8.1'
        option hostname 'turris'
        option proto 'pppoe'
        option device 'eth2.7'
        option keepalive '20 1'
        option username 'vodafone-vdsl.komplett/*redacted*-static'
        option password '*redacted*'
        option ipv6 '1'

config interface 'guest_turris'
        option enabled '1'
        option proto 'static'
        option ipaddr '10.111.222.1'
        option netmask '255.255.255.0'
        option device 'br-guest-turris'
        option ip6assign '64'

config interface 'wan6'
        option device '@wan'
        option reqprefix 'auto'
        option reqaddress 'try'
        option ip6gw '2a00:11:f982:2a00::1'
        option proto 'dhcpv6'

config device 'br_lan'
        option name 'br-lan'
        option bridge_empty '1'
        list ports 'lan0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option type 'bridge'

config device 'br_guest_turris'
        option bridge_empty '1'
        option type 'bridge'
        option name 'br-guest-turris'

config device 'dev_wan'
        option name 'eth2.7'

config route
        option gateway '192.168.178.42'
        option interface 'lan'
        option target '10.0.0.0'
        option netmask '255.0.0.0'
```

I did read somewhere, that `ip -6 route add default dev pppoe-wan` resolves the Permission denied error, which it does (not permanent that way). However, it still fails with `(Address not available)` afterwards.

I did see issue #55. My ISP does support IPv6 via dual stack for sure, but no link local address shows up with `ifstatus wan`:
```
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 25,
        "l3_device": "pppoe-wan",
        "proto": "pppoe",
        "device": "eth2.7",
        "updated": [
                "addresses",
                "routes"
        ],
        "metric": 0,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "176.95.*redacted*",
                        "mask": 32,
                        "ptpaddress": "84.58.120.1"
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [
                {
                        "target": "0.0.0.0",
                        "mask": 0,
                        "nexthop": "84.58.120.1",
                        "source": "0.0.0.0/0"
                }
        ],
        "dns-server": [
                "176.95.16.250",
                "176.95.16.251"
        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {

        }
}
```

It does work with another router (a fritzBox).

**Additional info:**
```
ip -6 addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1508 state UP qlen 1024
    inet6 fe80::da58:d7ff:fe01:9486/64 scope link 
       valid_lft forever preferred_lft forever
4: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1024
    inet6 fe80::da58:d7ff:fe01:9487/64 scope link tentative 
       valid_lft forever preferred_lft forever
14: br-guest-turris: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 2a00:11:f982:2aff::1/64 scope global deprecated dynamic 
       valid_lft 5133sec preferred_lft 0sec
    inet6 fd42:44d4:f241::1/64 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::c0f8:daff:fea5:34be/64 scope link 
       valid_lft forever preferred_lft forever
15: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fd42:44d4:f241:1::1/64 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::da58:d7ff:fe01:9486/64 scope link 
       valid_lft forever preferred_lft forever
22: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::6f0:21ff:fe45:e050/64 scope link 
       valid_lft forever preferred_lft forever
23: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::c2f8:daff:fea5:34be/64 scope link 
       valid_lft forever preferred_lft forever
24: guest_turris_1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::c0f8:daff:fea5:34be/64 scope link 
       valid_lft forever preferred_lft forever
25: guest_turris_0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::4f0:21ff:fe45:e050/64 scope link 
       valid_lft forever preferred_lft forever
854: eth2.7@eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::da58:d7ff:fe01:9487/64 scope link tentative 
       valid_lft forever preferred_lft forever
```

### OpenWrt version

r16862+127-170d9e447d

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

turris omnia

### Image kind

Official downloaded image

### Actual behaviour

I should get an IPv6 address

### Expected behaviour

I don't get an IPv6 address. Additionally, the connectio resets every minute, but I am not sure if that is connected.

### Additional info

I already filed in issue in the odhcp6c repo on github, because I didn't see that it isn't monitored: https://github.com/openwrt/odhcp6c/issues/77

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-06:
You are not using OpenWRT.

danpawlik, 2024-01-04:
I create a build from today code and I got similar issue with ipv6, which is odd - on old router which was not reinstalled for a 1 year it was working normally.
Current error: 
```
Thu Jan  4 11:39:27 2024 daemon.err odhcp6c[4383]: Failed to send RS (Address not available)
Thu Jan  4 11:39:28 2024 daemon.err odhcp6c[4383]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
```

What I spotted also:  the option `option ra_slaac '1' `  in `/etc/config/dhcp` even when you enable/disable via luci it does not appear there (comparing old dhcp config and new one).

brada4, 2024-01-04:
yes, it is odd to report turris original fw issues to openwrt.

danpawlik, 2024-01-04:
@brada4 It's not turris issue, but OpenWRT.
Llet me share more info:
```
ubus call system board
{
        "kernel": "6.1.70",
        "hostname": "BPI-R4",
        "system": "ARMv8 Processor rev 0",
        "model": "Bananapi BPI-R4",
        "board_name": "bananapi,bpi-r4",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r24664+77-85f59c8e27",
                "target": "mediatek/filogic",
                "description": "OpenWrt SNAPSHOT r24664+77-85f59c8e27"
        }
}
```
Network:
```
root@BPI-R4:~# uci export network
package network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fda2:8189:e40f::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        option stp '1'
        option igmp_snooping '1'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.88.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config device
        option name 'br-wan'
        option type 'bridge'
        list ports 'eth2'
        list ports 'wan'

config interface 'wan'
        option device 'br-wan'
        option proto 'static'
        option ipaddr 'XXX'
        option netmask '255.255.255.0'
        option gateway 'XXX'

config interface 'wan6'
        option proto 'dhcpv6'
        option device 'br-wan'
        option reqaddress 'try'
        option reqprefix 'auto'
```
[... 221 lines cut ...]
       valid_lft forever preferred_lft forever
default from 2a00:56e0:XX::/64 via fe80::e6d3:f1ff:fe61:f181 dev br-wan  metric 512
unreachable 2a00:56e0:XX::/64 dev lo  metric 2147483647
2a00:56e0:XX::/48 dev br-wan  metric 256
fe80::/64 dev eth0  metric 256
fe80::/64 dev br-wan  metric 256
fe80::/64 dev phy0-ap0  metric 256
fe80::/64 dev br-lan  metric 256
local ::1 dev lo table local  metric 0
anycast fe80:: dev eth0 table local  metric 0
anycast fe80:: dev br-wan table local  metric 0
anycast fe80:: dev phy0-ap0 table local  metric 0
anycast fe80:: dev br-lan table local  metric 0
local fe80::1e7e:51ff:fea1:491c dev phy0-ap0 table local  metric 0
local fe80::3cef:30ff:feee:d921 dev eth0 table local  metric 0
local fe80::5e02:14ff:fe30:1e5a dev br-wan table local  metric 0
local fe80::840f:8bff:fef6:969b dev br-lan table local  metric 0
multicast ff00::/8 dev eth0 table local  metric 256
multicast ff00::/8 dev wg_lan table local  metric 256
multicast ff00::/8 dev br-wan table local  metric 256
multicast ff00::/8 dev phy0-ap0 table local  metric 256
multicast ff00::/8 dev br-lan table local  metric 256
0:      from all lookup local
32766:  from all lookup main
```

brada4, 2024-01-04:
feel free to open a new issue against real snapshot or release (without +77), seems dhcp6 worked partially assigning routes but not address.

danpawlik, 2024-01-04:
@brada4 good idea. Maybe I will start from that point. If it's not working, I will continue in #14140. Thanks for advice

dioni21, 2024-11-22:
I'm getting this on OpenWRT x86_64-generic, all pkgs updated.

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.5'
DISTRIB_REVISION='r24106-10cc5fcd00'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.5 r24106-10cc5fcd00'
DISTRIB_TAINTS=''
```

Log:

```
Fri Nov 22 19:36:00 2024 daemon.err odhcp6c[9296]: Failed to send RS (Operation not permitted)
Fri Nov 22 19:36:00 2024 daemon.err odhcp6c[9296]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Fri Nov 22 19:36:01 2024 daemon.err odhcp6c[9296]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Fri Nov 22 19:36:03 2024 daemon.err odhcp6c[9296]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Fri Nov 22 19:36:04 2024 daemon.err odhcp6c[9296]: Failed to send RS (Operation not permitted)
Fri Nov 22 19:36:07 2024 daemon.err odhcp6c[9296]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Fri Nov 22 19:36:08 2024 daemon.err odhcp6c[9296]: Failed to send RS (Operation not permitted)
Fri Nov 22 19:36:12 2024 daemon.err odhcp6c[9296]: Failed to send RS (Operation not permitted)
Fri Nov 22 19:36:15 2024 daemon.err odhcp6c[9296]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
```

Firewall still at default settings

I have dual uplink, one PPPoE and another routed.  Both are working with IPv6 in an old Netgear WNDR3800 (OpenWrt 22.03.3 r20028-4d371ad93e).  I am configuring a new router to upgrade this old pal, but IPv6 is still not working on the new router, using the very same configurations or any variation I try.

Note: I am not even trying to assign IPv6 to backend.  Just to get IPv6 addrs in my upstream connections.

brada4, 2024-11-22:
Reset the device to clear turris configs and go to forum as you promised. No need to beat a dead horse.

san5ezez, 2025-01-30:
здравствуйте мой роутер на openWRT но выдает ошибку 
daemon.err odhcp6c[2627]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
daemon.err odhcp6c[2627]: Failed to send RS (Network unreachable)

Но она очень похожа на вашу.

san5ezez, 2025-01-30:
hello my router is on openWRT but it gives an error 
daemon.err odhcp6c[2627]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
daemon.err odhcp6c[2627]: Failed to send RS (Network unreachable)

But it is very similar to yours.

webysther, 2025-03-20:
Try this one, worked to me: https://tomthorogood.net/writing/openwrt-disabled-ipv6/