openwrt/openwrt #13613: [23.05.0-rc4] Ubiquiti EdgeRouter X has increaced cpu load and reduced network speed on eth0 when using it as an all ports Switch compared to 22.03.5, hardware offloading activated.

henielssen, 2023-10-06:
### Describe the bug

Ubiquiti EdgeRouter X has increaced cpu load and reduced network speed on eth0 when using it as an all ports Switch compared to 22.03.5, hardware offloading activated.

The same issue is identified on 23.05.0-rc3.

### OpenWrt version

23.05.0-rc4

### OpenWrt target/subtarget

ramips/mt7621

### Device

Ubiquiti EdgeRouter X

### Image kind

Official downloaded image

### Steps to reproduce

Identical configuration in both 22.03.5 and 23.05.0-rc4, as Switch on all ports:
interface lan:
- device br-lan bridging all ports
- assigned to green fw zone
hardware offloading activated

Network speed measured with Iperf 3.1.3, 30 second sessions, and averaged on 5 test runs per test 1 through 4.
CPU utilization measured with top.

1. Test max network speed on traffic between eth0 and eth2. Also observe CPU utilization.
2. Repeat 1, but replace eth0 with eth1, no cable in eth0.

Prior to tests 3 and 4, enable Bridge VLAN filtering, create a VLAN 3 and make it untagged on primary VLAN on all ports
Assign lan to br-lan.3

3. Test max network speed on traffic between eth0 and eth2. Also observe CPU utilization.
4. Repeat 3, but replace eth0 with eth1, no cable in eth0.

### Actual behaviour

23.05.0-rc4:
1. Speed: 914 Mbits/sec, CPU: 74% idle a)
2. Speed: 938 Mbits/sec, CPU: 99% idle
3. Speed: 902 Mbits/sec, CPU: 65% idle b)
4. Speed: 938 Mbits/sec, CPU: 99% idle

a) Fluctuates between 870 and 930 Mbits/sec between 30 sec test runs
b) Fluctuates between 830 and 910 Mbits/sec between 30 sec test runs

Same tests on 22.03.5:
1. Speed: 940 Mbits/sec, CPU: 99% idle
2. Speed: 941 Mbits/sec, CPU: 99% idle
3. Speed: 941 Mbits/sec, CPU: 99% idle
4. Speed: 944 Mbits/sec, CPU: 99% idle

Indicating that hardware offload might not be in effect on eth0 with this configuration on 23.05.0-rc4, while it is on 22.03.05.

### Expected behaviour

Same high gigabit speed on eth0 <-> eth2 traffic as on eth1 <-> eth2 traffic.
Same low cpu utilization on eth0 <-> eth2 traffic as on eth1 <-> eth2 traffic.

### Additional info

Running the same tests on EdgeRouter X SFP, also on the same target/subtarget ramips/mt7621 as EdgeRouter X

23.05.0-rc4:
1.  Speed: 938 Mbits/sec, CPU: 99% idle
2. Speed: 938 Mbits/sec, CPU: 99% idle
3. Speed: 938 Mbits/sec, CPU: 99% idle
4. Speed: 937 Mbits/sec, CPU: 99% idle

Same tests on 22.03.5:
1. Speed : 942 Mbits/sec, CPU: 99% idle
2. Speed : 942 Mbits/sec, CPU: 99% idle
3. Speed : 942 Mbits/sec, CPU: 99% idle
4. Speed : 944 Mbits/sec, CPU: 99% idle

EdgeRouter X SFP has no increased CPU load on 23.05.0-rc4 for tests 1 and 3, but a 4-5 Mbits/sec decreased network speed on all four tests compared to 22.03.5.

Perhaps see how the EdgeRouter X SFP, with basically the same hardware as EdgeRouter X, manage to achieve no increased CPU load in 23.05.0-rc4? In addition to investigating the 22.03.5 release for EdgeRouter X of course.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

simonvetter, 2023-10-06:
This is a side effect of https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478 i.e. eth0 on the ER-X now acts as a dedicated ethernet port (goes straight to the CPU), so frames forwarded between eth0 and other ports need to be software-switched.
Doing this allows vastly increased routing performance between LAN and WAN ports as the CPU now has two 1Gbit/s Ethernet links instead of just one earlier.

Since this optimization can't be done on the ER-X SFP (due to the way Ubiquiti changed the wiring on the board), eth0 on the ER-X SFP is still tied to the switch. That explains why the CPU isn't pegged on that router (switching still happens in hardware).

If you can, use eth0 as the WAN/uplink port and other ports as part of the switch. That should alleviate your issues and give you increased LAN<>WAN routing performance.

brada4, 2023-10-06:
To use 'offloading' you need to forward between interfaces via ??tables rule. Typically not allowed on a bridge.
You need to state the measurement tool. Gigabit does one gigabit up and anothet down.

Shine-, 2023-10-08:
> If you can, use eth0 as the WAN/uplink port and other ports as part of the switch. That should alleviate your issues and give you increased LAN<>WAN routing performance.

The hardwiring of `eth0` to be WAN is doubleplusungood. Since eth0 is the PoE input port, so if you are powering your ER-X via PoE, you would usually use `eth0` as one of the LAN ports, not WAN.

All in all, this "2GBps hw-accel" PR with its unchangeable hardwiring received much too few critical review, imo. Since for all affected devices, it broke wire-speed switching capability.

henielssen, 2023-10-09:
> This is a side effect of https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478 i.e. eth0 on the ER-X now acts as a dedicated ethernet port (goes straight to the CPU), so frames forwarded between eth0 and other ports need to be software-switched.

@simonvetter thank you for the link to the forum page, it explains a lot.

[arinc9](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/63)  labels sacrificing eth0 performance in switch mode in order to get increased router performance "a design choice". It's admitted that switching at CPU level, which is enforced by the "mt7621 refactor and mux phy0/4 to gmac1 to achieve 2 Gbps CPU bandwidth" change is "not very efficient" and we're advised to "do routing instead".

> If you can, use eth0 as the WAN/uplink port and other ports as part of the switch. That should alleviate your issues and give you increased LAN<>WAN routing performance.

I have five ER-X set up as managed VLAN switches, with POE on eth0. Upgrading to 23.05.0 will seriously degrade my network performance, while at the same time increasing cpu load and power usage. I wholeheartedly support [vevere](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/102)s and [hmh](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/144)s requests to make the "enhancement" selectable, and I strongly supports hmhs view when it comes to the suggested [proper way forward](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/213).

I'm all for increased performance in router mode, what I question is how easily switch performance has been sacrificed to achieve this, and how any concerns raised have been dismissed. I guess regular OpenWrt firmware users don't follow change initiatives (I certainly did not), and don't expect that new versions of OpenWrt firmware will decrease hardware performance compared to earlier releases, by design! It's likely that the decreased switch performance will be raised as an issue by more OpenWrt users once 23.05.0 is officially released, as it then will become evident to "the masses".

The only options I see now for my use case are to not upgrade from 22.3.05, switch back to vendor stock firmware or to start compiling images myself with changes to the devicetree. None of these options are attractive.

henielssen, 2023-10-09:
> To use 'offloading' you need to forward between interfaces via ??tables rule. Typically not allowed on a bridge. 

@brada4, regarding hw offload, yes I see your point. I saw it had a huge impact when using the device as a router, and did not verify it was not needed / had no effect when used as a switch. 

> You need to state the measurement tool. Gigabit does one gigabit up and anothet down.

I'll skip following up on the measurements for now, the issue was already well documented in the thread that @simonvetter provided in his post, and thus was known to the developers who still went ahead and made the change without providing an alternative for the non-router use of ramips/mt7621 devices, who are now severely affected by the change.

henielssen, 2023-10-09:
> All in all, this "2Gbps hw-accel" PR with its unchangeable hardwiring received much too few critical review before getting merged, imo. Since for all affected devices, it broke wire-speed switching capability.

@Shine- I totally agree with your points, see my reply to @simonvetter and @brada4. What do we do now?

Shine-, 2023-10-09:
> What do we do now?

Two options:
1. Live with it
2. PR a fix and hope for it to get accepted

brada4, 2023-10-09:
Offload:
Enable bridge filtering, then accept forwarding between bridge members that support offload, then make another flowtable that would take in tcp/udp between interfaces mentioned via quickpath.

Kinda lots of machinery to combat .1% performance loss due to 5% CPU load.

henielssen, 2023-10-09:
> 1. Live with it

No,  too vested in the current setup where investments and configuration choices have been made based on the functionality and performance of the current and previous official OpenWrt releases.

> 2. PR a fix and hope for it to get accepted

So non-OpenWRT programmers don't have a say?

henielssen, 2023-10-09:
> Offload: Enable bridge filtering, then accept forwarding between bridge members that support offload, then make another flowtable that would take in tcp/udp between interfaces mentioned via quickpath.

@brada4: How about simply keep providing the option to don't connect eth0 directly to cpu, which is the current functionality in the latest official release, 22.03.5?

> Kinda lots of machinery to combat .1% performance loss due to 5% CPU load.

Since you insist:

iperf2 (windows)
Server: iperf-win-isoch-fix -s
Client: iperf-win-isoch-fix -c [server ip] -t 20 -d

switch config, bridge on all ports, VLAN filtering enabled

### 23.05.0-rc4
**eth0<->eth2**
cpu ~20% idle
[  2] 0.00-20.04 sec  1.24 GBytes   531 Mbits/sec
[  1] 0.00-20.11 sec  1.42 GBytes   608 Mbits/sec
531+608 = 1139 Mbits/sec

**eth1<->eth2**
cpu 99% idle
[  1] 0.00-20.06 sec  2.19 GBytes   936 Mbits/sec
[  2] 0.00-20.03 sec  2.17 GBytes   931 Mbits/sec
936+931 = 1867 Mbits/sec

### 22.03.5
**eth0<->eth2**
cpu 99% idle
[  2] 0.00-20.01 sec  2.16 GBytes   927 Mbits/sec
[  1] 0.00-20.05 sec  2.18 GBytes   935 Mbits/sec
927+935 = 1862 Mbits/sec

**eth1<->eth2**
cpu 99% idle
[  2] 0.00-20.03 sec  2.17 GBytes   932 Mbits/sec
[  1] 0.00-20.05 sec  2.18 GBytes   934 Mbits/sec
932+934 = 1866 Mbits/sec

99-20 = _**79% increased cpu load**_

1867-1139 = 728
728/1867*100 = **_~39% performance loss_**

brada4, 2023-10-09:
Now we're talking.....
531+608 = 1139 Mbits/sec exceeds manufacturer's spec , but not flattering for your non-default usage. 
Can you check if `nft list ruleset` has both adapters in question bound to flowtable so that it is worth trying bridge 'oflioad' at all. Evens soft mode drops cpu consumption.

henielssen, 2023-10-09:
> Can you check if `nft list ruleset` has both adapters in question bound to flowtable so that it is worth trying bridge 'oflioad' at all. Evens soft mode drops cpu consumption.

`nft list ruleset` yields no output. The lan interface is assigned to the br-lan device and green lan firewall zone. The br-lan device bridges ports eth0 through eth4. Ports eth0 and eth2 are currently in use.

[image]

brada4, 2023-10-10:
> hardware offloading activated

???

brada4, 2023-10-10:
You need to load kmod-br-netfilter, then construct NEW flowtable and new tcp/udp rule and general rule to allow packets to pass between interfaces prepending the accelerator rule to that (it can be done in etc/nftabled.d.
basically upfront the ruleset listing if offload is enabled you see interfaces that checked out as offload-capable, they are usually different between hardware and sooftware offload modes. Please check you have eth0 and eth1 both next to eachother. I still dont get what offload you claimed in issue title though.

henielssen, 2023-10-10:
> I still dont get what offload you claimed in issue title though.

Already answered in my first response to you: It was an error on my part, hw offloading has no effect on an all ports bridged switch on either of the tests listed in the op, it's a router feature, not relevant in switch mode. Also, the speed measurements in that post done with iperf3 were simplex, not duplex, so not really that relevant. The tests done later with iperf2 listed in an earlier comment are relevant.

> You need to load kmod-br-netfilter, then construct NEW flowtable and new tcp/udp rule and general rule to allow packets to pass between interfaces prepending the accelerator rule to that (it can be [...]

Just so I understand what we're trying to achieve: Is the goal to have hardware offloading between eth0 and eth1-eth4 when eth0-eth4 are all part of the br-lan bridge? If I understand @arinc9 in [this post](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/63) correctly, this is not possible. What am I missing or not understanding here?

FYI, I'll be unable to provide any quick feedback on your suggestions, I have not made any manual nftable entries before and thus have a steep climb ahead of me in this regard.

brada4, 2023-10-10:
You said something about eth0 and eth1
What do you see in `ip link` then followed up by `ethtool -i eth0` `ethtool -i eth1`
If they are distinct adapters shortcut forwarding may help. If there is only one you are right. 
make that.file to test offload ability:
```
table inet testing {
        flowtable ft {
                hook ingress priority filter
                devices = { eth0,eth1 }
                #flags offload
        }
```
then run nft -c that.file altering commented line which means hardware support.

if there is offload support across slowness link over CPU it can work for betterment.

henielssen, 2023-10-10:
I think it's a no-go? [ER-X_eth0-eth4_bridged_switch.txt](https://github.com/openwrt/openwrt/files/12857777/ER-X_eth0-eth4_bridged_switch.txt)

brada4, 2023-10-10:
Check all 6 ports for (includIng 'dsa') for offload capability.
No problem whatsoever making states on a bridge with filter.
1 drop of confusion from eth3 being port label i.p.o being 4th netcard in system

henielssen, 2023-10-11:
> Check all 6 ports for (includIng 'dsa') for offload capability.

The config you provided resulted in `Error: syntax error, unexpected newline, expecting string or last`. Any config I try results in the same error: I added what I presume was a missing closing bracket and some semicolons as per the samples on [this page](https://docs.kernel.org/networking/nf_flowtable.html), but to no avail.

[image]

kmod-br-netfilter is installed, and these:

[image]

Am I missing something?

brada4, 2023-10-11:
Sory for mispelling....
```
root@OpenWrt:~# nft -c -f testing.nft ; echo $?
0
root@OpenWrt:~# cat testing.nft 
table inet testing {
        flowtable ft {
                hook ingress priority filter;
                devices = { eth0 }
                #flags offload
        }
}
root@OpenWrt:~# 
```

brada4, 2023-10-11:
(the whole idea is to make side-table next to inet fw4 one that does only bridge offloading while fw4 neutrally allows all traffic on the bridge)

henielssen, 2023-10-11:
I still don't get what we're trying to achieve. In 23.05.0-rc4, eth0 is directly connected to the CPU so it's no longer possible to benefit from the bridge offloading feature of the DSA subdriver, as the dev for the PR altering eth0 devicetree connection confirms in [this post](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/63).

> (the whole idea is to make side-table next to inet fw4 one that does only bridge offloading while fw4 neutrally allows all traffic on the bridge)

I'm able to load nft rules, and to remove them. I have tried eth0 with dsa, eth1, eth2, eth3 and eth4: There's no increased performance when flags offload is enabled on either of these five configurations, and there's always a massive CPU load involved when traffic goes over eth0. As I understand it, CPU is always involved when using eth0, and there's no way around it other than removing the eth0 to CPU direct connection in devicetree that has been introduced in the in the 23.05.0-rc* releases.

Note that traffic between either of the ports eth1, eth2, eth3 and eth4 already are at full duplex gigabit speed, with no cpu usage. Using the current official stable release build of the firmware, 22.03.5, eth0 is also attached to dsa, there is no "eth0" listed in `ip link`, and we get full duplex gigabit speed between any of the five ports eth0 through eth4 i.e. also including eth0 that is not directly attached to the CPU in this and previous versions of the firmware.

Shine-, 2023-10-11:
> > 1. Live with it
> 
> No, too vested in the current setup where investments and configuration choices have been made based on the functionality and performance of the current and previous official OpenWrt releases.

If you don't want to live with it, yet still want to upgrade to 23.05, simply undo the change. It's trivial, at least for the ER-X, and as of now, there have been zero DTS changes for the ER-X that may require manual rebasing work. Just remove the handful of lines that were added by the commit in question by hand, or automatically:

<details>
<summary>by using a patch file</summary>

```
diff --git a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts
index 2b5993db44..bca7c61690 100644
--- a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts
+++ b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts
@@ -4,26 +4,3 @@
 	model = "Ubiquiti EdgeRouter X";
 	compatible = "ubnt,edgerouter-x", "mediatek,mt7621-soc";
 };
-
-&gmac1 {
-	status = "okay";
-	label = "eth0";
-	phy-handle = <&ethphy0>;
-
-	nvmem-cells = <&macaddr_factory_22>;
-	nvmem-cell-names = "mac-address";
-};
-
-&mdio {
-	ethphy0: ethernet-phy@0 {
-		reg = <0>;
-	};
-};
-
-&switch0 {
-	ports {
-		port@0 {
-			status = "disabled";
-		};
-	};
-};
```
</details>

<details>
<summary>or using shell commands only</summary>

```
git reset --hard origin
git checkout v23.05.0-rc4
git checkout 81bd10e target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts
commit
```
</details>

> So non-OpenWrt programmers don't have a say?

They have, by contributing. For your voice to be heard, you should, however, bring forward convincing reasoning.

> > I don't have an ER-X, though, and even if I had one, I wouldn't be using the non-standard "passive PoE" port for powering the device, as I'm using solely standard 802.3af/at PoE equipment.
>
> The vendor supports it, the hardware takes it, and I easily power five ER-X with PoE from one ER-X SFP. Rock solid for years, a great setup supporting managed gigabit switching on the cheap providing four available VLAN ports in five different locations without a single external psu.

Just describing my use case, as I prefer standards-compliant PoE over UBNT's el-cheapo version. Your use case may be different, and I'm not questioning it. Rather off-topic, anyway :-)

arinc9, 2023-10-13:
> If you don't want to live with it, yet still want to upgrade to 23.05, simply undo the change.

@henielssen, this is the best thing to do for your own use case at the moment. I am working on the MT7530 DSA subdriver to allow changing the CPU port the switch port connects to. I will adapt all of the devicetrees accordingly after this, including ER-X. In the end, it will be possible to change the CPU port the switch ports connect to from userspace which will address the issue described here.

brada4, 2023-10-13:
@arinc9 I think here the use case is to disconnect 2nd CPU NIC completely and have all sockets in an unmanaged switch.

henielssen, 2023-10-16:
> @henielssen, this is the best thing to do for your own use case at the moment. 

Not only for me I think, there will be more disappointed faces when 23.05.0 now officially released is installed en masse on one of the most popular OpenWrt hardware platforms. Most OpenWrt users do not compile their own images from scratch, the option left is then to keep the 22.03.5 version for managed switch and AP use cases if eth0/wan port is needed.

> I am working on the MT7530 DSA subdriver to allow changing the CPU port the switch port connects to. I will adapt all of the devicetrees accordingly after this, including ER-X. In the end, it will [...]

@arinc9 That's great news! Do you have an estimated timeframe for this effort? 
Is this related to [your dialogue](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/64) with [drbrains](https://forum.openwrt.org/u/drbrains) back in August 22?

Kudos on the [2Gbps CPU bandwidth patch](https://github.com/openwrt/openwrt/pull/10238) @arinc9, it works a treat and caused me to switch my router from an ER-X SFP (using SFP and fiber directly from ISP) to an ER-X with eth0 RJ45 connection to ISP hardware in bridged mode (that I previously ditched). I now get up to 940+900 duplex over the router, not too shabby for that tiny piece of inexpensive hardware :-)

henielssen, 2023-10-16:
> I think here the use case is to disconnect 2nd CPU NIC completely and have all sockets in an unmanaged switch.

@brada4: I do use VLANs so the unit is used as a managed switch then, right? Or are we talking about different things?

arinc9, 2023-10-16:
> Not only for me I think, there will be more disappointed faces when 23.05.0 now officially released is installed en masse on one of the most popular OpenWrt hardware platforms. Most OpenWrt users do [...]

Most devices with MT7530-included MT7621 SoCs utilise all 5 ports of the switch. That means there are 4 ports remaining to use for switching without hindered performance. I expect a very small amount of people to be affected negatively by this.

> @arinc9 That's great news! Do you have an estimated timeframe for this effort?

No, sorry. Though I can promise it will be done.

> Is this related to [your dialogue](https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478/64) with [drbrains](https://forum.openwrt.org/u/drbrains) back in August 22?

Yes, the DSA subsystem didn't support multiple CPU ports back then.

henielssen, 2023-10-18:
On many ramips/mt7621 devices, eth0/wan port is now directly attached to CPU, due to [mt7621 refactor and mux phy0/4 to gmac1 to achieve 2 Gbps CPU bandwidth](https://github.com/openwrt/openwrt/pull/10238). This patch has the adverse effect that if this port is included and used in a bridge and becomes part of a switch setup, it will no longer benefit from the bridge offloading feature and will have lower Mbps and higher CPU usage than in earlier versions of the OpenWrt firmware, i.e. <= 22.03.5.

There are two known workarounds:

1. Don't upgrade your firmware beyond 22.03.5.
2. Build your own firmware image, with the pull request [10238](https://github.com/openwrt/openwrt/pull/10238) removed. See Shines [post above](https://github.com/openwrt/openwrt/issues/13613#issuecomment-1757724074).

arinc9 is [working on the MT7530 DSA subdriver](https://github.com/openwrt/openwrt/issues/13613#issuecomment-1761696133) to allow changing the CPU port the switch port connects to. In the end, it will be possible to change the CPU port the switch ports connect to from userspace, thus solving this issue. There's no known timeframe for this patch, but he promises [it will be done](https://github.com/openwrt/openwrt/issues/13613#issuecomment-1764960126).

I close this issue.

Thanks to @simonvetter, @brada4, @Shine- and @arinc9 for contributing.

closed 2023-10-18 by henielssen

howels, 2026-10-30:
Note this edgerouter x target is currently broken in 25.12 when trying to switch raw ethernet. Frames are either dropping or being mangled whilst realtek targets work correctly.