openwrt/openwrt #13631: Any way to fix?

ChaingTsung, 2023-10-08:
### Describe the bug

```
make[3]: Entering directory '/home/yoki/openwrt/feeds/routing/ohybridproxy'
touch /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.prepared_f5a3eeefbd76c8cfd0e8f26c5557cfbd_6664517399ebbbc92a37c5bb081b5c53_check
. /home/yoki/openwrt/include/shell.sh; xzcat /home/yoki/openwrt/dl/ohybridproxy-2020-05-22-0dfef1eb.tar.xz | tar -C /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb
touch /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.prepared_f5a3eeefbd76c8cfd0e8f26c5557cfbd_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.configured_*
rm -f /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.ohybridproxy_installed
mkdir -p /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb
(cd /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb; CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb=ohybridproxy-2020-05-22-0dfef1eb -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb=ohybridproxy-2020-05-22-0dfef1eb -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -znow -zrelro " cmake --no-warn-unused-cli -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=aarch64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr;/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl -DCMAKE_PREFIX_PATH=/home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb )
Not searching for unused variables given on the command line.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 12.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb
touch /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.built
touch /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.built_check
MAKEFLAGS="" /home/yoki/openwrt/staging_dir/host/bin/ninja  -j1 -C /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb 
ninja: Entering directory `/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb'
[2/20] Building C object CMakeFiles/ohybridproxy.dir/src/dns2mdns.c.o
FAILED: CMakeFiles/ohybridproxy.dir/src/dns2mdns.c.o 
/home/yoki/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/bin/aarch64-openwrt-linux-musl-gcc -D_GNU_SOURCE -I/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/src -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb=ohybridproxy-2020-05-22-0dfef1eb -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -g -std=c99 -DNDEBUG   -Wall -Wextra -MD -MT CMakeFiles/ohybridproxy.dir/src/dns2mdns.c.o -MF CMakeFiles/ohybridproxy.dir/src/dns2mdns.c.o.d -o CMakeFiles/ohybridproxy.dir/src/dns2mdns.c.o -c /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/src/dns2mdns.c
In file included from /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/src/dns2mdns.c:18:
/home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/src/dns2mdns_i.h:21:10: fatal error: dns_sd.h: No such file or directory
   21 | #include <dns_sd.h>
      |          ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:55: /home/yoki/openwrt/build_dir/target-aarch64_cortex-a53_musl/ohybridproxy-2020-05-22-0dfef1eb/.built] Error 1
make[3]: Leaving directory '/home/yoki/openwrt/feeds/routing/ohybridproxy'
time: package/feeds/routing/ohybridproxy/compile#0.47#0.38#0.82
    ERROR: package/feeds/routing/ohybridproxy failed to build.
make[2]: *** [package/Makefile:120: package/feeds/routing/ohybridproxy/compile] Error 1
make[2]: Leaving directory '/home/yoki/openwrt'
make[1]: *** [package/Makefile:114: /home/yoki/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/openwrt'
make: *** [/home/yoki/openwrt/include/toplevel.mk:232：world] 错误 2
```

### OpenWrt version

`main` branch

### OpenWrt target/subtarget

lastest

### Device

banana pi r3

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ChaingTsung, 2023-10-08:
[Here is the seeds conf](https://github.com/openwrt/openwrt/files/12840182/seeds.txt)

ChaingTsung, 2023-10-08:
I do not choose the ` ohybridproxy`.
[image]

brada4, 2023-10-08:
Use openwrt?
CONFIG_PACKAGE_luci-app-aliddns is not from here, you are using an unsupported fork.

ChaingTsung, 2023-10-08:
> Use openwrt? CONFIG_PACKAGE_luci-app-aliddns is not from here, you are using an unsupported fork.

Yes,I use this official source code . `luci-app-aliddns` is a custom app package.

brada4, 2023-10-08:
Please repeat problem with genuine package collection.
Origins absent no telling which of hundreds of custom packages pull ohybridproxy as a dependency. 
The dns_sd header is correctly pulled by package here https://github.com/openwrt/routing/blob/c2124dd2ad542f526ee84e1984f03dd2f5b34dfe/ohybridproxy/Makefile#L33

ChaingTsung, 2023-10-09:
custom packages is from `https://github.com/kenzok8/openwrt-packages` and `https://github.com/kenzok8/small`, however  no one use `ohybridproxy` as a dependency.

brada4, 2023-10-09:
You should report this issue to the fork you are using.