openwrt/openwrt #13640: Networking (bridge ?) broken on IPQ40xx since kernel move to 6.1

Codemonkey1973, 2023-10-09:
### Describe the bug

Since d686f2153ca2fd30ff9ad1e022e084b4e891106f, networking seems to be broken for me, prior to this works fine.

I have a bridge (br-lan) which consists of eth0, lan1, lan2, lan3 and lan4

On the 4 lan ports are connected some IP cameras.

My application starts up and disables the lan1-4 ports (done by my application running the command "ip link set lanX down" where X=1 to 4), and then enables them one by one (done by my application running the command "ip link set lanX up" where X=1 to 4). This causes the camera attached to each port to reset.

Once each lan port comes up, the camera attached to it (previously configured with a fixed IP address) spits out some UDP messages to allow it to be detected. This works fine and the application receives these device announcements.

However, next, the application attempts to send some UDP messages back to configure the camera and isn't able to. Running wireshark on the br-lan interface reveals that the ARP "who has" request is never answered.

Other info...

I've tried changing things to enable the 4 lan ports at startup, this has no effect.

If I sniff traffic on the lan1-4 ports, I see each camera start up and begin sending messages (some config, some packets containing video), however I don't see any attempt to communicate back, no ARP who has messages, nothing.
If I sniff traffic on br-lan bridge, I don't see any of the messages from the cameras, but I DO see the ARP who has messages when my application tries to communicate with a camera.

So basically it looks like the bridge isn't bridging!

Any suggestions ?

### OpenWrt version

r24099-e66eed033f

### OpenWrt target/subtarget

ipq40xx/generic

### Device

8devices Habanero DVK

### Image kind

Self-built image

### Additional info

```

root@tedd-mesh-node-10-64-186-130:/# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.56 (ubuntu@ubuntu-VirtualBox) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.3.0 r23902-948ad2ec7a) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sun Oct  8 12:05:12 2023
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: 8devices Habanero DVK
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000009fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000009fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009fffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s33812 r8192 d23532 u65536
[    0.000000] pcpu-alloc: s33812 r8192 d23532 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: rootfstype=squashfs
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[... 652 lines cut ...]
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK_ENABLED=y
# CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_ENABLED is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_PSK_ENABLED is not set
CONFIG_MBEDTLS_NIST_KW_C=y
# CONFIG_MBEDTLS_PLATFORM_C is not set
# CONFIG_MBEDTLS_RIPEMD160_C is not set
CONFIG_MBEDTLS_RSA_NO_CRT=y
# CONFIG_MBEDTLS_SELF_TEST is not set
# CONFIG_MBEDTLS_SSL_RENEGOTIATION is not set
# CONFIG_MBEDTLS_SSL_TRUNCATED_HMAC is not set
# CONFIG_MBEDTLS_VERSION_C is not set
# CONFIG_MBEDTLS_VERSION_FEATURES is not set
# CONFIG_MBEDTLS_XTEA_C is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_libmbedtls=y
# CONFIG_PCRE_JIT_ENABLED is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Codemonkey1973, 2023-10-09:
/etc/config/network

```
config interface 'loopback'
option proto 'static'
option ipaddr '127.0.0.1'
option netmask '255.0.0.0'
option device 'lo'

config globals 'globals'
option ula_prefix 'fd8b:59e7:0201::/48'
option packet_steering '1'

config interface 'lan'
option proto 'static'
option ipaddr '192.168.40.254'
option netmask '255.255.255.0'
option device 'br-lan'

config interface 'wan'
option proto 'static'
option device 'br-wan'
option ipaddr '10.0.0.10'
option netmask '255.0.0.0'
option ip6ifaceid 'eui64'
option ip6assign '64'

config switch
option name 'switch0'
option reset '1'
option enable_vlan '1'

config switch_vlan
option device 'switch0'
option vlan '1'
option ports '1 2 3 4 0'

config device
option name 'br-lan'
option type 'bridge'
list ports 'eth0'
list ports 'lan1'
list ports 'lan2'
list ports 'lan3'
list ports 'lan4'
option ipv6 '0'
option bridge_empty '1'

config device
option name 'br-wan'
option type 'bridge'
list ports 'wan'
option bridge_empty '1'

config device
option name 'phy0-ap0'

config device
option name 'eth0'

config device
option name 'lan1'

config device
option name 'lan2'

config device
option name 'lan3'

config device
option name 'lan4'

```

john-tho, 2023-10-09:
> bridge (br-lan) which consists of eth0, lan1, lan2, lan3 and lan4

> ```
> config switch
> option name 'switch0'
> option reset '1'
> option enable_vlan '1'
>
> config switch_vlan
> option device 'switch0'
> option vlan '1'
> option ports '1 2 3 4 0'
>
> config device
> option name [...]

Looks like config is a mix between the old `swconfig` driver, and the new DSA driver.
Could you please remove `eth0` (DSA root) from the bridge, and see if you still get issues?
Also, the `switch` and `switch_vlan` stanzas will not do anything for a DSA device, so they can be removed as well.

brada4, 2023-10-09:
Why do you have swconfig section alongside dsa one? The one with 0 1 2 3 is old, check swconfig vs ip link which hardwaee you have.

Codemonkey1973, 2023-10-09:
Thanks for the responses. Removing the switch and switch_vlan sections together with the eth0 bit did the trick and its now working again

They were in there simply due to being required when I started the project a couple of years ago! Global supply chain issues meant various bits of it were not available until recently and it's just dragged on and on. I just keep updating the openwrt stuff to try and keep things up to date.

I'll mark this as closed.

closed 2023-10-09 by Codemonkey1973