openwrt/openwrt #13642: Bug: [23.05.0-rc4] mt76/mt7921: AP fail to start after reboot

davidngrc, 2023-10-09:
### Describe the bug

I am on proxmox 7.4-17 with Kernel Linux 5.15.116-1-pve
I have only one wifi card passthrough to this x64 system, which is a 
MediaTek Model: MT7921
AW-XB468NF REV0A
the AP mode works fine, but after I reboot OpenWrt, the AP will not associate with the m2 card anymore.
I need to manually disable the AP in the UI, and then re-enable it, then the wifi works.

root@OpenWrt:~# lspci
`01:00.0 Network controller: MEDIATEK Corp. MT7921 802.11ax PCI Express Wireless Network Adapter`

root@OpenWrt:~# iw list
`Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: managed
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
				HE SU PPDU with 1x HE-LTF and 0.8us GI
				NDP with 4x HE-LTF and 3.2us GI
				STBC Tx <= 80MHz
				STBC Rx <= 80MHz
				Full Bandwidth UL MU-MIMO
				Partial Bandwidth UL MU-MIMO
				DCM Max Constellation: 2
				DCM Max Constellation Rx: 2
				SU Beamformee
				Beamformee STS <= 80Mhz: 3
[... 358 lines cut ...]
[  482.568976] br-lan: port 2(phy0-ap0) entered forwarding state
[  482.570194] br-lan: port 2(phy0-ap0) entered disabled state
[  483.153391] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[  483.154634] br-lan: port 2(phy0-ap0) entered blocking state
[  483.155178] br-lan: port 2(phy0-ap0) entered forwarding state`

### OpenWrt version

r23482-7fe85ce1f2

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-09:
`iwinfo`
`/etc/config/wireless` (key option removed)
`logread -e hostapd`
And please attach files bigger than screen.

davidngrc, 2023-10-09:
I try to use the "add code" function here, it just not display nicely, not sure why.

the command you want me to run, I attached it here.
[after_reboot_no_network.txt](https://github.com/openwrt/openwrt/files/12847994/after_reboot_no_network.txt)

[then_disable_ap_then_enable_ap.txt](https://github.com/openwrt/openwrt/files/12847996/then_disable_ap_then_enable_ap.txt)

brada4, 2023-10-09:
add regulatory domain like `option country 'GB'` to radio section. 

Ir simply cannot start on any channel unless receiving something first in global least-common-capability "region"

brada4, 2023-10-09:
Another try:
in /etc/modules.d find file containing module
`grep mt7921e /etc/modules.d/*`
Then after module name add the only parameter - supposedly `mt7921e disable_aspm=Y`
And try rebooting. (It should have been more evident if you attached full dmesg initially)
If it magically starts working just add the respective filename to /etc/sysupgrade.conf to keep setting over sysupgrade or auc.

brada4, 2023-10-09:
Also you might need to power off the card completely to change country code.

davidngrc, 2023-10-09:
vim /etc/modules.d/mt7921e:mt7921e
before content was empty, then add content mt7921e disable_aspm=Y

root@OpenWrt:~# grep mt7921e /etc/modules.d/*
/etc/modules.d/mt7921e:mt7921e disable_aspm=Y

I power off, and pull the AC wall plug, so there is no power on the PC and the wifi card inside.
re-plugin the AC wall plug, computer on, proxmox on, start openwrt vm
AP did not start.
then I get this full demsg
[full_dmesg.txt](https://github.com/openwrt/openwrt/files/12849394/full_dmesg.txt)

davidngrc, 2023-10-09:
I notice I might do something wrong, so 
vim /etc/modules.d/mt7921e:mt7921e
change content to 
disable_aspm=Y

reboot, AP still not associate with m2 card.

root@OpenWrt:~# grep mt7921e /etc/modules.d/*
show no output

root@OpenWrt:~# cat /etc/modules.d/mt7921e
disable_aspm=Y

here is the dmesg after I run above command.
[full_dmesg v2.txt](https://github.com/openwrt/openwrt/files/12849448/full_dmesg.v2.txt)

brada4, 2023-10-09:
You need to have line in that file in place of module name alone:

`mt7921e disable_aspm=Y`

brada4, 2023-10-09:
Explanation - you disabled module loading, no worries yet.

In dmesg:
```
 extended config space not accessible
```
Which is usually characteristic of a virtualisation.

Does it come up after power off/on of a VM?

brada4, 2023-10-09:
It is how proxmox and bios set the device.  Like around ACPI/APIC/PM....

If machine powerswitch fixes the problem -  it is fixable in domain.xml
<on_reboot>restart</on_reboot>

davidngrc, 2023-10-09:
I am on another proxmox server, inside a ubuntu vm, 
then using this ubuntu to ssh into that openwrt vm, 
doing all the vim in the ssh.

the proxmox is newly setup, all and bios are in default.

brada4, 2023-10-09:
It is how proxmox sets up devices according to bios first.
Then VM tries to  set them up, but does not see some config parts.
Then on shutdown it may not reset device from VM the way BIOS would reset it completely.
It is a lot of moving parts to sync up.

davidngrc, 2023-10-10:
I removed proxmox, and install openwrt-23.05.0-rc4-x86-64-generic-ext4-combined.img.gz to a usb drive.
and boot that usb in the same PC.

without country code and without disable_aspm=Y
the AP is up after reboot.

so this is related to proxmox.
but I think many people will use proxmox with openwrt, so could you guys help?
or I close this issue?

---------

in proxmox, I use command
# nano /etc/modprobe.d/vfio.conf
options vfio-pci ids=14c3:7961    #m2 mt7921 wifi card

# lspci -nnk
03:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7961]
	Subsystem: AzureWave Device [1a3b:4682]
	Kernel driver in use: vfio-pci
	Kernel modules: mt7921e

as you can see, it already using vfio-pci, so passthrough should not be a problem.
but still got this reboot issue.

brada4, 2023-10-10:
You isolated problem to poxmox, there is not much Openwrt can do.
You can  try to change pcie slots and bios settings around acpi/apic/power, it may work differently.
Already suggested to map reboot to domain restart you did not consider or try.

davidngrc, 2023-10-10:
my bad, I am new to this, 
I search the keywork on "proxmox domain.xml"
there is no result, andbI nerver hear

closed 2023-10-10 by davidngrc

brada4, 2023-10-11:
quite frankly no idea, with normal libvirt/qemu/kvm that gives same machine/bios string one does `virsh edit vm-name`  and that way achieves VM poweroff and clean start in place of reboot inside running vm.

davidngrc, 2023-10-12:
When VM openwrt rebooting, there is no dmesg on proxmox, so no idea how to trace the proxmox problem.
However, I found a temp solution here
https://forum.openwrt.org/t/fixing-disabled-radio-on-reboot/142113
since I only got 1 m2 wifi card, so the radio index is 0
vim /etc/rc.local
`/sbin/wifi down radio0
sleep 1
/sbin/wifi up radio0
exit 0
`

doublep21, 2024-12-04:
@davidngrc sorry for the offtopic , can you use 2.4ghz and 5ghz bands at the same time in ap mode?