openwrt/openwrt #13647: WNDAP660 kernel fails to boot with official snapshot build (Wed Oct 11 2023)

linuxtim, 2023-10-11:
### Describe the bug

Not sure if this is expected, but whilst debugging an pcie problem, I tried the latest available snapshot on a Netgear WNDAP660.  This is the image:

netgear_wndap660-squashfs-sysupgrade.bin | ed881e3b4654a0b31d19b3d28faa5bf763f6d75fcec771e22ea6f09cf945df49 | 7390.7 KB | Wed Oct 11 00:06:05 2023

[netgear_wndap660-squashfs-sysupgrade.bin](https://downloads.openwrt.org/snapshots/targets/apm821xx/nand/openwrt-apm821xx-nand-netgear_wndap660-squashfs-sysupgrade.bin)	ed881e3b4654a0b31d19b3d28faa5bf763f6d75fcec771e22ea6f09cf945df49	7390.7 KB	Wed Oct 11 00:06:05 2023

... looks like a failure during early kernel init.  Serial console messages below:

```
Loading from NAND 32MiB 3,3V 8-bit, offset 0x110000
   Image Name:    initramfs
   Image Type:   PowerPC Linux Multi-File Image (gzip compressed)
   Data Size:    4083668 Bytes =  3.9 MB
   Load Address: 00000000
   Entry Point:  00000000
   Contents:
      Image 0: 4050852 Bytes =  3.9 MB
      Image 1: 32 Bytes =  0 kB
      Image 2: 32768 Bytes = 32 kB
## Booting kernel from Legacy Image at 00600000 ...
   Image Name:    initramfs
   Image Type:   PowerPC Linux Multi-File Image (gzip compressed)
   Data Size:    4083668 Bytes =  3.9 MB
   Load Address: 00000000
   Entry Point:  00000000
   Contents:
      Image 0: 4050852 Bytes =  3.9 MB
      Image 1: 32 Bytes =  0 kB
      Image 2: 32768 Bytes = 32 kB
   Verifying Checksum ... OK
## Loading init Ramdisk from multi component Legacy Image at 00600000 ...
## Flattened Device Tree from multi component Image at 00600000
   Booting using the fdt at 0x9dd014
   Uncompressing Multi-File Image ... OK
WARNING: legacy format multi component image overwritten
   Loading Ramdisk to 0fdef000, end 0fdef020 ... OK
```

... then nothing.  Only the power light is illuminated.

The device was then recovered via tftp boot and subsequently flashed with `OpenWrt 23.05.0-rc4, r23482-7fe85ce1f2` - which worked correctly as expected.

### OpenWrt version

snapshot 11 Oct 2023

### OpenWrt target/subtarget

apm821xx/nand

### Device

Netgear WNDAP660

### Image kind

Official downloaded image

### Steps to reproduce

Flash snapshot build (netgear_wndap660-squashfs-sysupgrade.bin	5efb24d83bba0a3c3088777a7eaef2617ab524c84d8e914d1749dda2e4dc96f0	7680.7 KB	Fri Apr 28 02:55:59 2023) from https://downloads.openwrt.org/snapshots/targets/apm821xx/nand/openwrt-apm821xx-nand-netgear_wndap660-squashfs-sysupgrade.bin

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

chunkeey, 2023-10-21:
Hmm, I just tried the last snapshot from the 19.

netgear_wndap660-squashfs-sysupgrade.bin | a6c37a12355c48cac827817714f1c468775cc9f373a64ea82c37aefcce3709a0 | 7390.7 KB | Thu Oct 19 18:22:32 2023
-- | -- | -- | --

[netgear_wndap660-squashfs-sysupgrade.bin](https://downloads.openwrt.org/snapshots/targets/apm821xx/nand/openwrt-apm821xx-nand-netgear_wndap660-squashfs-sysupgrade.bin)	a6c37a12355c48cac827817714f1c468775cc9f373a64ea82c37aefcce3709a0	7390.7 KB	Thu Oct 19 18:22:32 2023

and the device boots here. (Granted, mine is stock. Never swapped the wifis)

One quirk with the WNDAP 620+660 since 5.15 is that the kernel is always on the brink of being to big for the volume.
Though, you should be able to unbrick these through uboot (no reflash necessary) with the instructions in:

https://github.com/openwrt/openwrt/commit/b2c9c4dda28dd9bd82df629d96d626d2c236f382

    Please note: Netgear WNDAP660 & WNDAP620 users:
    
    Due to the kernel's size increase, uboot will likely break
    because it is overwrite the kernel during decompression.
    
    To fix this (and debrick affected devices, no reflash
    necessary), attach the RJ45-Serial-Console cable and
    enter the following in the uboot prompt during bootup:
    
     setenv kernel_addr_r 1100000
     saveenv
     run bootcmd

joanandk, 2026-01-11:
What should one do if the flash is corrupted too?

`tftpboot ${kernel_addr_r} openwrt-24.10.5-apm821xx-nand-netgear_wndap660-squashfs-factory.img;nand erase 0x110000 0x600000;nand write.jffs2 ${kernel_addr_r} 0x110000 ${kernel_size}`

Leads to 
`Wrong Image Format for bootm command ERROR: can't get kernel image!`

If I use factory.img, then the boot is looped unable to mount mtd4. This results from changed kernel_addr_r iIsuppose?

Thanks for any help getting the addresses right.