openwrt/openwrt #13649: IPQ4019  ethernet

zhangsd20, 2023-10-11:
### Describe the bug

Network cable directly connects router and computer.
Now the ping packet delay is greater than 2ms. This delay seems too large and abnormal.
And only openwrt firmware has this problem. Using qsdk firmware, the delay is less than 1ms.

root@OpenWrt:/# ping 192.168.1.10
PING 192.168.1.10 (192.168.1.10): 56 data bytes
64 bytes from 192.168.1.10: seq=0 ttl=64 time=0.612 ms
64 bytes from 192.168.1.10: seq=1 ttl=64 time=2.852 ms
64 bytes from 192.168.1.10: seq=2 ttl=64 time=2.719 ms
64 bytes from 192.168.1.10: seq=3 ttl=64 time=2.741 ms
64 bytes from 192.168.1.10: seq=4 ttl=64 time=2.808 ms
64 bytes from 192.168.1.10: seq=5 ttl=64 time=2.757 ms
64 bytes from 192.168.1.10: seq=6 ttl=64 time=2.626 ms

### OpenWrt version

openwrt21.02/22.03/23.05

### OpenWrt target/subtarget

ipq40xx/generic

### Device

e2600ac c1/c2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xback, 2023-10-11:
Please try this:

`echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor`

The problem is that the CPU clocks all the way down to 48MHz.
I have done power mearsurements while testing CPU scaling, and using the performance governor (and running the soc at full speed) has a power difference of 0.01 Watts ...

I have a patch locally changing the governor from Ondemand to Performance during build.

zhangsd20, 2023-10-11:
I tested according to the method you gave and found that this problem can be solved.
Could you please let me take a look at your patch? Rebuild the image locally and test it.

xback, 2023-10-11:
This is for kernel 6.1
It's similar for 5.15

```
From 1315dd46881995f3418f49bd4f7aea315c6147c9 Mon Sep 17 00:00:00 2001
From: Koen Vandeputte <koen.vandeputte@citymesh.com>
Date: Tue, 12 Sep 2023 15:38:27 +0200
Subject: [PATCH] ipq40xx: switch to performance governor by default

---
 target/linux/ipq40xx/config-6.1  | 4 ++--
 1 files changed, 2 insertions(+), 2 deletions(-)

diff --git a/target/linux/ipq40xx/config-6.1 b/target/linux/ipq40xx/config-6.1
index 2619f3f87c..e84823aacd 100644
--- a/target/linux/ipq40xx/config-6.1
+++ b/target/linux/ipq40xx/config-6.1
@@ -74,8 +74,8 @@ CONFIG_CPU_COPY_V6=y
 CONFIG_CPU_CP15=y
 CONFIG_CPU_CP15_MMU=y
 CONFIG_CPU_FREQ=y
-CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND=y
-# CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE is not set
+# CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND is not set
+CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE=y
 CONFIG_CPU_FREQ_GOV_ATTR_SET=y
 CONFIG_CPU_FREQ_GOV_COMMON=y
 # CONFIG_CPU_FREQ_GOV_CONSERVATIVE is not set
-- 
2.34.1
```

aiamadeus, 2023-10-12:
Perhaps we should disable 48MHz? Just like this commit did: d0d6b8e

xback, 2023-10-12:
> Perhaps we should disable 48MHz? Just like this commit did: [d0d6b8e](https://github.com/openwrt/openwrt/commit/d0d6b8e1833c587d0c50cac4f6324aa93b0bc8fc)

I understand to just upper the lowlimit here, but as the soc does not consume more power (and thus does not generate more heat), why not simply switch to performance?

It boots faster, feels a lot less jittery etc.

Just my 2 cents :-)