openwrt/openwrt #13650: USB does not work on compex wpj563

alex-pnzv, 2023-10-11:
### Describe the bug

Hi, I have compex wpj563 router with connected kingston usb flash drive that works on openwrt 19.07.6 without any problem. After updating to OpenWrt 22.03.5 router does not detect connected usb devices at all. I installed all the packages from the [manual](https://openwrt.org/docs/guide-user/storage/usb-installing)  but it did not help, the router does not see the connected flash drive. Please help with this problem.

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

Compex WPJ563

### Image kind

Official downloaded image

### Additional info

> [    0.000000] Linux version 5.10.176 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Thu Apr 27 20:28:15 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is Compex WPJ563
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 121092K/131072K available (5884K kernel code, 605K rwdata, 748K rodata, 1200K init, 210K bss, 9980K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000008] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008281] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074811] pid_max: default: 32768 minimum: 301
[    0.079804] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087522] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097718] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.108117] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.118526] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.125794] pinctrl core: initialized pinctrl subsystem
[    0.135285] NET: Registered protocol family 16
[    0.140727] thermal_sys: Registered thermal governor 'step_wise'
[    0.190160] clocksource: Switched to clocksource MIPS
[    0.203001] NET: Registered protocol family 2
[    0.207804] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.216202] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.225116] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... 135 lines cut ...]
[   35.879002] br-lan: port 1(eth0.1) entered disabled state
[   35.884917] device eth0.1 entered promiscuous mode
[   35.889877] device eth0 entered promiscuous mode
[   35.900259] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   35.938161] br-lan: port 1(eth0.1) entered blocking state
[   35.943793] br-lan: port 1(eth0.1) entered forwarding state

> lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M

> cat /sys/kernel/debug/usb/devices
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.10
S:  Manufacturer=Linux 5.10.176 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=1b000000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-11:
Do you have voltage tester to measure persistent 5V between USB sidemost pins i.e if it is powered. Or something like usb mouse or eqally cheap that lights up when powered?
No fault on your side. Your router supports USB 2.0 only, you have no effect from USB1.1 or USB 3.0 or UAS drivers. They are safe to remove.

xback, 2023-10-11:
@alex-pnzv 

Can you try this patch?

```
From c4d592eac979c1bf441a99c086c98b6a2ba2e84b Mon Sep 17 00:00:00 2001
From: Koen Vandeputte <koen.vandeputte@citymesh.com>
Date: Wed, 11 Oct 2023 15:53:24 +0200
Subject: [PATCH] ath79: wpj563: enable 2nd usb controller

testpatch

---
 target/linux/ath79/dts/qca9563_compex_wpj563.dts | 8 ++++++++
 1 file changed, 8 insertions(+)

diff --git a/target/linux/ath79/dts/qca9563_compex_wpj563.dts b/target/linux/ath79/dts/qca9563_compex_wpj563.dts
index ceac53f415..407e912ad2 100644
--- a/target/linux/ath79/dts/qca9563_compex_wpj563.dts
+++ b/target/linux/ath79/dts/qca9563_compex_wpj563.dts
@@ -130,6 +130,14 @@
 	status = "okay";
 };
 
+&usb_phy1 {
+	status = "okay";
+};
+
+&usb1 {
+	status = "okay";
+};
+
 &wmac {
 	status = "okay";
 
-- 
2.34.1

```

alex-pnzv, 2023-10-12:
@brada4 I have verified that there is indeed 5 volts present on the USB pins.
@xback I created an image with your patch and see that the USB port is now working.

> lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=usb-storage, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M

Can you add this patch to the openwrt repository? It would be nice to have this fix in future releases.

Thanks for help, guys!

xback, 2023-10-12:
@alex-pnzv 
Great,

Thanks for testing.
I'll push patches to openwrt tomorrow, including stable branches.
This fix will pop-up on later service releases then.

Enjoy the board :)

closed 2023-10-13 by openwrt-bot

xback, 2023-10-13:
@alex-pnzv 

Patches pushed to:
- master
- 23.05
- 22.03
- 21.02

commit 9188c77c (2023-10-13): ath79: wpj563: enable 2nd USB controller

The compex WPJ563 actually has both usb controllers wired:

usb0 --> pci-e slot
usb1 --> pin header

As the board exposes it for generic use, enable this controller too.

fixes: #13650
commit 0bc33a65 (2023-10-13): ath79: wpj563: enable 2nd USB controller

The compex WPJ563 actually has both usb controllers wired:

usb0 --> pci-e slot
usb1 --> pin header

As the board exposes it for generic use, enable this controller too.

fixes: #13650

(cherry picked from commit 9188c77cbee55a933d0fa75c74e175fbc52c556d)
commit 9e8c9592 (2023-10-13): ath79: wpj563: enable 2nd USB controller

The compex WPJ563 actually has both usb controllers wired:

usb0 --> pci-e slot
usb1 --> pin header

As the board exposes it for generic use, enable this controller too.

fixes: #13650

(cherry picked from commit 9188c77cbee55a933d0fa75c74e175fbc52c556d)
commit 42374bce (2023-10-13): ath79: wpj563: enable 2nd USB controller

The compex WPJ563 actually has both usb controllers wired:

usb0 --> pci-e slot
usb1 --> pin header

As the board exposes it for generic use, enable this controller too.

fixes: #13650

(cherry picked from commit 9188c77cbee55a933d0fa75c74e175fbc52c556d)