openwrt/openwrt #13668: Lucu has wrong option for "multicast to unicast"

micw, 2023-10-12:
### Describe the bug

The Option "multicast_to_unicast" for wifi has been renamed to "multicast_to_unicast_all" in https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=ab61232b0abe2158c346f9fbf17f34e64de5288a.

The option has not be renamed in luci

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ALL

### Device

ALL

### Image kind

Official downloaded image

### Steps to reproduce

* In luci open a wifi network
* Go to advanced
* Check "Multi To Unicast"
* Save
* click on "unsaved changes" to see what is changed

### Actual behaviour

Changes:
/etc/config/wireless
uci set wireless.lan_2g.multicast_to_unicast='1'

### Expected behaviour

Changes:
/etc/config/wireless
uci set wireless.lan_2g.multicast_to_unicast_all='1'

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ditti4, 2023-10-14:
You're running 22.03, right? This was fixed on master as well as on the openwrt-23.05 branch:
* https://github.com/openwrt/luci/commit/6069f4f228305e3e0b59cb6deb9de9934aa68b6f
* https://github.com/openwrt/luci/commit/895a508251bddbdaec1f1232c093d178555b7d00

It looks like that commit simply needs to be cherry-picked onto the openwrt-22.03 branch as well. But since this is LuCI-specific, it might be more fruitful to file another issue on the LuCI repository.

micw, 2023-10-14:
Might be. I'm running the stable which is the latest 22.x.

micw, 2023-10-16:
Just upgraded to 23.05.0 and can confirm that it's fixed there.

closed 2023-10-16 by micw