openwrt/openwrt #13669: The entire download section of „firmware-selector.openwrt.org“ and the associated „downloads.openwrt.org“ are down

dziugas1959, 2023-10-12:
### Describe the bug

In short, all of the links go to a 404 page. 
[image]
Doesn't matter if it's an snapshot or a release build.
I think this also affects „sysupgrade“ as it also doesn't find anything“
And yes, this is included in this „Github“ repo, so it can be called a bug, since it's an official link/website.

### OpenWrt version

X

### OpenWrt target/subtarget

X

### Device

X

### Image kind

Official downloaded image

### Steps to reproduce

Go to this: https://downloads.openwrt.org/undefined/openwrt-23.05.0-ramips-mt7621-mercusys_mr70x-v1-squashfs-factory.bin 
Or this: https://downloads.openwrt.org/undefined/openwrt-23.05.0-ath79-generic-asus_pl-ac56-squashfs-sysupgrade.bin 

### Actual behaviour

A 404 page is displayed.

### Expected behaviour

A download starts.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-12:
Probably some glitch during r23 preparation. You can find not yet announced firmwares here:
https://downloads.openwrt.org/releases/23.05.0/targets/

brada4, 2023-10-12:
ping @aparcar 
Al archive unreachable via selector
https://firmware-selector.openwrt.org/?version=19.07.10&target=ar71xx%2Fgeneric&id=tl-wr1043nd-v1

aparcar, 2023-10-12:
I rolled back the firmware selector and it works again.

@mwarning please check what broke it and do another release. Keep in mind that whatever changes you do to the collect.py script are not automatically deployed on the production servers.

closed 2023-10-12 by aparcar

mwarning, 2023-10-13:
@aparcar isn't collect.py called in the gitlab ci? The pipeline output suggests that only the output is deployed.