openwrt/openwrt #13670: USB Stick MT7601 not working after installing drivers (Solved)

Arku3S, 2023-10-12:
### Describe the bug

After installing the drivers, the USB stick does not work.

### OpenWrt version

LuCI openwrt-22.03 branch (git-23.119.80898-65ef406) / OpenWrt 22.03-SNAPSHOT r20220-92a0dd2447 The official version plus several add-ons and packages downloaded from https://eko.one.pl/

### OpenWrt target/subtarget

MT7601u

### Device

TP-LINK TL-WR1043ND V4

### Image kind

Official downloaded image

### Steps to reproduce

Simply connect the 7601 USB Stick to your router and install the drivers.

### Actual behaviour

It doesn't work and nothing happens.

### Expected behaviour

I expected it to work after installing the drivers.

### Additional info

Here is the solution:
After installing kmod-mt7601u and mt7601u-firmware
The file named mt7601u.bin is in the wrong location.
The wrong location is: "/lib/firmware/Mediatek"
The correct location is "/lib/firmware"
Moving the file to the correct location and restarting the router solves the problem.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-13:
You must report your problem to the fork you downloaded from your link. 
https://dl.eko.one.pl/openwrt-22.03/targets/ath79/generic/

brada4, 2023-10-13:
Firmware was  moved with v19
https://github.com/openwrt/mt76/pull/342

mt76x0u works for me uninterrupted since v18 btw.

Arku3S, 2023-10-14:
It seems that the firmware path was changed in the latest version 23.05 and I am using 22, so it is not an error.

closed 2023-10-14 by Arku3S

brada4, 2023-10-14:
Wrong bug tracker for your problem.