openwrt/openwrt #13671: bpf-headers

oskarirauta, 2023-10-12:
### Describe the bug

With testing-kernel (6.1) selected, bpf header attempts to download 5.1 branch kernel from 6.1 url.
I think that if you use testing kernel, bpf headers from 6.1 should be used, both work; fixing download url for 5.1 kernel, or to change it to download 6.1 kernel.

With 6.1 kernel patches work (while writing this) just fine, though there is some minor offset reported, but if bpf-headers source should be chosen depending on selected kernel version, I think there should directories way they are in targets: patches-5.1 and patches 6.1.

It seems that there is this issue every now and then, because I had this one earlier, resolved it (fixed url that time), and now I had it again. But probably because kernel source file's sum matched, dl url didn't matter until I decided to save some space and remove _some old downloads_..

Maybe someone should take a look at it, severity of issue is not very bad; overcoming it is quite easy to fix.

### OpenWrt version

current git

### OpenWrt target/subtarget

x86/64

### Device

Interceptor 10210U

### Image kind

Self-built image

### Steps to reproduce

build with nearly all?

### Actual behaviour

file for download missing due to wrong url.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

anomeome, 2023-10-12:
Seems to happen every time there is a 5.15.x push

schm0, 2023-11-05:
However, the use of the 6.1 header breaks qosify.

commit a15eb1cd (2023-11-06): bpf-headers: Fix download URL with kernel 6.1

This fixes the download of the kernel 5.15 for the bpf-headers when
kernel 6.1 is build for the target.

Even if kernel 6.1 was selected for the target we still use kernel 5.15
for the bpf-headers. The download script tried to download the 5.15
kernel from the 6.x directory on kernel,org and this failed. Define
PKG_SOURCE_URL based on PKG_PATCHVER and not KERNEL_BASE like done in
kernel.mk.

Without this change it tries to download the kernel from this URL:
ttps://cdn.kernel.org/pub/linux/kernel/v6.x/linux-5.15.129.tar.xz

Fixes: #13190
Fixes: #13 [... cut]
PR #13884 (2023-11-06): bpf-headers: Fix download URL with kernel 6.1
This fixes the download of the kernel 5.15 for the bpf-headers when kernel 6.1 is build for the target.

Even if kernel 6.1 was selected for the target we still use kernel 5.15 for the bpf-headers. The download script tried to download the 5.15 kernel from the 6.x directory on kernel,org and this failed. Define PKG_SOURCE_URL based on PKG_PATCHVER and not KERNEL_BASE like done in kernel.mk.

Fixes: #13190
Fixes: #13671
Fixes: #13814
closed 2023-11-06 by openwrt-bot