openwrt/openwrt #13674: Asus RT-N56U “Image metadata not present” on upgrade

DoughMucker, 2023-10-13:
### Describe the bug

Upgrading the Asus RT-N56U requires a "force upgrade". The update completes successfully after forced.

### OpenWrt version

r23482-7fe85ce1f2 (has also happened in previous 2 stable releases)

### OpenWrt target/subtarget

ramips/rt3883

### Device

Asus RT-N56U

### Image kind

Official downloaded image

### Steps to reproduce

System->Backup/Flash Firmware->"Flash new firmware image". Select dowloaded firmware.

### Actual behaviour

https://imgur.com/V13j6AP

### Expected behaviour

An upgrade without a force being necessary.

### Additional info

A few users on the forum have experienced this.
https://forum.openwrt.org/t/asus-rt-n56u-image-metadata-not-present-on-upgrade/158472

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-13:
Can you go through the sha256 which firmware you download?

DoughMucker, 2023-10-14:
Yes. For example, 23.05.0 rc4 (r23497-6637af95aa) has matching hash:
`PS C:\Users\dougm\Downloads> certUtil -hashfile .\openwrt-23.05.0-ramips-rt3883-asus_rt-n56u-squashfs-sysupgrade.bin SHA256
SHA256 hash of .\openwrt-23.05.0-ramips-rt3883-asus_rt-n56u-squashfs-sysupgrade.bin:
644e414574bba5c5ea3e0eba99cee3d3f45942936fe13f366edfe9d8e5f03fbe
CertUtil: -hashfile command completed successfully.`

lmorrisonnautel, 2024-04-24:
This still appears to be an issue with 23.05.3.

goosnarrggh, 2024-04-25:
Confirmed, this is still an issue with the official 23.05.3 image.

**fwtool -i** reports "CRC error".

I note that the image builder goes through the normal process of `append-kernel | append-rootfs | pad-rootfs | check-size | append-metadata` but then the asus_rt-n56u target inside the rt3883.mk supplemental makefile goes on to append another step `| mkrtn56uimg -s`.

I wonder if maybe the mkrtn56uimg might manipulate the image in ways that would invalidate the previously applied metadata and/or its CRC signature in ways that fwtool doesn't know about and/or isn't prepared to accommodate? (I don't know enough about the image file structure to be able dig too deeply in this area right away.)

closed 2024-07-13 by openwrt-bot