openwrt/openwrt #13676: Lighthttpd random start after boot

Loafter, 2023-10-13:
### Describe the bug

Lighthttpd service random start at boot, force restart service execute normal and after that lighthttpd operate normal

log contain strange message about shutdown:

# 'cat' '/var/log/lighttpd/error.log'
2023-10-13 08:05:25: (../src/server.c.1909) server started (lighttpd/1.4.71)
2023-10-13 08:05:30: (../src/server.c.1216) [note] graceful shutdown started
2023-10-13 08:05:30: (../src/server.c.2308) server stopped by UID = 0 PID = 1

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

x86/64

### Device

Acer Veriton N2510G

### Image kind

Official downloaded image

### Steps to reproduce

install lighthttp reboot router and check status lighthttpd

### Actual behaviour

it's not work

### Expected behaviour

lighthttpd service operate normal

### Additional info

```
### Documentation
# https://wiki.lighttpd.net/
#
### Configuration Syntax
# https://wiki.lighttpd.net/Docs_Configuration
#
### Configuration Options
# https://wiki.lighttpd.net/Docs_ConfigurationOptions
#
### Configuration Variables (potentially used in /etc/lighttpd/conf.d/*.conf)
var.log_root    = "/var/log/lighttpd/"
var.server_root = "/www/"
var.state_dir   = "/var/run/"
var.home_dir    = "/var/lib/lighttpd/"
var.conf_dir    = "/etc/lighttpd"
var.vhosts_dir  = server_root + "/vhosts"
var.cache_dir   = "/var/cache/lighttpd"
var.socket_dir  = home_dir + "/sockets"
server.port = 8888
server.document-root        = server_root
server.upload-dirs          = ("/mnt/data/tmp")
#server.errorlog-use-syslog  = "enabled"
server.errorlog            = log_root + "error.log"
server.pid-file             = state_dir + "lighttpd.pid"
server.username             = "http"
server.groupname            = "www-data"

include "/etc/lighttpd/mime.conf"
include "/etc/lighttpd/conf.d/*.conf"

$HTTP["url"] =~ "^/photos($|/)" {
	server.dir-listing := "enable"
	dir-listing.encoding := "utf-8"
	server.document-root := "/mnt/data/"
	auth.backend = "plain"
	webdav.activate := "enable"
	auth.backend.plain.userfile = "/etc/lighttpd/webdav.shadow"
	auth.require = (
		"/photos/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
	)
}
$HTTP["url"] =~ "^/dav($|/)" {
	server.dir-listing := "enable"
	dir-listing.encoding := "utf-8"
	server.document-root := "/mnt/data/"
	auth.backend = "plain"
	webdav.activate := "enable"
	auth.backend.plain.userfile = "/etc/lighttpd/webdav.shadow"
	auth.require = (
		"/dav/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
	)
}
$HTTP["url"] =~ "^/pub($|/)" {
	server.dir-listing := "enable"
	dir-listing.encoding := "utf-8"
	server.document-root := "/mnt/data/"
	auth.backend = "plain"
	webdav.activate := "enable"
	auth.backend.plain.userfile = "/etc/lighttpd/public.shadow"
	auth.require = (
		"/pub/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
	)
}

$SERVER["socket"] == ":1443" {
	ssl.engine = "enable"
	ssl.pemfile ="/etc/acme/ai.XXX.de/ai.XXX.de.cer"
	ssl.privkey ="/etc/acme/ai.XXX.de/ai.XXX.de.key"
		$HTTP["url"] =~ "^/photos($|/)" {
			server.dir-listing := "enable"
			dir-listing.encoding := "utf-8"
			server.document-root := "/mnt/data/"
			auth.backend = "plain"
			webdav.activate := "enable"
			auth.backend.plain.userfile = "/etc/lighttpd/webdav.shadow"
			auth.require = (
				"/photos/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
			)
		}
		$HTTP["url"] =~ "^/dav($|/)" {
			server.dir-listing := "enable"
			dir-listing.encoding := "utf-8"
			server.document-root := "/mnt/data/"
			auth.backend = "plain"
			webdav.activate := "enable"
			auth.backend.plain.userfile = "/etc/lighttpd/webdav.shadow"
			auth.require = (
				"/dav/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
			)
		}
		$HTTP["url"] =~ "^/pub($|/)" {
			server.dir-listing := "enable"
			dir-listing.encoding := "utf-8"
			server.document-root := "/mnt/data/"
			auth.backend = "plain"
			webdav.activate := "enable"
			auth.backend.plain.userfile = "/etc/lighttpd/public.shadow"
			auth.require = (
				"/pub/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
			)
		}
### Services
		$HTTP["url"] =~ "^/websockify" {
			proxy.server = ( "" => ( ( "host" => "192.168.2.28", "port" => "27000" ) ) )
			proxy.header = ( "https-remap" => "enable", "upgrade" => "enable","map-urlpath" => ("/websockify/" =>"/" ) )
			auth.backend = "plain"
			auth.backend.plain.userfile = "/etc/lighttpd/pwservices.shadow"
			auth.require = (
			"/websockify/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
			)
		} 
		$HTTP["url"] =~ "^/car_switch" {
			proxy.server = ( "" => ( ( "host" => "192.168.6.3", "port" => "80" ) ) )
			proxy.header = ( "https-remap" => "enable", "upgrade" => "enable","map-urlpath" => ("/car_switch/" =>"/" ) )
			auth.backend = "plain"
			auth.backend.plain.userfile = "/etc/lighttpd/pwservices.shadow"
			auth.require = (
			"/car_switch/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
			)
		}
		$HTTP["url"] =~ "^/dell_switch" {
			proxy.server = ( "" => ( ( "host" => "192.168.2.17", "port" => "80" ) ) )
			proxy.header = ( "https-remap" => "enable", "upgrade" => "enable","map-urlpath" => ("/dell_switch/" =>"/" ) )
			auth.backend = "plain"
			auth.backend.plain.userfile = "/etc/lighttpd/pwservices.shadow"
			auth.require = (
			"/dell_switch/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
			)
		}
}
### Port Proxies

$SERVER["socket"] == ":2283" {                                                                                    
    ssl.engine = "enable"                                                     
    ssl.pemfile ="/etc/acme/ai.XXX.de/ai.XXX.de.cer"              
    ssl.privkey ="/etc/acme/ai.XXX.de/ai.XXX.de.key"     
    proxy.server = ( "" => ( ( "host" => "10.9.9.13", "port" => "2283" ) ) )                                                             
}

$SERVER["socket"] == ":8083" {                                                    
    ssl.engine = "enable"                                                         
    ssl.pemfile ="/etc/acme/ai.XXX.de/ai.XXX.de.cer"                  
    ssl.privkey ="/etc/acme/ai.XXX.de/ai.XXX.de.key"
    auth.backend = "plain"
    auth.backend.plain.userfile = "/etc/lighttpd/pwservices.shadow"
    auth.require = (
    "/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
   )
    proxy.header = ( "upgrade" => "enable")
    proxy.server = ( "" => ( ( "host" => "10.9.9.13", "port" => "8083" ) ) )      
}
 
$SERVER["socket"] == ":8084" {
    ssl.engine = "enable"
    ssl.pemfile ="/etc/acme/ai.XXX.de/ai.XXX.de.cer"
    ssl.privkey ="/etc/acme/ai.XXX.de/ai.XXX.de.key"
    auth.backend = "plain"
    auth.backend.plain.userfile = "/etc/lighttpd/pwservices.shadow"
    auth.require = (
    "/" => ("method" => "basic", "realm" => "disk", "require" => "valid-user")
   )
    proxy.header = ( "upgrade" => "enable")
    proxy.server = ( "" => ( ( "host" => "10.9.9.13", "port" => "8084" ) ) )
}
 

# customizations should generally be placed in separate files such as
#   /etc/lighttpd/conf.d/00_vars.conf    # override variables for conf.d/*.conf
#   /etc/lighttpd/conf.d/zz_custom.conf  # override other conf.d/*.conf settings

```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-13:
it should be reported to ../[ackages/
Not far on vendor doc you will find preflight check
`lighttpd -tt -f /etc/lighttpd/lighttpd.conf`
You probably have its output in `logread` already.

musashino205, 2023-10-13:
And please don't multi-post. (https://github.com/openwrt/openwrt/issues/13675)

gstrauss, 2024-01-05:
```
2023-10-13 08:05:25: (../src/server.c.1909) server started (lighttpd/1.4.71)
2023-10-13 08:05:30: (../src/server.c.1216) [note] graceful shutdown started
```
If lighttpd detects that the clock jumps (default: 1800 seconds; 30 mins), lighttpd will attempt to gracefully restart, waiting up to 5 seconds before attempting a non-graceful restart.  The restart is done due to underlying libraries such as openssl and TLSv1.2 session tickets not handling clock jumps well.  (The lighttpd core internally uses monotonic time where possible, so is less affected by clock jumps.)

There are probably more lines in your lighttpd error log which you did not share.

All openwrt maintained branches have lighttpd 1.4.73, so I suggest upgrading, or filing an issue with Acer to update their image.

closed 2024-01-05 by ynezz