openwrt/openwrt #13679: [Netgear DGN3500] Can't upgrade to 23.05.0: Image metadata not present

GabriRuflex, 2023-10-13:
### Describe the bug

Can't upgrade Netgear DG3500 from 22.03.5 to 23.05.0

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

lantiq/xway

### Device

Netgear DGN3500

### Image kind

Official downloaded image

### Steps to reproduce

Go to the `Flash operations` page (https://192.168.1.1/cgi-bin/luci/admin/system/flash) and click on `Flash new firmware image`

### Actual behaviour

This errors are returned:
```
It appears that you are trying to flash an image that does not fit into the flash memory, please verify the image file!
```

```
Thu Apr 27 20:35:27 UTC 2023 upgrade: Image metadata not present Thu Apr 27 20:35:27 UTC 2023 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware Image check failed.

The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform.
```

If I try to force the upgrade, the router can't boot anymore.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-13:
Are you flasing sysupgrade as noted in text where you upload it?

GabriRuflex, 2023-10-13:
I'm flashing the factory image:

https://downloads.openwrt.org/releases/23.05.0/targets/lantiq/xway/openwrt-23.05.0-lantiq-xway-netgear_dgn3500-squashfs-factory.img

dave14305, 2023-10-13:
Flash the sysupgrade image and then come back to close this issue.

brada4, 2023-10-13:
@GabriRuflex dont do it again, it may lead to hard brick.

GabriRuflex, 2023-10-13:
Flashing the sysupgrade image fixed the issue.

closed 2023-10-13 by GabriRuflex