openwrt/openwrt #13681: COMFAST CF-EW72 V2 5Ghz radio not working reliably

felipecrs, 2023-10-13:
### Describe the bug

I am having some problems with my 5Ghz radio for COMFAST CF-EW72 V2 which was recently integrated by https://github.com/openwrt/openwrt/pull/13328.

Sometimes it works, but most times it does not. On Luci it appears like this:

[image]

Also, the power shows as 3dbm for some reason.

And when I trigger a radio restart, this is what shows in the logs:

```
Wed Oct 11 22:25:48 2023 daemon.notice hostapd: Set new config for phy phy1:
Wed Oct 11 22:25:48 2023 kern.info kernel: [ 1792.945530] device phy1-ap0 left promiscuous mode
Wed Oct 11 22:25:48 2023 kern.info kernel: [ 1792.950283] br-lan: port 3(phy1-ap0) entered disabled state
Wed Oct 11 22:25:49 2023 daemon.notice netifd: lan (2165): udhcpc: sending renew to server 192.168.1.10
Wed Oct 11 22:25:49 2023 daemon.notice netifd: lan (2165): udhcpc: lease of 192.168.1.3 obtained from 192.168.1.10, lease time 86400
Wed Oct 11 22:25:49 2023 daemon.notice netifd: radio1 (2925): Command failed: Not found
Wed Oct 11 22:25:49 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Wed Oct 11 22:25:49 2023 daemon.notice netifd: radio1 (2957): WARNING: Variable 'data' does not exist or is not an array/object
Wed Oct 11 22:25:49 2023 daemon.notice hostapd: Set new config for phy phy1:
Wed Oct 11 22:25:49 2023 daemon.notice netifd: radio1 (2957): Command failed: Not found
Wed Oct 11 22:25:49 2023 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Wed Oct 11 22:25:49 2023 daemon.notice hostapd: Restart interface for phy phy1
Wed Oct 11 22:25:49 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=44 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-A-MPDU-LEN-EXP7] channel=44  interface=phy1-ap0 bssid=e0:e1:a9:b1:59:64 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=melancia wpa_psk_file=/var/run/hostapd-phy1-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=Felipe bridge=br-lan wds_bridge= snoop_
Wed Oct 11 22:25:49 2023 kern.info kernel: [ 1794.050447] br-lan: port 3(phy1-ap0) entered blocking state
Wed Oct 11 22:25:49 2023 kern.info kernel: [ 1794.056200] br-lan: port 3(phy1-ap0) entered disabled state
Wed Oct 11 22:25:49 2023 kern.info kernel: [ 1794.062182] device phy1-ap0 entered promiscuous mode
Wed Oct 11 22:25:50 2023 daemon.err hostapd: Could not set interface phy1-ap0 flags (UP): I/O error
Wed Oct 11 22:25:50 2023 daemon.err hostapd: nl80211: Could not set interface 'phy1-ap0' UP
Wed Oct 11 22:25:50 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Wed Oct 11 22:25:50 2023 daemon.err hostapd: nl80211 driver initialization failed.
Wed Oct 11 22:25:50 2023 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Wed Oct 11 22:25:50 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy1-ap0 wasn't started
Wed Oct 11 22:25:50 2023 daemon.notice hostapd: hostapd.add_iface failed for phy phy1 ifname=phy1-ap0
Wed Oct 11 22:25:50 2023 daemon.notice netifd: lan (2165): udhcpc: sending renew to server 192.168.1.10
Wed Oct 11 22:25:50 2023 daemon.notice netifd: lan (2165): udhcpc: lease of 192.168.1.3 obtained from 192.168.1.10, lease time 86400
Wed Oct 11 22:25:50 2023 daemon.notice netifd: Wireless device 'radio1' is now up
```

And this is the kernel logs:
<details>

```
[    0.000000] Linux version 5.15.134 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24111-66f6c20e45) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Oct 9 22:22:25 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is COMFAST CF-EW72 V2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s14736 r8192 d22128 u45056
[    0.000000] pcpu-alloc: s14736 r8192 d22128 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000870
[    0.000000] Readback ErrCtl register=00000870
[... 309 lines cut ...]

</details>

_Originally posted by @felipecrs in https://github.com/openwrt/openwrt/issues/13328#issuecomment-1758810585_
            

### OpenWrt version

r24124-518923178c

### OpenWrt target/subtarget

ramips/mt7621

### Device

COMFAST CF-EW72 V2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-14:
Screamed missing firmware files are here:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

felipecrs, 2023-10-14:
Hm... that's interesting. But am I supposed to do something with this information? I am new here.

brada4, 2023-10-14:
check presence and sha256sum of firmware files in /lib/firmware/mediatek

felipecrs, 2023-10-14:
Such file `mt7663pr2h.bin` is indeed missing, apparently:

```console
root@ap-da-rua:~# ls -la /lib/firmware/mediatek
drwxr-xr-x    2 root     root            68 Oct 13 04:27 .
drwxr-xr-x    3 root     root            94 Oct 13 04:27 ..
-rw-r--r--    1 root     root        346216 Oct 13 04:27 mt7663_n9_rebb.bin
-rw-r--r--    1 root     root        211518 Oct 13 04:27 mt7663pr2h_rebb.bin
```

brada4, 2023-10-14:
well, prepare some reset pin.

Try opkg search /lib/firmware/mediatek/ .... on each file there, uninstall the firmware package then upload files one by one from firmware repo as kernel asks

@nbd168 made some patch around semaphore message. Might be related.

felipecrs, 2023-10-14:
Ok... I am not very sure what you mean by "uninstall the firmware package", but here's what I did:

```
root@ap-da-rua:~# opkg search /lib/firmware/mediatek/mt7663pr2h.bin
root@ap-da-rua:~# opkg search /lib/firmware/mediatek/mt7663pr2h_rebb.bin
kmod-mt7663-firmware-ap - 5.15.135+2023-09-18-2afc7285-1
root@ap-da-rua:/lib/firmware/mediatek# wget https://git.kernel.org/pub/scm/linux
/kernel/git/firmware/linux-firmware.git/tree/mediatek/mt7663pr2h.bin -O /tmp/fil
e.bin
Downloading 'https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek/mt7663pr2h.bin'
Connecting to 139.178.84.217:443
Writing to '/tmp/file.bin'
/tmp/file.bin         1747k --:--:-- ETA
Download completed (1870868 bytes)
root@ap-da-rua:/lib/firmware/mediatek# mv -fv /tmp/file.bin /lib/firmware/mediat
ek/mt7663pr2h.bin
root@ap-da-rua:/lib/firmware/mediatek# ls -la /lib/firmware/mediatek/
drwxr-xr-x    1 root     root             0 Oct 13 09:34 .
drwxr-xr-x    1 root     root             0 Oct 13 04:27 ..
-rw-r--r--    1 root     root        346216 Oct 13 04:27 mt7663_n9_rebb.bin
-rw-r--r--    1 root     root       1870868 Oct 13 09:33 mt7663pr2h.bin
-rw-r--r--    1 root     root        211518 Oct 13 04:27 mt7663pr2h_rebb.bin
```

I will monitor how the AP behaves now.

felipecrs, 2023-10-14:
Ok... the file is there. But apparently nothing has changed. Even in the logs, it's still complaining about the file.

But if @nbd168 is indeed working on something, maybe I should just wait. Is there a PR I can follow?

felipecrs, 2023-10-14:
Thanks you by the way.

brada4, 2023-10-14:
like here on similar device:
https://github.com/openwrt/mt76/issues/513

felipecrs, 2023-10-18:
@filimonic is there any chance you just missed this file in #13328? I'm sorry, I'm just being naive wondering if it would be a simple fix.

darknblack, 2023-11-15:
hi any updates on this one?

brada4, 2023-11-15:
> Wed Oct 11 22:25:49 2023 daemon.notice netifd: radio1 (2957): WARNING: Variable 'data' does not exist or is not an array/object

This one is from ubus that some uci file has imperfect syntax like list vs option mismatch or missing tickmarks etc, probably due to restoring ancient conf file

> Configuration file: data: driver=nl80211

You try to run 80mhz wide channel specifying middle channel, it is ch36 by all means, some drivers take just first channel as valid id.

felipecrs, 2023-11-15:
> You try to run 80mhz wide channel specifying middle channel, it is ch36 by all means, some drivers take just first channel as valid id.

Do you have a suggested radio configuration for me to try?

brada4, 2023-11-15:
Auto + 160 or 80 depending on what clients can do....
The problem is you pick channel mid-band and half of it is dfs, no idea how that plays out...

felipecrs, 2023-11-18:
I tried these combinations and several others, unfortunately nothing changes. I can't make it work. :(

brada4, 2023-11-18:
Some logs from `logread -e  phy.-ap.` could help (passke ligged, use synthetic)

felipecrs, 2023-11-18:
```console
$ logread -e  phy.-ap. | grep phy1
Fri Nov 17 18:38:10 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-A-MPDU-LEN-EXP7] channel=36  interface=phy1-ap0 bssid=e0:e1:a9:b1:59:64 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=melancia wpa_psk_file=/var/run/hostapd-phy1-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=Felipe (5GHz) bridge=br-lan wds_bridge=
Fri Nov 17 18:38:10 2023 daemon.err hostapd: Could not set interface phy1-ap0 flags (UP): I/O error
Fri Nov 17 18:38:10 2023 daemon.err hostapd: nl80211: Could not set interface 'phy1-ap0' UP
Fri Nov 17 18:38:10 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Fri Nov 17 18:38:10 2023 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Fri Nov 17 18:38:10 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy1-ap0 wasn't started
Fri Nov 17 18:38:10 2023 daemon.notice hostapd: hostapd.add_iface failed for phy phy1 ifname=phy1-ap0
```

brada4, 2023-11-18:
It notes 160MHz in command line, public spec says 866Mbps ie 80MHz TX-STBC-2BY1 ..... It is rater cosmetic bug having unsupported rate in interface, but make it 80 and it should start (or at least change error message)

Please check `iw list` if it comes from the driver or luci invents....

felipecrs, 2023-11-18:
I am actually using 80MHz:

[image]

But I will do a power cycle to make sure I get new logs.

brada4, 2023-11-18:
3dBm is kind of placeholder of no transmission often.
The last config dump says channel list is 36 only, should be no channel list at all in 'auto' case.

felipecrs, 2023-11-18:
> if you set reg domain once

I am very sorry, but I don't know what do you mean by that.

After power cycle:

```console
$ logread -e  phy.-ap. | grep -v phy0
Fri Nov 17 18:38:22 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=US ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=42 vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160-80PLUS80][MAX-A-MPDU-LEN-EXP7] channel=36  interface=phy1-ap0 bssid=e0:e1:a9:b1:59:64 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=melancia wpa_psk_file=/var/run/hostapd-phy1-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=Felipe (5GHz) bridge=br-lan wds_bridge=
Fri Nov 17 18:41:56 2023 daemon.err hostapd: Could not set interface phy1-ap0 flags (UP): I/O error
Fri Nov 17 18:41:56 2023 daemon.err hostapd: nl80211: Could not set interface 'phy1-ap0' UP
Fri Nov 17 18:41:56 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Fri Nov 17 18:41:56 2023 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Fri Nov 17 18:41:56 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy1-ap0 wasn't started
Fri Nov 17 18:41:56 2023 daemon.notice hostapd: hostapd.add_iface failed for phy phy1 ifname=phy1-ap0
```

Which is apparently the same thing as before.

iw list:

<details>

  <summary>Click here to show</summary>

```console
$ iw list
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 4
        max scan IEs length: 539 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                Frequencies:
                        * 2412 MHz [1] (27.0 dBm)
                        * 2417 MHz [2] (27.0 dBm)
                        * 2422 MHz [3] (27.0 dBm)
                        * 2427 MHz [4] (27.0 dBm)
                        * 2432 MHz [5] (27.0 dBm)
[... 168 lines cut ...]
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported extended features:
                * [ RRM ]: RRM
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

</details>

brada4, 2023-11-18:
> t=100 chanlist=36 tx

This config file content is logged on failure:
```
# grep chan /var/run/hostapd-phy1.conf
channel=acs_survey
chan_util_avg_period=600
```
Is chanlist in /etc/config/wireless ????

(dd to delete line in vi, Esc:wq to exit)

regdom seems well applied......

felipecrs, 2023-11-19:
```console
$ grep chan /var/run/hostapd-phy1.conf
channel=36
chanlist=36
chan_util_avg_period=600
```

> Is chanlist in /etc/config/wireless ????

I guess it is not...?

```console
$ cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel 'auto'
        option band '2g'
        option htmode 'HT20'
        option country 'US'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Felipe'
        option encryption 'psk2'
        option key 'hidden'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option country 'US'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Felipe (5GHz)'
        option encryption 'psk2'
        option key 'hidden'
```

brada4, 2023-11-19:
> option channel '36'

make it auto in the file. Not significant in this case but you can name both AP-s same and your mobile devices will roam without dropping connections.

felipecrs, 2023-11-19:
> make it auto in the file.

Ok, done:

```console
$ cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel 'auto'
        option band '2g'
        option htmode 'HT20'
        option country 'US'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Felipe'
        option encryption 'psk2'
        option key 'hidden'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel 'auto'
        option band '5g'
        option htmode 'VHT80'
        option country 'US'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Felipe (5GHz)'
        option encryption 'psk2'
        option key 'hidden'
```

After restarting the radio in Luci, now Luci no longer shows channel 36 as being used:

[image]

> Not significant in this case but you can name both AP-s same and your mobile devices will roam without dropping connections.

I usually do that. This time I renamed the 5Ghz to make it easier to know when it's working or not.

brada4, 2023-11-19:
Very bad luck https://openwrt.org/unsupported/dfs , it will select only no-dfs channels, like 36 or 100
Please check ACS/DFS in `logread`

felipecrs, 2023-11-20:
Ok, thank you for providing this link. I understand that there is some missing implementation of the driver for the chip I'm using, which is indeed bad luck for me.

However, I don't get it. Can't I just stay away from these DFS channels? For example, could not I manually pick a non-DFS channel for example? I am struggling to understand this.

As per `logread`, I don't see anything relevant:

```console
root@ap-da-rua:~# logread | grep -i dfs
root@ap-da-rua:~# logread | grep -i acs
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
[... cut ...]
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: phy0-ap0: ACS-COMPLETED freq=2452 channel=9
Sat Nov 18 12:51:58 2023 daemon.notice hostapd: phy0-ap0: interface state ACS->ENABLED
```

brada4, 2023-11-20:
5ghz radio does not come up at all. 2.4 up-s correctly, just non-critical complaints about driver responses.

felipecrs, 2023-12-18:
Here is what I did, I flashed the sysupgrade image built by @filimonic attached to https://github.com/openwrt/openwrt/pull/13328, while performing a factory reset.

It's working ever since. I haven't attempted to upgrade again to a newer snapshot yet.

@brada4, if you are in the same situation, perhaps you can try doing the same.

brada4, 2023-12-18:
thanks for info, 23.05.2 got me enough of mt76 updates and i dont see the weird errors rhyming with your anymoe...

felipecrs, 2023-12-18:
Where exactly did you download your firmware from? Is it the regular snapshot from https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-comfast_cf-ew72-v2-squashfs-sysupgrade.bin?

felipecrs, 2024-01-08:
OMG, I found what my issue is.

I was following the [dumbap guide](https://openwrt.org/docs/guide-user/network/wifi/dumbap), and I had this in my `/etc/rc.local` (as per [this section](https://openwrt.org/docs/guide-user/network/wifi/dumbap)):

```bash
rm /usr/sbin/wpa_supplicant
```

Removing this line (and re-flashing firmware to restore the file) fixes the issue for me.

I wonder if that's an expected behavior. If it is, I would probably just remove that section from the dumbap guide as it can cause such issues even on true dumb aps.

closed 2024-01-08 by felipecrs

brada4, 2024-01-08:
> rm /usr/sbin/wpa_supplicant

im dropping bricks. literarily.

One can make more compact image swapping wpad-basic-mbedtls for hostapd-basic-mbedtls in firmware selector. Deleting file in overlayfs does literarily zero for space and function.