openwrt/openwrt #13683: hostapd may fail to setup 40MHz channel

InsaneKnight, 2023-10-14:
### Describe the bug

hostapd now start with ht_capab=[HT40+] or ht_capab=[HT40-] in the first place, making it unable to setup 40MHz channel when landing control channel lower (36,44,52,60) with ht_capab=[HT40-] or landing control channel higher (40,48,56,64) with ht_capab=[HT40+], mainly in auto mode, but it may also occur when channel is manually set.

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

ath79/nand

### Device

Netgear WNDR4500 v3

### Image kind

Self-built image

### Steps to reproduce

/etc/config/wireless:
config wifi-device 'radio1'
	option type 'mac80211'
	option country 'CN'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:00.0'
	option cell_density '0'
	option htmode 'HT40'
	option channels '36-64'
	option channel 'auto'

### Actual behaviour

Sat Oct 14 08:13:17 2023 daemon.notice hostapd: Set new config for phy phy1:
Sat Oct 14 08:13:18 2023 daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Sat Oct 14 08:13:18 2023 daemon.notice hostapd: Restart interface for phy phy1
Sat Oct 14 08:13:18 2023 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=CN ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 chanlist=36-64 #num_global_macaddr=1 ieee80211n=1 ht_coex=0 ht_capab=**[HT40+]**[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][DSSS_CCK-40] channel=acs_survey  interface=phy1-ap0 bssid=<bssid> ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=<passwd> wpa_psk_file=/var/run/hostapd-phy1-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=<ssid> bridge=br-lan wds_bridge= snoop_iface=br-lan ft_iface=br-lan mobility_domain=98f7 ft_psk_generate_local=1 ft_over_ds=1 reassociation_deadline=1000 wpa_disable_eapol_key_retries=1 wpa_key_mgmt=WPA-PSK FT-PSK okc=0 disable_pmksa_caching=1 dynami
Sat Oct 14 08:13:18 2023 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Oct 14 08:13:18 2023 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Sat Oct 14 08:13:18 2023 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->ACS
Sat Oct 14 08:13:18 2023 daemon.notice hostapd: phy1-ap0: ACS-STARTED
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5745 is missing noise floor
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5745 is missing channel time
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5745 is missing RX and busy time (at least one is required)
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5765 is missing noise floor
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5765 is missing channel time
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5765 is missing RX and busy time (at least one is required)
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5785 is missing noise floor
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5785 is missing channel time
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5785 is missing RX and busy time (at least one is required)
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5805 is missing noise floor
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5805 is missing channel time
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5805 is missing RX and busy time (at least one is required)
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5825 is missing noise floor
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5825 is missing channel time
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: ACS: Survey for freq 5825 is missing RX and busy time (at least one is required)
Sat Oct 14 08:13:23 2023 daemon.notice **hostapd: phy1-ap0: ACS-COMPLETED freq=5280 channel=56**
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: phy1-ap0: interface state ACS->HT_SCAN
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: Switch own primary and secondary channel to get secondary channel with no Beacons from other BSSes
Sat Oct 14 08:13:23 2023 daemon.err **hostapd: HT40 channel pair (60, 56) not allowed**
Sat Oct 14 08:13:23 2023 daemon.notice **hostapd: Fallback to 20 MHz**
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->DFS
Sat Oct 14 08:13:23 2023 daemon.notice hostapd: phy1-ap0: DFS-CAC-START freq=5300 chan=60 sec_chan=0, width=0, seg0=0, seg1=0, cac_time=60s
Sat Oct 14 08:13:54 2023 daemon.err hostapd: nl80211: kernel reports: key addition failed
Sat Oct 14 08:13:54 2023 daemon.info hostapd: phy0-ap0: STA b4:ee:b4:b7:09:5a IEEE 802.11: associated (aid 3)
Sat Oct 14 08:13:54 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED b4:ee:b4:b7:09:5a auth_alg=ft
Sat Oct 14 08:14:25 2023 daemon.notice hostapd: phy1-ap0: DFS-CAC-COMPLETED success=1 freq=5300 ht_enabled=1 chan_offset=0 chan_width=1 cf1=5300 cf2=0
Sat Oct 14 08:14:26 2023 daemon.notice hostapd: phy1-ap0: interface state DFS->ENABLED
[... 64 lines cut ...]
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_netgear_wndr4500-v3=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_netgear_wndr4500-v3=""
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_zte_mf281=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_zte_mf281=""
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_zte_mf282=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_zte_mf282=""
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_zte_mf286=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_zte_mf286=""
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_zte_mf286a=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_zte_mf286a=""
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_zte_mf286r=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_zte_mf286r=""
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_zyxel_emg2926_q10a=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_zyxel_emg2926_q10a=""
CONFIG_TARGET_DEVICE_ath79_nand_DEVICE_zyxel_nbg6716=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_nand_DEVICE_zyxel_nbg6716=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-14:
52..54..56 are DFS in CN. When ch60 wins acs survey it becomes 60-62-64

What does `iwinfo` say about channels?

InsaneKnight, 2023-10-14:
> 52..54..56 are DFS in CN. When ch60 wins acs survey it becomes 60-62-64

In the log above, it is ch56 which wins, and "HT40 channel pair (60, 56) not allowed".

And these even occurs:
```
Sat Oct 14 10:41:01 2023 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Oct 14 10:41:01 2023 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Sat Oct 14 10:41:01 2023 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->ACS
Sat Oct 14 10:41:01 2023 daemon.notice hostapd: phy1-ap0: ACS-STARTED
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5745 is missing noise floor
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5745 is missing channel time
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5745 is missing RX and busy time (at least one is required)
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5765 is missing noise floor
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5765 is missing channel time
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5765 is missing RX and busy time (at least one is required)
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5785 is missing noise floor
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5785 is missing channel time
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5785 is missing RX and busy time (at least one is required)
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5805 is missing noise floor
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5805 is missing channel time
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5805 is missing RX and busy time (at least one is required)
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5825 is missing noise floor
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5825 is missing channel time
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: ACS: Survey for freq 5825 is missing RX and busy time (at least one is required)
Sat Oct 14 10:41:06 2023 daemon.warn hostapd: phy1-ap0: IEEE 802.11 Configured channel (64) or frequency (5320) (secondary_channel=1) not found from the channel list of the current mode (2) IEEE 802.11a
Sat Oct 14 10:41:06 2023 daemon.warn hostapd: phy1-ap0: IEEE 802.11 Hardware does not support configured channel
Sat Oct 14 10:41:06 2023 daemon.err hostapd: ACS: Could not (err: -3) select hw_mode for freq=5320 channel=64
Sat Oct 14 10:41:06 2023 daemon.err hostapd: Interface initialization failed
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: phy1-ap0: interface state ACS->DISABLED
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Sat Oct 14 10:41:06 2023 daemon.err hostapd: ACS: Possibly channel configuration is invalid, please report this along with your config file.
Sat Oct 14 10:41:06 2023 daemon.err hostapd: ACS: Failed to start
Sat Oct 14 10:41:06 2023 daemon.notice hostapd: phy1-ap0: AP-DISABLED
```
with the same configs above, that is, "HT40+, ch64" failed, and unable to switch to "HT40-, ch64" to become (60, 64).
 
> What does `iwinfo` say about channels?

After the log same as the first reproduced, iwinfo says:
```
phy1-ap0  ESSID: "_ssid_"
          Access Point: _bssid_
          Mode: Master  Channel: 56 (5.280 GHz)  HT Mode: HT20
          Center Channel 1: 56 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -86 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11a/n
          Hardware: 168C:ABCD 0000:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
```

brada4, 2023-10-14:
Seems never recovers after? Like hostapd bug?
> Sat Oct 14 08:13:23 2023 daemon.notice hostapd: Switch own primary and secondary channel to get secondary channel with no Beacon

Does it work around if you set channels option to `36 44 52 60` (or those +4 for vice-versa)

InsaneKnight, 2023-10-14:
> Seems never recovers after?

Yes, recovery should occur only when it is kicked out by a DFS radar signal.

> Does it work around if you set channels option to `36 44 52 60` (or those +4 for vice-versa)

I have tried this, resulting in failing to set up 40MHz, as `36 44 52 60` means no neighbor channel available.

brada4, 2023-10-14:
It looks like it swaps channels and mixes up ht mode from then on?
What if you force HT40+ in conf?
(your assumtion on dfs operation is correct, wprst case fallback ch36 40+ or even 80 should always work)

InsaneKnight, 2023-10-14:
> It looks like it swaps channels and mixes up ht mode from then on? 

Yes.

> What if you force HT40+ in conf? 

Currently the best walkaround is to config like this:
```
config wifi-device 'radio1'
	option type 'mac80211'
	option country 'CN'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:00.0'
	option cell_density '0'
	option htmode 'HT40'
	option channels '36-64'
	option channel '64'
```

So initially hostapd can choose the correct direction (HT40-/+) to extend the default channel, and when DFS radar signal arrives, it can evade to another channel automatically, with correct extending direction.

brada4, 2023-10-14:
Still a bug. Attach `iw list` , your conf is quite generic, just unclear which driver capability triggers wrong radio config.

InsaneKnight, 2023-10-15:
> Still a bug. Attach `iw list` , your conf is quite generic, just unclear which driver capability triggers wrong radio config.
>
~# iw list
Wiphy phy1
    wiphy index: 1
    max # scan SSIDs: 4
    max scan IEs length: 2261 bytes
    max # sched scan SSIDs: 0
    max # match sets: 0
    Retry short limit: 7
    Retry long limit: 4
    Coverage class: 0 (up to 0m)
    Device supports AP-side u-APSD.
    Device supports T-DLS.
    Available Antennas: TX 0x7 RX 0x7
    Configured Antennas: TX 0x7 RX 0x7
    Supported interface modes:
         * IBSS
         * managed
         * AP
         * AP/VLAN
         * monitor
         * mesh point
         * P2P-client
         * P2P-GO
         * outside context of a BSS
    Band 2:
        Capabilities: 0x11ef
            RX LDPC
            HT20/HT40
            SM Power Save disabled
            RX HT20 SGI
            RX HT40 SGI
            TX STBC
            RX STBC 1-stream
            Max AMSDU length: 3839 bytes
            DSSS/CCK HT40
        Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
        Minimum RX AMPDU time spacing: 8 usec (0x06)
        HT TX/RX MCS rate indexes supported: 0-23
        Frequencies:
            * 5180 MHz [36] (20.0 dBm) (radar detection)
            * 5200 MHz [40] (20.0 dBm) (radar detection)
            * 5220 MHz [44] (20.0 dBm) (radar detection)
            * 5240 MHz [48] (20.0 dBm) (radar detection)
            * 5260 MHz [52] (20.0 dBm) (radar detection)
            * 5280 MHz [56] (20.0 dBm) (radar detection)
            * 5300 MHz [60] (20.0 dBm) (radar detection)
            * 5320 MHz [64] (20.0 dBm) (radar detection)
            * 5500 MHz [100] (disabled)
            * 5520 MHz [104] (disabled)
            * 5540 MHz [108] (disabled)
            * 5560 MHz [112] (disabled)
            * 5580 MHz [116] (disabled)
            * 5600 MHz [120] (disabled)
            * 5620 MHz [124] (disabled)
            * 5640 MHz [128] (disabled)
            * 5660 MHz [132] (disabled)
            * 5680 MHz [136] (disabled)
            * 5700 MHz [140] (disabled)
            * 5745 MHz [149] (28.0 dBm)
            * 5765 MHz [153] (28.0 dBm)
            * 5785 MHz [157] (28.0 dBm)
            * 5805 MHz [161] (28.0 dBm)
            * 5825 MHz [165] (28.0 dBm)
    valid interface combinations:
         * #{ managed } <= 2048, #{ AP, mesh point } <= 8, #{ P2P-client, P2P-GO } <= 1, #{ IBSS } <= 1,
           total <= 2048, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz }

    HT Capability overrides:
[... 79 lines cut ...]
    HT Capability overrides:
         * MCS: ff ff ff ff ff ff ff ff ff ff
         * maximum A-MSDU length
         * supported channel width
         * short GI for 40 MHz
         * max A-MPDU length exponent
         * min MPDU start spacing
    max # scan plans: 1
    max scan plan interval: -1
    max scan plan iterations: 0
    Supported extended features:
        * [ RRM ]: RRM
        * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
        * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
        * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
        * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
        * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
        * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
        * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
        * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
        * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
        * [ DEL_IBSS_STA ]: deletion of IBSS station support
        * [ MULTICAST_REGISTRATIONS ]: mgmt frame registration for multicast
        * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
        * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support

Neustradamus, 2024-11-03:
@InsaneKnight: Have you always the problem?

InsaneKnight, 2024-11-03:
> @InsaneKnight: Have you always the problem?
>
Yes, this problem persists in 23.05.5. I should walk around this as in https://github.com/openwrt/openwrt/issues/13683#issuecomment-1762977206

Neustradamus, 2024-11-03:
For me the hostapd updates have not been for 23.05.x branch.
Have you tested master code?

InsaneKnight, 2024-11-04:
> For me the hostapd updates have not been for 23.05.x branch.
> Have you tested master code?
>
Master seems to always select "ch8n-4, HT40+" when performing automatic channel selection.