openwrt/openwrt #13688: Instructions destroyed device! Thank you!

KackTuT, 2023-10-14:
### Describe the bug

I followed everything strictly according to the instructions! Unfortunately my new device now only flashes! These weird warnings about not installing something but still posting the link. Why isn't this marked and numbered properly? I uploaded the 19.** kernel via the OEM interface and everything seemed to work smoothly. Then the shock! Blink blink! So 5 times and then break! What can I do now? If I left it as it was, I urgently need to use this device! Since the device was the same before and Openwrt was flashed.I wanted it   

### OpenWrt version

As explained in your description!

### OpenWrt target/subtarget

Ha?

### Device

Strong1200

### Image kind

Official downloaded image

### Steps to reproduce

Your instructions are out of date!

### Actual behaviour

It is like it is

### Expected behaviour

Different!

### Additional info

NopeNope

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MichaelUray, 2023-10-14:
Probably duplicate of https://github.com/openwrt/openwrt/issues/12637

There is already a post in the forum regarding this matter:
https://forum.openwrt.org/t/strong-1200-firmware-crossflash-2-times-failure/172213

KackTuT, 2023-10-14:
Unfortunately, this only confirms my suspicion that the router is now worthless to me! Because they publish a faulty tutorial! Super thank you! In der Zeit in der ich den Router zum laufen bringen verdiene ich so viel Geld das ich mir 10 neue kaufen könnte! Daher ist es sinnlos meine Zeit zu verschwenden. Aber danke für die Antwort!--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

raenye, 2023-10-14:
@KackTuT this approach gets you nothing.

There's inherent risk in updating operating systems, no matter whether it's a desktop, a server, or an embedded device. While commercial vendors are supposed (perhaps obliged) to help paying customers when a software update goes wrong,

OpenWrt is a volunteer-based project where you could find kind people who will try to assist you in fixing your device. You have to invest some effort, you have to provide useful debugging information, and above all you have to change your attitude.

As per the issue above, your router can be restored via serial (UART) access. If you care, read all the comments there, try to make it work for your device, and ask for specific help if you get stuck.

jow-, 2023-10-19:
Duplicate of #12637

closed 2023-10-19 by jow-