openwrt/openwrt #13691: Ip route command doesn't bring a line starting with default

fluttirci, 2023-10-14:
### Describe the bug

Cannot connect to the internet anymore since I flashed 23.05. I have Asus RT-N18U router I use as my main router, connected to the GPON. No weird configs, no VPN's etc. Just default config. I don't even have WIFI because of Broadcom chipset.

For example pinging google.com brings this:
ping: bad address 'google.com'

Then I setup the internet as PPPOE as what my ISP requires, exactly the same setup which worked in 22.03.05. But now exact same setup doesn't work. Here is my ip route command in 23.05:
192.168.2.0/24 dev br-lan scope link src 192.168.2.1

And my /etc/config/network file is like this below. I have written personal data as SECRET. Also wan.35 is required for my ISP requires vlan id 35.

config interface 'loopback'
option device 'lo'
option proto 'static'
option ipaddr '127.0.0.1'
option netmask '255.0.0.0'

config globals 'globals'
option packet_steering '1'
option ula_prefix 'SOMEIPV6PREFIX'

config device
option name 'br-lan'
option type 'bridge'
list ports 'lan1'
list ports 'lan2'
list ports 'lan3'
list ports 'lan4'

config interface 'lan'
option device 'br-lan'
option proto 'static'
option netmask '255.255.255.0'
option ip6assign '60'
option ipaddr '192.168.2.1'

config device
option name 'wan'
option macaddr 'SECRET'

config interface 'wan'
option proto 'pppoe'
option device 'wan.35'
option username 'SECRET'
option password 'SECRET'
option ipv6 'auto'

config interface 'wan6'
option device 'wan'
option proto 'dhcpv6'

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Asus RT-N18U (BCM47081)

### Image kind

Official downloaded image

### Steps to reproduce

Just flashing the 23.05 firmware said in the wiki link, and flashing it through luci interface.

### Actual behaviour

Internet broke. No default ip route. Pings doesn't work.

### Expected behaviour

It should of just simply connect to the internet, also ip route must return some lines starting with default. 

### Additional info

I flashed back 22.03.05, a working configuration I had and did ip route. It brings this. I removed the ip's and wrote someip1 and someip2 instead. Cuz those might be personal data.

default via someip1 dev pppoe-wan

someip1 dev pppoe-wan scope link src someip2

192.168.2.0/24 dev br-lan scope link src 192.168.2.1

Whereas in 23.05 my ip route is like this:
192.168.2.0/24 dev br-lan scope link src 192.168.2.1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-14:
You should mention it might be related to #13598.

fluttirci, 2023-10-14:
> You should mention you asked about it on #13598.
> 
> Also, since this is a `bcm53xx` device, this could be related to #13548. Can you try 23.05-rc2 and -rc3 and see if rc2 works while rc3 doesn't?

Hi, I've been trying what you said in the mean time. I tried all rc2, rc3 and even rc1 and none of them worked. With each of them, I flashed, I performed reset, I incorrected the time(when flashed the time was wrong, I corrected it via browser) and tried to set up PPPOE the way my ISP intended, still nothing. None of it work. There in #13548 , there is a patch file. That wouldn't work for my problem would it?

raenye, 2023-10-14:
If it doesn't work with -rc2, it's not #13548, sorry.
Not an expert about VLANs, but your network config sounds fine to me.

BenD780x9, 2023-10-15:
I have exactly the samw issue with Xiaomi Mi Router 4A Gigabit Edition v2, but I have VLAN's.
Nothing seems to work.

fluttirci, 2023-10-17:
> I have exactly the same issue with Xiaomi Mi Router 4A Gigabit Edition v2, but I have VLAN's. Nothing seems to work.

I also have vlan's. But on WAN interface, as you see wan.35 means wan's vlan id is 35. This setup is used to work but not anymore.

fluttirci, 2023-10-18:
Some more info on this, on the working 22.03.05, there is a device automatically created as "pppoe-wan" but on 23.05 no such device. See on the ss below. This is the 22.03.05, working one:
[image]
Here this is the 23.05, no pppoe-wan:
[image]
Also this is the 22.03.05 again:
[image]
And the same page on 23.05:
[image]

fluttirci, 2023-12-09:
I have this exact same issue on RT-AC88U, they use that bcm53xx driver, or target whatever. Although much faster CPU. No default route, no internet.