openwrt/openwrt #13692: Wireless adapters disapeared

jglooije, 2023-10-14:
### Describe the bug

After upgrading to the latest trunk my wireless adapters did not work anymore also after a hard reset. Reverting to
OpenWrt SNAPSHOT r24100-22b0c36ae7 / LuCI Master git-23.266.27574-7744ad0 solved the problem.

### OpenWrt version

ccf24b5dcd14fbdd5fb24a3f1f1eadcb89b22ea0

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Link DAP-X1860 A1

### Image kind

Self-built image

### Steps to reproduce

Build latest trunk and upgrade target

### Actual behaviour

Empty wireless config file after sysupgrade -n

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_dlink_dap-x1860-a1=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
# CONFIG_BINUTILS_USE_VERSION_2_40 is not set
CONFIG_BINUTILS_USE_VERSION_2_41=y
CONFIG_BINUTILS_VERSION="2.41"
CONFIG_BINUTILS_VERSION_2_41=y
# CONFIG_GCC_USE_VERSION_12 is not set
CONFIG_GCC_USE_VERSION_13=y
CONFIG_GCC_VERSION="13.2.0"
CONFIG_GCC_VERSION_13=y
CONFIG_KERNEL_IO_URING=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_hostapd-openssl=y
# CONFIG_PACKAGE_jansson is not set
CONFIG_PACKAGE_kmod-cryptodev=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
# CONFIG_PACKAGE_kmod-nft-fib is not set
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libmbedtls is not set
# CONFIG_PACKAGE_libmnl is not set
# CONFIG_PACKAGE_libnftnl is not set
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libopenssl-devcrypto=y
CONFIG_PACKAGE_libopenssl-legacy=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libuci-lua=y
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-ledtrig-rssi=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-json=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_nftables-json is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
CONFIG_PACKAGE_openssl-util=y
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_usteer=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PACKAGE_zoneinfo-europe=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-16:
Perhaps try something similar to [this](https://github.com/openwrt/openwrt/issues/13700#issuecomment-1764127003)?

neheb, 2023-10-17:
@jglooije can you post output of

```
hexdump -C /dev/mtd2ro| head
```

jglooije, 2023-10-17:
> @jglooije can you post output of
> 
> ```
> hexdump -C /dev/mtd2ro| head
> ```

00000000  15 79 00 00 04 ba d6 09  89 88 00 0c 43 26 59 97  |.y..........C&Y.|
00000010  15 79 c3 14 00 80 02 00  15 79 c3 14 ec 94 01 00  |.y.......y......|
00000020  16 79 c3 14 00 80 02 00  16 79 c3 14 cc 94 01 00  |.y.......y......|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000040  00 00 00 00 00 00 00 00  00 00 80 00 00 00 00 00  |................|
00000050  01 00 80 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  00 00 07 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  30 04 0b 00 00 00 06 af  00 00 00 00 00 00 00 00  |0...............|
00000080  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

neheb, 2023-10-17:
Another 80

raenye, 2023-10-22:
Please check whether `kmod-mt7915-firmware` is installed.

closed 2023-11-28 by jglooije