openwrt/openwrt #13694: RPI4 - brcmfmac crash and errors - snapshot

AQPSal, 2023-10-14:
### Describe the bug

RPI4 running OpenWrt SNAPSHOT r24112-67e465b32e; US regulatory

brcmfmac sdio crashes, eventually seems to load and stay functional; then multiple obss errors (see below). This is with Wifi not enabled.

When WIFI is enabled, then there is multiple "channel" errors and performance seems to suffer a bit (anecdotal - but LUCI does seem sluggish). Second section of dmesg shows it.

[    5.469589] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.478455] usbcore: registered new interface driver brcmfmac
[    5.486323] kmodloader: done loading kernel modules from /etc/modules.d/*
[    5.545781] usb 2-1: reset SuperSpeed USB device number 2 using xhci_hcd
[    5.613197] r8152 2-1:1.0: load rtl8153a-4 v2 02/07/20 successfully
[    5.656262] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    5.656296] r8152 2-1:1.0 eth1: v1.12.13
[    5.664290] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jul 29 2022 02:15:20 version 7.45.250 (70e9766 CY) FWID 01-e53e306b
[    8.060748] br-lan: port 1(eth1) entered blocking state
[    8.066121] br-lan: port 1(eth1) entered disabled state
[    8.071503] device eth1 entered promiscuous mode
[    8.077972] br-lan: port 1(eth1) entered blocking state
[    8.083221] br-lan: port 1(eth1) entered forwarding state
[    8.106203] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.115431] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.115371] br-lan: port 1(eth1) entered disabled state
[   10.735380] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   10.740934] r8152 2-1:1.0 eth1: carrier on
[   10.745731] br-lan: port 1(eth1) entered blocking state
[   10.750971] br-lan: port 1(eth1) entered forwarding state
[   10.756594] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   10.763206] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.2: link becomes ready
[   10.769867] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.7: link becomes ready
[   10.776484] IPv6: ADDRCONF(NETDEV_CHANGE): eth1.8: link becomes ready
[   10.783429] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   10.789128] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   11.275800] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.282455] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   11.767747] CIFS: Attempting to mount \\10.0.1.38\OpenWRT
[   11.846720] CIFS: Attempting to mount \\10.0.1.38\Transmission
[   12.325912] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   12.331619] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   12.485533] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   12.491373] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   12.511628] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[   15.045219] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   15.051070] brcmfmac: brcmf_dump_obss: dump_obss error (-110)
[   15.062253] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   15.155707] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   15.168547] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   15.186045] mmc1: card 0001 removed
[   15.295189] mmc1: new high speed SDIO card at address 0001
[   15.304614] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   15.458015] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   15.466102] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jul 29 2022 02:15:20 version 7.45.250 (70e9766 CY) FWID 01-e53e306b
[   35.136115] r8152 2-1:1.0 eth1: Promiscuous mode enabled
[   42.544486] ieee80211 phy1: brcmf_fw_crashed: Firmware has halted or crashed
[   45.037103] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   45.043005] brcmfmac: brcmf_dump_obss: dump_obss error (-110)
[   45.127802] mmc1: card 0001 removed
[   45.231212] mmc1: new high speed SDIO card at address 0001
[   45.240291] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   45.393121] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   45.401152] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jul 29 2022 02:15:20 version 7.45.250 (70e9766 CY) FWID 01-e53e306b
[   50.786625] ieee80211 phy2: brcmf_fw_crashed: Firmware has halted or crashed
[   53.279188] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   53.285063] brcmfmac: brcmf_dump_obss: dump_obss error (-110)
[   53.480492] mmc1: card 0001 removed
[   53.587113] mmc1: new high speed SDIO card at address 0001
[   53.597374] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   53.750822] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   53.758846] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jul 29 2022 02:15:20 version 7.45.250 (70e9766 CY) FWID 01-e53e306b
[... 56 lines cut ...]
[ 1388.943119] net_ratelimit: 50 callbacks suppressed

### OpenWrt version

r24112-67e465b32e

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Official downloaded image

### Steps to reproduce

Normal operation from boot

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-17:
Could you please post your `/etc/config/wireless` (with redacted SSID and passphrase)?

AQPSal, 2023-10-17:
As requested. Thanks.

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option channel '153'
        option band '5g'
        option htmode 'VHT20'
        option country 'US'
        option cell_density '0'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key 'xxxxxxxxxx'
        option network 'lan'

raenye, 2023-10-17:
And what was the last working version?

AQPSal, 2023-10-17:
Not really sure - it must have happened over the last month or so - I only noticed some sluggishness last week and when investigating I found dmesg errors. When disabling wifi the sluggishness seems to improve, and the errors change. I do not have the prior snapshot I was using. I did setup another RPi4 around the end of August with a snapshot version and had no errors. Similar simple wifi configuration.

Sorry I cannot be more helpful

AQPSal, 2023-10-17:
I did some further experimentation to verify it was not a hardware issue. I downloaded a plain image of the latest snapshot and set up a wireless network. Everything worked with no error messages. However, after playing with the new image a bit, I noticed the wireless network was using the "default" country setting. I changed it to "US" and the "set channel" errors started. 

I went back to the RPi4 in question and selected driver default there - error messages stopped - both types.

Hope this helps.

raenye, 2023-10-17:
Thanks. What is the output of `iw reg get` (with or without US settings)?

AQPSal, 2023-10-17:
With US setting

global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#3
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

raenye, 2023-10-20:
No idea, sorry.

AQPSal, 2023-11-10:
I pinned down the slow response in Luci to Luci  calling iwinfo. 

I seemed to have fixed the issue by using the latest NVRAM file used in bullseye from the RP OS.

satmandu, 2024-01-24:
> I pinned down the slow response in Luci to Luci calling iwinfo.
> 
> I seemed to have fixed the issue by using the latest NVRAM file used in bullseye from the RP OS.

What firmware file did you copy over to make this work?

Noltari, 2024-07-02:
> I pinned down the slow response in Luci to Luci calling iwinfo.

This was fixed with 512b762ddbb602b744f0e7a443cf4ac7422155c1

> I seemed to have fixed the issue by using the latest NVRAM file used in bullseye from the RP OS.

This is no longer hapenning on latest main branch.

closed 2024-07-02 by Noltari