openwrt/openwrt #13695: armsr: sysupgrade loses settings with the squashfs image

Kistelini, 2023-10-14:
### Describe the bug

Applying an armsr armv8 squashfs image via sysupgrade wipes the configuration instead of preserving it.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

armsr/armv8

### Device

QEMU QEMU Virtual Machine

### Image kind

Official downloaded image

### Steps to reproduce

Apply an armsr armv8 squashfs image via sysupgrade
I tried 23.05.0-rc4 -> 23.05.0 and 23.05.0 -> 23.05.0

### Actual behaviour

The update applies, but all configuration is lost

### Expected behaviour

Keep the configuration

### Additional info

This does not affect the ext4 images.

CC @mcbridematt 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-14:
what did you TRY ? like in steps.

mcbridematt, 2023-10-14:
armsr was developed on ext4 only and the squashfs images basically happened by accident.

This problem isn't hard to fix, the sysupgrade script needs to be modified to put sysupgrade.tgz in the /boot partition (rather than /) and a pre-init script needs to be added to copy the config backup into / again (same as https://github.com/openwrt/openwrt/blob/main/target/linux/x86/base-files/lib/preinit/79_move_config )

closed 2023-10-19 by openwrt-bot