openwrt/openwrt #13696: Test kernel 6.1 on ramips mt7620

InBomBi, 2023-10-15:
### Describe the bug

`Applying /workdir/openwrt-build/target/linux/ramips/patches-6.1/000-v5.18-02-staging-mt7621-dts-align-resets-with-binding-documen.patch using plaintext:
can't find file to patch at input line 24
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 64b2d6ffff862c0e7278198b4229e42e1abb3bb1 Mon Sep 17 00:00:00 2001
|From: Sergio Paracuellos <sergio.paracuellos@gmail.com>
|Date: Mon, 10 Jan 2022 12:49:30 +0100
|Subject: [PATCH 2/2] staging: mt7621-dts: align resets with binding documentation
|MIME-Version: 1.0
|Content-Type: text/plain; charset=UTF-8
|Content-Transfer-Encoding: 8bit
|
|Binding documentation for compatible 'mediatek,mt7621-sysc' has been updated
|to be used as a reset provider. Align reset related bits and system controller
|node with binding documentation along the dtsi file.
|
|Tested-by: Arınç ÜNAL <arinc.unal@arinc9.com>
|Reviewed-by: Philipp Zabel <p.zabel@pengutronix.de>|Signed-off-by: Sergio Paracuellos <sergio.paracuellos@gmail.com>
|Link: https://lore.kernel.org/r/20220110114930.1406665-5-sergio.paracuellos@gmail.com
|Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>
|---
| drivers/staging/mt7621-dts/mt7621.dtsi | 21 +++++++++------------
| 1 file changed, 9 insertions(+), 12 deletions(-)
|
|--- a/drivers/staging/mt7621-dts/mt7621.dtsi
|+++ b/drivers/staging/mt7621-dts/mt7621.dtsi
--------------------------
No file to patch.  Skipping patch.
10 out of 10 hunks ignored
Patch failed!  Please fix /workdir/openwrt-build/target/linux/ramips/patches-6.1/000-v5.18-02-staging-mt7621-dts-align-resets-with-binding-documen.patch!`

### OpenWrt version

Snapshot

### OpenWrt target/subtarget

Ramips, Mt7620

### Device

Miwifi mi r3

### Image kind

Self-built image

### Steps to reproduce

Turn on test kernel 6.1 in .config
Include patches 5.15 in Makefile.

### Actual behaviour

build image as normal.
Make -j

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2023-10-15:
The 6.1 testing kernel is not yet ready for ramips target.

InBomBi, 2023-10-15:
Waiting for release

closed 2023-10-15 by InBomBi