openwrt/openwrt #13700: Broken 5GHz radio on Unifi 6 Lite

d4nys3k, 2023-10-15:
### Describe the bug

On Unifi 6 Lite  with recent [snapshot](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=ubnt_unifi-6-lite) build I'm noticiing issues similar  to https://github.com/openwrt/mt76/issues/462 with 5GHz interface (_2GHz radio isn't affected_).  Snapshot image I downloaded on September 29  (_running on kernel 5.15.132, distrib revision r24054-fe10f97439_) is still working as expected, but recent build (_running on kernel 5.15.135_) from October 14 isn't, there's  `error -22` reported by mt7915e driver. 

I tried to review recent changes and I noticed commit https://github.com/openwrt/openwrt/commit/9ac891f8c44124e931c15f1257347cd8ba311a19 - which may be cause of this issue. I also tried reprogram latest factory image from UBNT and then migrate back to OpenWrt, but there're still issues with 5GHz radio and I believe that eeprom content (_factory/mtd2_) isn't corrupt after stock firmware was reuploaded on

### OpenWrt version

r24124-518923178c

### OpenWrt target/subtarget

ramips/mt7621

### Device

Ubiquiti UniFi 6 Lite

### Image kind

Official downloaded image

### Steps to reproduce

Install build revision r24054-fe10f97439 on U6lite and 5GHz radio will be working as expected. Install build revision r24124-518923178c, you'll see `[   50.736668] mt7915e: probe of 0000:02:00.0 failed with error -22` in dmesg and `iw phy` will not show 5GHz radio.

### Actual behaviour

5GHz unusable with snapshot release r24124-518923178c

### Expected behaviour

Usable 5GHz radio

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-15:
Anything peculiar in the bootlog?

d4nys3k, 2023-10-16:
> Anything peculiar in the bootlog?

Nothing. `mt7915e` module loading just fails in modprobe stage (_also tried unload&modprobe manually_), there's nothing logged elsewhere, only error mentioned above in dmesg.

d4nys3k, 2023-10-16:
I tried to revert https://github.com/openwrt/openwrt/commit/9ac891f8c44124e931c15f1257347cd8ba311a19 DTS modification on revision r24137-2e57028424 from git, no other modifications except this one:

```
--- target/linux/ramips/dts/mt7621_ubnt_unifi-6-lite.dts.orig	2023-10-16 08:13:05.576677277 +0200
+++ target/linux/ramips/dts/mt7621_ubnt_unifi-6-lite.dts	2023-10-16 10:49:14.589946899 +0200
@@ -108,8 +108,8 @@
 &wlan_5g {
 	compatible = "mediatek,mt76";
 
-	nvmem-cells = <&eeprom_factory_20000>, <&macaddr_eeprom_6>;
-	nvmem-cell-names = "eeprom", "mac-address";
+	nvmem-cells = <&macaddr_eeprom_6>;
+	nvmem-cell-names = "mac-address";
 
 	/* This is a workaround.
 	 *
```

Without DTS modification, modprobe fails with error -22....
```
[   50.947143] mt7915e 0000:02:00.0: WM Firmware Version: ____000000, Build Time: 20220929104145
[   50.980466] mt7915e 0000:02:00.0: WA Firmware Version: DEV_000000, Build Time: 20220929104205
[   51.098586] mt7915e: probe of 0000:02:00.0 failed with error -22
[   51.105634] kmodloader: done loading kernel modules from /etc/modules.d/*
```

With modified DTS mt7915e module loads without any error and 5GHz wifi works as expected (clients can connect to it)
```
[   50.956559] mt7915e 0000:02:00.0: WM Firmware Version: ____000000, Build Time: 20220929104145
[   50.990360] mt7915e 0000:02:00.0: WA Firmware Version: DEV_000000, Build Time: 20220929104205
[   51.153931] kmodloader: done loading kernel modules from /etc/modules.d/*
```

So it's clear that @DragonBluep / @Ansuel 's commit https://github.com/openwrt/openwrt/commit/9ac891f8c44124e931c15f1257347cd8ba311a19  breaks 5GHz wifi at least on on Unifi 6 Lite hardware. Maybe there's some custom UBNT modification in factory eeprom, which causes failure in `mt7915e` modprobe.

DragonBluep, 2023-10-16:
@d4nys3k Can you upload the 'factory' partition here so that I can do some analysis?

d4nys3k, 2023-10-16:
> @d4nys3k Can you upload the 'factory' partition here so that I can do some analysis?

Sure, [here](https://github.com/openwrt/openwrt/files/12915432/factory.bin.gz)'s content of factory partition (_/dev/mtd2_) from my Unifi 6 Lite.

DragonBluep, 2023-10-16:
@Ansuel Maybe we need to partial revert MT7915 nvmem eeprom commit https://github.com/openwrt/openwrt/commit/9ac891f8c44124e931c15f1257347cd8ba311a19.

I found the MT7915 is different from other chips. The driver will load first eeprom part (size 0xe00) and then check if Byte[62] is equal to `0x07`. If yes, the driver will load second part eeprom (pre calibration data).

@d4nys3k's eeprom:
[image]

mt76 eeprom load code:
```
static int mt7915_eeprom_load_precal(struct mt7915_dev *dev)
{
	struct mt76_dev *mdev = &dev->mt76;
	u8 *eeprom = mdev->eeprom.data;
// u32 val = eeprom[62];
	u32 val = eeprom[MT_EE_DO_PRE_CAL];
	u32 offs;

	if (!dev->flash_mode)
		return 0;
//if (val != 0x07)
	if (val != (MT_EE_WIFI_CAL_DPD | MT_EE_WIFI_CAL_GROUP))
		return 0;

	val = MT_EE_CAL_GROUP_SIZE + MT_EE_CAL_DPD_SIZE;
	dev->cal = devm_kzalloc(mdev->dev, val, GFP_KERNEL);
	if (!dev->cal)
		return -ENOMEM;

	offs = is_mt7915(&dev->mt76) ? MT_EE_PRECAL : MT_EE_PRECAL_V2;
// load second part, pre calibration data
	return mt76_get_of_eeprom(mdev, dev->cal, offs, val);
}
```

We can still keep the changes for `JCG`, `H3C` and `Haier` series devices, I can sure they didn't need second eeprom part. @neheb should know if linksys e7350 /belkin rt1800 are working properly with the patch.

Ansuel, 2023-10-16:
@DragonBluep if it's way too much work to fix it quickly, go for the partial revert.

neheb, 2023-10-16:
eeprom looks like

```
root@OpenWrt:~# hexdump -C /dev/mtd2ro | head
00000000  15 79 00 00 00 0c 43 26  46 e8 00 0c 43 26 59 97  |.y....C&F...C&Y.|
00000010  15 79 c3 14 00 80 02 00  15 79 c3 14 ec 8c 01 00  |.y.......y......|
00000020  16 79 c3 14 00 80 02 00  16 79 c3 14 ec 90 01 00  |.y.......y......|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000040  00 00 00 00 00 00 00 00  00 00 80 00 00 00 00 00  |................|
00000050  01 00 80 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  30 24 01 00 00 00 07 ac  da 00 00 00 00 00 00 00  |0$..............|
00000080  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
```

closed 2023-10-17 by openwrt-bot