openwrt/openwrt #13701: 5Ghz radio not working on 23.05.0 - Linksys E8450/Belkin RT3200

CrazyLemon, 2023-10-15:
### Describe the bug

Upgraded from 22.03 to 23.05 and notice that 5Ghz radio is not active.

> Mon Oct  9 23:45:39 2023 daemon.notice netifd: radio1 (2153): WARNING: Variable 'data' does not exist or is not an array/object
Mon Oct  9 23:45:39 2023 daemon.notice netifd: radio0 (2152): WARNING: Variable 'data' does not exist or is not an array/object
Mon Oct  9 23:45:39 2023 daemon.notice netifd: radio1 (2153): Phy not found
Mon Oct  9 23:45:39 2023 daemon.notice netifd: radio1 (2153): Could not find PHY for device 'radio1'
Mon Oct  9 23:45:39 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
Mon Oct  9 23:45:39 2023 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Mon Oct  9 23:45:39 2023 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon Oct  9 23:45:39 2023 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/network
Mon Oct  9 23:45:39 2023 daemon.notice netifd: radio1 (2211): WARNING: Variable 'data' does not exist or is not an array/object
Mon Oct  9 23:45:39 2023 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/wireless
Mon Oct  9 23:45:39 2023 daemon.notice netifd: radio1 (2211): Bug: PHY is undefined for device 'radio1'
Mon Oct  9 23:45:39 2023 daemon.notice hostapd: Set new config for phy wl0:
Mon Oct  9 23:45:39 2023 daemon.notice netifd: Wireless device 'radio1' is now down

After i manually disable and reenable the interface i get the same message

> Sun Oct 15 20:59:05 2023 daemon.notice netifd: radio1 (6556): WARNING: Variable 'data' does not exist or is not an array/object
Sun Oct 15 20:59:05 2023 daemon.notice netifd: radio1 (6556): Phy not found
Sun Oct 15 20:59:05 2023 daemon.notice netifd: radio1 (6556): Could not find PHY for device 'radio1'
Sun Oct 15 20:59:05 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
Sun Oct 15 20:59:05 2023 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Sun Oct 15 20:59:05 2023 daemon.notice netifd: radio1 (6585): WARNING: Variable 'data' does not exist or is not an array/object
Sun Oct 15 20:59:05 2023 daemon.notice netifd: radio1 (6585): Bug: PHY is undefined for device 'radio1'
Sun Oct 15 20:59:05 2023 daemon.notice netifd: Wireless device 'radio1' is now down

[image]

### OpenWrt version

23.05.0 - r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade Linksys E8450/Belkin RT3200 from 22.03 to 23.05 via Attended Sysupgrade

### Actual behaviour

5Ghz radio is not working

### Expected behaviour

5Ghz radio should work

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-15:
Could be related to #13700.

CrazyLemon, 2023-10-17:
> config wifi-device 'radio0'
> 	option type 'mac80211'
> 	option path 'platform/18000000.wmac'
> 	option channel '1'
> 	option band '2g'
> 	option htmode 'HT20'
> 	option txpower '20'
> 	option cell_density '0'
> 
> config wifi-iface 'default_radio0'
> 
> 	option device 'radio0'
> 	option network 'lan'
> 	option mode 'ap'
> 	option ssid 'OpenWRT'
> 	option encryption 'psk2'
> 	option key 'password'
> 	list maclist 'AA:BB:CC:DD:EE:11'
> 
> config wifi-device 'radio1'
> 	option type 'mac80211'
> 	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
> 	option cell_density '0'
> 	option country 'SI'
> 	option disabled '1'
> 
> config wifi-iface 'default_radio1'
> 	option device 'radio1'
> 	option network 'lan'
> 	option mode 'ap'
> 	option ssid 'OpenWRT'
> 	option encryption 'psk2'
> 	option key 'password'
> 	list maclist 'AA:BB:CC:DD:EE:11'
> 	option disabled '1'

CrazyLemon, 2023-10-17:
Doesn't work. After that change everything is disabled (2.4ghz and 5ghz)

raenye, 2023-10-17:
What happens if you delete the file and let openwrt reconfigure?

buggsdummy, 2023-10-18:
+1 on this issue, just picked up this router too

CrazyLemon, 2023-10-18:
> What happens if you delete the file and let openwrt reconfigure?

This happens

> config wifi-device 'radio0'
> 	option type 'mac80211'
> 	option phy 'wl0'
> 	option channel '1'
> 	option band '2g'
> 	option htmode 'HT20'
> 
> config wifi-iface 'default_radio0'
> 	option device 'radio0'
> 	option network 'lan'
> 	option mode 'ap'
> 	option ssid 'OpenWrt'
> 	option encryption 'none'
>
doesn't work/help.

raenye, 2023-10-18:
Hmm, so neither radio is even detected.
Does the last part of your `/etc/board.json` look like this:
```
        "wlan": {
                "wl0": {
                        "path": "platform/18000000.wmac"
                },
                "wl1": {
                        "path": "1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0"
                }
        }
```

Also, does this happen also with release candidate versions (-rc1, -rc2, -rc3 or -rc4)?
This could help pinpoint the code change responsible.

raenye, 2023-10-18:
There's a report [here](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1602) of a similar problem on a similar `mt7622+mt7915` device.

@nbd168 could this be a `netifd` issue, as per [this](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1605)?

CrazyLemon, 2023-10-19:
> Hmm, so neither radio is even detected. Does the last part of your `/etc/board.json` look like this:
> 
> ```
>         "wlan": {
>                 "wl0": {
>                         "path": "platform/18000000.wmac"
>                 },
>                 "wl1": {
>                         "path": "1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0"
>                 }
>         }
> ```
> 
> Also, does this happen also with release candidate versions (-rc1, -rc2, or -rc3)? This could help pinpoint the code change responsible.

It does

```
 "wlan": {
                "wl0": {
                        "path": "platform/18000000.wmac"
                },
                "wl1": {
                        "path": "1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0"
                },
                "wl2": {
                        "path": "1a145000.pcie/pci0001:00/0001:00:01.0/0001:01:00.0"
                }
        }
```

Did not test any of the RCs and honestly don't have the time.

raenye, 2023-10-20:
I don't think your `/etc/board.json` makes sense. E8450 has only one 5 GHz chip AFAIK.

CrazyLemon, 2023-10-20:
'wl2' does not make sense  but also does not make any difference.

I tried changing my config/wireless settings from
`option path 'platform/18000000.wmac'`
to
`option phy 'wl0'`

and 2.4GHz radio works as it should.

But 5GHz radio "behaves" like it's missing entirely.

nilsbebelaar, 2023-10-21:
Just wanted to chip in to mention I'm having the same issue with both of my Belkin RT3200 units.
The 5GHz radio is being detected in LuCI, but as 'generic unknown'. If there's any testing I can do I'm happy to tryout some ideas.
[image]

StaRipper, 2023-10-21:
[    7.050974] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_rom_patch.bin failed with error -2
[    7.060924] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_rom_patch.bin
[    7.286665] mt7915e: probe of 0000:01:00.0 failed with error -12

raenye, 2023-10-21:
`kmod-mt7915-firmware` should be on by default for RT3200/E8450... very weird.

CrazyLemon, 2023-10-22:
Can confirm that installing `kmod-mt7915-firmware` fixes the issue.

I as well used AUC to upgrade to 23.05 and it's interesting didn't install/upgrade it.

Thank you for your help.

closed 2023-10-22 by CrazyLemon